arendst/Tasmota #9261: URL redirect: tasmota.github.io gets 404

carras, 2020-09-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

**tasmota.github.io shouldn't get 404 but redirect to tasmota.github.io/docs/**

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2020-09-09:
It actually should since its a github limitation, we don't control the domain name or redirects

closed 2020-09-09 by blakadder

carras, 2020-12-16:
https://osxfuse.github.io this one works

I guess you just need to put a page in the root. Then tasmota.github.io will give a proper root page, or a least redirect to avoid 404.

blakadder, 2020-12-16:
can you tell me how to do that?

carras, 2020-12-18:
> can you tell me how to do that?

Sorry, I may know how to do that in a normal web server, but idk much about GitHub... Sorry. I would have done an index.html... but that seems too basic... idk.

But yeah, it seems to redirect now. With what you did here: https://github.com/tasmota/tasmota.github.io

So well done! 👍

blakadder, 2020-12-18:
yeah i found it but forgot to mention it sorry. thanks for telling me its possible!