arendst/Tasmota #9262: Tasmota for Tuya WRG1 Module (RTL8711AM)

kirovilya, 2020-09-08:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I have not found Tasmota firmware for Tuya WRG1 Module. 

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I would like to use a cheap zigbee gateway in a local network (without cloud and Tuya App). Same as with Sonoff Zigbee Bridge with Tasmota firmware.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Sonoff Zigbee Bridge with Tasmota firmware.

**Additional context**  
_Add any other context or screenshots about the feature request here._
Tuya wireless Hub (https://aliexpress.ru/item/4000465421911.html)

[image]
[image]
[image]
[image]
[image]

Datasheet for WRG1 module https://developer.tuya.com/en/docs/iot/device-development/module/wifi-module/wrg1-datasheet

A similar issue https://github.com/arendst/Tasmota/issues/8583

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-08:
As long there is no full Arduino RTL8711 support this will not be possible since Tasmota needs the Arduino layer

sfromis, 2020-09-08:
For the foreseeable future, the only way of getting Tasmota on that device would be to replace the WRG1 module with something including a supported chip, which may need a birds nest of wiring to get pin compatibility.

Jason2866, 2020-09-08:
Changing the WIFI module is not enough. The zigbee module is not supported. It is a different chip as in the zigbeebridge.
There will be no support for this device

ascillato2, 2020-09-08:
Closing this issue as this device has unsupported chips. 

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-08 by ascillato2

Stoatwblr, 2020-12-27:
It looks like realtek might be helping out: 

https://www.amebaiot.com/en/ameba-arduino-getting-started-rtl8710/

ascillato, 2020-12-27:
Good news!!!!!

Arduino Core for RT wifi SoC: https://github.com/Ameba8195/Arduino

Arduino Board Package: https://github.com/Ameba8195/Arduino/raw/master/release/package_realtek.com_ameba_index.json