arendst/Tasmota #9263: Energy Totals Incorrect - Arlec PC399HA Smart Energy Plug

smithy2167, 2020-09-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The Energy Totals displayed and in MQTT messages are incorrect - they are about 1/6 what they should be.
The Voltage, Current and Power readings have been calibrated and are all correct.
First noticed when running version 8.4.0.3
Downgrading to version 8.4.0 fixed the problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Arlec PC399HA Smart Energy Plug
- [x] Tasmota binary firmware version number used: 8.5.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
17:02:07 CMD: Backlog Template; Module; GPIO 255
17:02:07 MQT: stat/Smart_Energy_Switch/RESULT = {"NAME":"PC399HA","GPIO":[0,0,0,17,134,132,0,0,131,158,21,0,0],"FLAG":0,"BASE":52}
17:02:08 MQT: stat/Smart_Energy_Switch/RESULT = {"Module":{"0":"PC399HA"}}
17:02:08 MQT: stat/Smart_Energy_Switch/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"17":"Button1"},"GPIO4":{"134":"BL0937 CF"},"GPIO5":{"132":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"158":"LedLinki"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
17:04:13 CMD: status 0
17:04:13 MQT: stat/Smart_Energy_Switch/STATUS = {"Status":{"Module":0,"DeviceName":"Smart Energy Switch","FriendlyName":["Smart Energy Switch"],"Topic":"Smart_Energy_Switch","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:04:13 MQT: stat/Smart_Energy_Switch/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:18:57","StartupUTC":"2020-09-08T06:45:16","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2020-08-27T00:43:26","SaveCount":148,"SaveAddress":"F8000"}}
17:04:13 MQT: stat/Smart_Energy_Switch/STATUS2 = {"StatusFWR":{"Version":"8.5.0(tasmota)","BuildDateTime":"2020-09-07T10:35:04","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"450/699"}}
17:04:13 MQT: stat/Smart_Energy_Switch/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Froogle","LostArlec4"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
17:04:13 MQT: stat/Smart_Energy_Switch/STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
17:04:14 MQT: stat/Smart_Energy_Switch/STATUS5 = {"StatusNET":{"Hostname":"Smart_Energy_Switch-2802","IPAddress":"192.168.0.41","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.232","Mac":"24:62:AB:42:8A:F2","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:04:14 MQT: stat/Smart_Energy_Switch/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.245","MqttPort":1883,"MqttClientMask":"Smart_Energy_Switch","MqttClient":"Smart_Energy_Switch","MqttUser":"Smart_Energy_Switch","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:04:14 MQT: stat/Smart_Energy_Switch/STATUS7 = {"StatusTIM":{"UTC":"2020-09-08T07:04:14","Local":"2020-09-08T17:04:14","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+10:00","Sunrise":"15:17","Sunset":"04:17"}}
17:04:14 MQT: stat/Smart_Energy_Switch/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:04:14 MQT: stat/Smart_Energy_Switch/STATUS10 = {"StatusSNS":{"Time":"2020-09-08T17:04:14","ENERGY":{"TotalStartTime":"2020-08-27T00:43:26","Total":0.034,"Yesterday":0.000,"Today":0.034,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":243,"Current":0.000}}}
17:04:14 MQT: stat/Smart_Energy_Switch/STATUS11 = {"StatusSTS":{"Time":"2020-09-08T17:04:14","Uptime":"0T00:18:58","UptimeSec":1138,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Froogle","BSSId":"70:3A:CB:91:5A:94","Channel":6,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Always present on Main Screen - Energy Today, Energy Yesterday and Energy Total

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Energy Totals should be ~ 6 x the values displayed

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

vosjevos, 2020-09-08:
Similar misbehaviour experienced on GoSund SP111 smart plugs. Downgrade to 8.4.0 fixed it.

coolman111, 2020-09-08:
Similar misbehaviour experienced also with **Tasmota 8.5.0** on Gosund SP111 Hardware Version 1.1.
The energy values in kWh should be about 15x greater than displayed.
The other readings (Volt, Watt, Ampere, Factor=1.00) seem to be correct.
Everything had been calibrated before updating from Tasmota 8.2.0 to 8.5.0.
Downgrading to 8.4.0 solved the problem.
Theo, I hope this information could be helpful.

smithy2167, 2020-09-09:
Note that this was tested with power levels of 350W, 2200W and 2450W so is assumed not to be related to low power level issues.

arendst, 2020-09-09:
I just reverted the change #9160 locally and I see the counters start to work again. As the above mentionend devices use the same code I'm quite sure this is the cause of your issues too.

See also https://github.com/arendst/Tasmota/issues/9266#issuecomment-689400273

arendst, 2020-09-09:
Fix has been integrated in latest release v8.5.0 binary downloads

closed 2020-09-09 by ascillato2

coolman111, 2020-09-09:
Confirmation: I got the last release 8.5.0 Build 2020.09.09 11:47:14, tested and works 100% perfectly on Gosund SP111.
Even better than before 8.4. The Energy kWh display is only +1.3% different to my old CONRAD Energy Meter EKM265.
At 8.2.0 the difference was more than +10%. I did not make any change in calibration.
Congratulations, excellent fix.

FISO291, 2020-09-10:
Ich kann auch bestätigen dass mit der aktuellsten Firmware das Problem beseitigt ist. Getestet auf einer Gosund SP111. Werte werden korrekt berechnet :)

I can also confirm that with the latest firmware the problem is solved. Tested on a Gosund SP111. Values are calculated correctly :)