arendst/Tasmota #9264: Support of Zigbee Device "Mijia Light Intensity Sensor" GZCGQ01LM

SirUli, 2020-09-08:
### PROBLEM DESCRIPTION
The Zigbee Device "Mijia Light Intensity Sensor" GZCGQ01LM can be paired without any issues but then doesn't report luminance but some scrambled data instead. The weblog 4 output below comes when i press the button on the sensor. Over the period of 24 hours the output was somewhat similar.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos D1 mini + 
- [X] Tasmota binary firmware version number used: 8.4.0
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: PlatformIO
- [X] Flashing tools used: Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
08:28:43 CMD: Backlog Template; Module; GPIO 255
08:28:43 SRC: WebConsole from 192.168.100.10
08:28:43 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
08:28:43 SRC: Backlog
08:28:43 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
08:28:43 MQT: stat/tasmota_AC6F9A/RESULT = {"NAME":"Zigbee","GPIO":[255,255,255,255,255,255,255,255,255,166,255,165,255],"FLAG":15,"BASE":18}
08:28:43 SRC: Backlog
08:28:43 CMD: Group 0, Index 1, Command "MODULE", Data ""
08:28:43 MQT: stat/tasmota_AC6F9A/RESULT = {"Module":{"0":"Zigbee"}}
08:28:43 SRC: Backlog
08:28:43 CMD: Group 0, Index 1, Command "GPIO", Data "255"
08:28:43 MQT: stat/tasmota_AC6F9A/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"166":"Zigbee Rx"},"GPIO14":{"0":"None"},"GPIO15":{"165":"Zigbee Tx"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
08:28:55 CMD: Backlog Rule1; Rule2; Rule3
08:28:55 SRC: WebConsole from 192.168.100.10
08:28:55 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
08:28:55 SRC: Backlog
08:28:55 CMD: Group 0, Index 1, Command "RULE", Data ""
08:28:55 MQT: stat/tasmota_AC6F9A/RESULT = {"Rule1":"off","Once":"off","StopOnError":"off","Length":0,"Free":511,"Rules":""}
08:28:55 SRC: Backlog
08:28:55 CMD: Group 0, Index 2, Command "RULE", Data ""
08:28:55 MQT: stat/tasmota_AC6F9A/RESULT = {"Rule2":"off","Once":"off","StopOnError":"off","Length":0,"Free":511,"Rules":""}
08:28:55 SRC: Backlog
08:28:55 CMD: Group 0, Index 3, Command "RULE", Data ""
08:28:55 MQT: stat/tasmota_AC6F9A/RESULT = {"Rule3":"off","Once":"off","StopOnError":"off","Length":0,"Free":511,"Rules":""}
```
- [X] Provide the output of this command: `Status 0`:
```
08:28:58 CMD: Status 0
08:28:58 SRC: WebConsole from 192.168.1.10
08:28:58 CMD: Group 0, Index 1, Command "STATUS", Data "0"
08:28:58 MQT: stat/tasmota_AC6F9A/STATUS = {"Status":{"Module":0,"DeviceName":"tasmota1","FriendlyName":["TasmotaHome"],"Topic":"tasmota_AC6F9A","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:28:58 MQT: stat/tasmota_AC6F9A/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:39:43","StartupUTC":"2020-09-08T06:49:15","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2020-09-07T18:26:02","SaveCount":28,"SaveAddress":"F8000"}}
08:28:58 MQT: stat/tasmota_AC6F9A/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-09-06T21:36:57","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"394/699"}}
08:28:58 MQT: stat/tasmota_AC6F9A/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["KM.MM",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["04008009","2805C8000100060000005A00000000000000","00000000","00006082"]}}
08:28:58 MQT: stat/tasmota_AC6F9A/STATUS4 = {"StatusMEM":{"ProgramSize":677,"Free":324,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAE797","04768001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,20,21,22,23,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
08:28:58 MQT: stat/tasmota_AC6F9A/STATUS5 = {"StatusNET":{"Hostname":"tasmota1","IPAddress":"192.168.1.16","Gateway":"192.168.1.1","Subnetmask":"255.255.255.224","DNSServer":"192.168.1.2","Mac":"FC:F5:C4:AC:6F:9A","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:28:58 MQT: stat/tasmota_AC6F9A/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt-server","MqttPort":1883,"MqttClientMask":"tasmota1","MqttClient":"tasmota1","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:28:58 MQT: stat/tasmota_AC6F9A/STATUS7 = {"StatusTIM":{"UTC":"2020-09-08T07:28:58","Local":"2020-09-08T08:28:58","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:17","Sunset":"19:17"}}
08:28:58 MQT: stat/tasmota_AC6F9A/STATUS10 = {"StatusSNS":{"Time":"2020-09-08T08:28:58"}}
08:28:58 MQT: stat/tasmota_AC6F9A/STATUS11 = {"StatusSTS":{"Time":"2020-09-08T08:28:58","Uptime":"0T00:39:43","UptimeSec":2383,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"KM.MM","BSSId":"76:AC:B9:2E:34:B8","Channel":10,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:04"}}}
08:28:59 WIF: Checking connection...
```

- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_

```
08:15:06 CMD: weblog 4
08:15:06 MQT: stat/tasmota_AC6F9A/RESULT = {"WebLog":4}
08:15:06 CFG: Saved to flash at F8, Count 28, Bytes 4096
08:15:15 ZIG: {"ZbZNPReceived":"448100000300808E010B011F00A5D749000003110601808E1D"}
08:15:15 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":3,"srcaddr":"0x8E80","srcendpoint":1,"dstendpoint":11,"wasbroadcast":1,"LinkQuality":31,"securityuse":0,"seqnumber":0,"fc":"0x11","manuf":"0x0000","transact":6,"cmdid":"0x01","payload":""}}
08:15:15 ZIG: ZbZCLRawReceived: {"0x8E80":{"0003!01":""}}
08:15:15 MQT: tele/tasmota_AC6F9A/8E80/SENSOR = {"ZbReceived":{"XMLUX01":{"Device":"0x8E80","Name":"XMLUX01","0003!01":"","Endpoint":1,"LinkQuality":31}}}
08:15:15 ZIG: {"ZbZNPReceived":"448100000300808E0101011F00A5D749000003110601808E1D"}
08:15:15 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":3,"srcaddr":"0x8E80","srcendpoint":1,"dstendpoint":1,"wasbroadcast":1,"LinkQuality":31,"securityuse":0,"seqnumber":0,"fc":"0x11","manuf":"0x0000","transact":6,"cmdid":"0x01","payload":""}}
08:15:15 ZIG: ZbZCLRawReceived: {"0x8E80":{"0003!01":""}}
08:15:15 MQT: tele/tasmota_AC6F9A/8E80/SENSOR = {"ZbReceived":{"XMLUX01":{"Device":"0x8E80","Name":"XMLUX01","0003!01":"","Endpoint":1,"LinkQuality":31}}}
08:15:19 WIF: Checking connection...
08:15:39 WIF: Checking connection...
08:15:59 WIF: Checking connection...

```

### TO REPRODUCE

Connect [GZCGQ01LM](https://zigbee.blakadder.com/Xiaomi_GZCGQ01LM.html) to a Tasmota with ZigBee enabled.

### EXPECTED BEHAVIOUR

Luminance (and potentially battery stats) is available.

### SCREENSHOTS

None.

### ADDITIONAL CONTEXT

None.

s-hadinger, 2020-09-08:
Hi, what you receive as `"0003!01":""` is an Identify Query (I've never seen this before). Your device is broadcasting on group 0 asking for devices in Identification mode. I have no clue why.

To receive Illuminance readings, you need to bind the device to your coordinator:
`ZbBind {"Device":"0x8E80","Cluster":"Illuminance"}`

Then you might also need to configure attribute reporting:
`ZbSend {"Device":"0x8E80","Config":{"Illuminance":{"MinInterval":15,"MaxInterval":3600,"MaxInterval":}}}`

The documentation wasn't updated yet, details can be found here: https://github.com/arendst/Tasmota/pull/9050

Don't forget to wake-up the device before sending these commands. It's also possible that they work only for tens of seconds after pairing.

M4rt1nCh, 2020-09-09:
I'm having the same issue and tried to follow the guidance with this result:
```
20:56:26 CMD: ZbBind {"Device":"0x29C9","Cluster":"Illuminance"}
20:56:26 SRC: WebConsole from 192.168.100.12
20:56:26 CMD: Group 0, Index 1, Command "ZBBIND", Data "{"Device":"0x29C9","Cluster":"Illuminance"}"
20:56:26 MQT: stat/tasmota_AC6F9A/RESULT = {"ZbBind":"Missing "ToDevice" or "ToGroup“"}
```
When I'm adding "ToDevice:0x000" to the command, it is accepted.

For the second command, I'm getting this message here:
```
20:58:06 CMD: ZbSend {"Device":"0x29C9","Config":{"Illuminance":{"MinInterval":15,"MaxInterval":3600,"MaxInterval":}}}
20:58:06 SRC: WebConsole from 192.168.100.12
20:58:06 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x29C9","Config":{"Illuminance":{"MinInterval":15,"MaxInterval":3600,"MaxInterval":}}}"
20:58:06 ZIG: guessing endpoint 1
20:58:06 MQT: stat/tasmota_AC6F9A/RESULT = Missing zigbee 'Send', 'Write', 'Report' or ‚Response'
```

I've also ensured to wake up the device, repaired them just to be sure as well - no success. Any suggestions are appreciated :-) Thanks

s-hadinger, 2020-09-09:
Please upgrade to v8.5.

M4rt1nCh, 2020-09-13:
works great, thanks for your support 👍

SirUli, 2020-09-13:
Agree - it does. Thanks for the quick resolution!

closed 2020-09-13 by SirUli

jeroenst, 2021-06-12:
Is this illuminance value the actual lux measured or does the lux value have to be calculated?

M4rt1nCh, 2021-06-12:
No, it’s not the actual mix value. I’m using this formula to calculate it:

pow(10,$x/10000)-1

where $x is the actual value from the sensor.

jeroenst, 2021-06-12:
I actually tried this but the value is way too low after this calculation... 🤔