arendst/Tasmota #9265: Inaccurate timing in Buzzer output

hello-world-dot-c, 2020-09-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU v3
- [x] Tasmota binary firmware version number used: 8.5.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
10:36:05 MQT: stat/tasmota_EC462F/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
10:36:05 MQT: stat/tasmota_EC462F/RESULT = {"Module":{"18":"Generic"}}
10:36:05 MQT: stat/tasmota_EC462F/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"158":"LedLinki"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"160":"Buzzer"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
10:37:02 MQT: stat/tasmota_EC462F/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_EC462F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:37:02 MQT: stat/tasmota_EC462F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:02","StartupUTC":"2020-09-08T09:18:00","Sleep":50,"CfgHolder":4617,"BootCount":83,"BCResetTime":"2020-07-02T14:24:38","SaveCount":320,"SaveAddress":"FA000"}}
10:37:02 MQT: stat/tasmota_EC462F/STATUS2 = {"StatusFWR":{"Version":"8.5.0(tasmota)","BuildDateTime":"2020-09-07T10:35:04","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"354/699"}}
10:37:02 MQT: stat/tasmota_EC462F/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["monsternet",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00040000","00106000","00000000"]}}
10:37:02 MQT: stat/tasmota_EC462F/STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
10:37:02 MQT: stat/tasmota_EC462F/STATUS5 = {"StatusNET":{"Hostname":"tasmota_EC462F-1583","IPAddress":"192.168.1.173","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"50:02:91:EC:46:2F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:37:02 MQT: stat/tasmota_EC462F/STATUS6 = {"StatusMQT":{"MqttHost":"homeslave.fritz.box","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EC462F","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:37:02 MQT: stat/tasmota_EC462F/STATUS7 = {"StatusTIM":{"UTC":"2020-09-08T09:37:02","Local":"2020-09-08T10:37:02","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:17","Sunset":"19:17"}}
10:37:02 MQT: stat/tasmota_EC462F/STATUS10 = {"StatusSNS":{"Time":"2020-09-08T10:37:02"}}
10:37:02 MQT: stat/tasmota_EC462F/STATUS11 = {"StatusSTS":{"Time":"2020-09-08T10:37:02","Uptime":"0T00:19:02","UptimeSec":1142,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"monsternet","BSSId":"C8:0E:14:A1:36:94","Channel":4,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Set one GPIO to Buzzer(160).
Issue command "Buzzer -1".
Listen to buzzer (if connected) or measure the on/off pattern with an oscilloscope.
Get irritated because the beeps and pauses are not equally long.

### EXPECTED BEHAVIOUR
Get accurate timing for the on/off pattern according to the definition of the Buzzer command.

### SCREENSHOTS
Measuring the signal on the Buzzer pin after the Buzzer -1 command:

[image]

Even visually, it is very clear to see that the timing is "off". This is not a one-time effect, either but happens frequently.

### ADDITIONAL CONTEXT
If someone could explain how the Buzzer timing is generated and what could throw it off I could start looking into it.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-09-08:
Thanks for fixing this on PR #8994 

Very appreciated :+1:

hello-world-dot-c, 2020-09-09:
This is not fixed, I only mentioned it in https://github.com/arendst/Tasmota/pull/8994 because I noticed the problem while testing the new feature.

Please re-open, thanks!

hello-world-dot-c, 2020-09-09:
> 
> 
> Thanks for fixing this on PR #8994
> 
> Very appreciated 👍

This is not fixed, I only mentioned it in #8994 because I noticed the problem while testing the new feature.

Please re-open, thanks!

hello-world-dot-c, 2020-09-10:
May I ask you to replace label 'feature request' and re-add 'enhancement'? The feature is there but the output doesn't work as described. It is at least enhancement, in my opinion it is even a bug because it's not behaving according to spec.

Thank you.

stefanbode, 2020-09-10:
I already mentioned it in a push request. There is no guarantee that the 50ms loop will be executed all the time. misses are quite frequent. We have two options to overcome the issue. One is to set the start time and then decide on on/off based on the difference to the start time. This will avoid the error move during the complete beep cycle. A one-second beep will take 1 second+.50ms even if in the meantime some 50ms loops are missed. The other option is to move the ON/OFF mechanism into an RTC timer. I do not recommend but in this case, you can be sure the time will be accurate.

stefanbode, 2020-09-11:
I would suggest to make it on the buzzer module. should be no big deal. One Question: why is this important to you? as mentioned the on/off will always have some imperfections.

hello-world-dot-c, 2020-09-11:
When generating an audible alert using an on/off pattern for a longer period of time, the fluctuations are **very** noticeable, especially when the "on" phase is shortened. May be already an improvement if the phases can only become longer, not shorter. Which, btw. based on your explanation I still don't understand how that can happen unless there already is some "catching up" logic implemented that is however malfunctioning.

Where is the 50 ms loop located that you were talking about earlier?

stefanbode, 2020-09-12:
50ms is a Tasmota restriction on the shortest Intervall to call a function.  The buzzer is in this loop

stefanbode, 2020-09-12:
@arendst : I assume the only way we get rid of the audio problem is to shift the on/off control of the beeper GPIO into a 50ms RTC Timer. Nothing large. Also using the loop call will in my mind not help to get clear control on the on/off synchronization. Should I try to move this part? I would enable and disable the rtc timer with the start of the beeping sequence and cleanup at the end.

s-hadinger, 2020-09-12:
Another way would be to temporarily reduce the sleep value to 10 or below. This gives you a 100Hz tick and better resolution. I'm doing this trick for a smoother fading with lights.

arendst, 2020-09-12:
I prefer adjusting sleep over another interrupt especially for something trivial as a beeper...

hello-world-dot-c, 2020-10-12:
> Another way would be to temporarily reduce the sleep value to 10 or below. This gives you a 100Hz tick and better resolution. I'm doing this trick for a smoother fading with lights.

@s-hadinger: Could you perhaps give me the location where already you do this? I'll then open a PR and try to apply this to the buzzer logic.

s-hadinger, 2020-10-12:
Sleep is set here: https://github.com/arendst/Tasmota/blob/40ad4f7ed01411d9b38e7ff230840885cc27ce4b/tasmota/xdrv_04_light.ino#L1860-L1868

Although the comment is wrong:
https://github.com/arendst/Tasmota/blob/40ad4f7ed01411d9b38e7ff230840885cc27ce4b/tasmota/my_user_config.h#L265

The logic is applied in `LightApplyFade()`
https://github.com/arendst/Tasmota/blob/40ad4f7ed01411d9b38e7ff230840885cc27ce4b/tasmota/xdrv_04_light.ino#L2110-L2117

And triggered here: i.e. fading is not finished, we had at least 5ms since last time (see above) and the timer sets ticks every ~10ms (not guaranteed)
https://github.com/arendst/Tasmota/blob/40ad4f7ed01411d9b38e7ff230840885cc27ce4b/tasmota/xdrv_04_light.ino#L2148-L2149

closed 2020-10-13 by arendst

hello-world-dot-c, 2020-10-13:
@s-hadinger: Thanks, did that and accuracy is better. Not perfect, but as good as it gets with this mechanism.

grmpl, 2025-02-02:
If anybody is still struggling with inaccurate timing of buzzer, especially when running complex (blocking) berry code: On ESP32 you can use the ULP coprocessor for accurate buzzer timing.
I have a working example in my AlarmClock project [UlanziAlarmClockTasmotaBerry](https://github.com/grmpl/UlanziAlarmClockTasmotaBerry) (Currently in branch ULP-Buzzer).

PR #9530 (2020-10-13): Reduce sleep during buzzer cycles to improve on/off period accuracy.
## Description:

**Related issue (if applicable):** fixes #9265

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.3
  - [ ] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change m [... cut]