arendst/Tasmota #9266: After Upgrade 7.1.2 -> 8.5.0 Sonoff Pow is not counting the produced 'Energy Today' from PV

Sab3rRid3r, 2020-09-08:
### PROBLEM DESCRIPTION
I use Sonoff Pow for visualizing the produced energy per day 'Energy Today' which worked fine in Version 7.1.2
The produced 'Energy Today' is usually between 2-5 kWh from the PV. After the Upgrade to 8.5.0 it is about 0.005 kWh which could be the needed Power from the inverter.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff POW
- [X] Tasmota binary firmware version number used: 8.5.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
15:13:42 CMD: Backlog Template; Module; GPIO 255
15:13:42 MQT: /smarthome/garten/photovoltaik/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:13:43 MQT: /smarthome/garten/photovoltaik/stat/RESULT = {"Module":{"6":"Sonoff Pow"}}
15:13:43 MQT: /smarthome/garten/photovoltaik/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"130":"HLWBL SEL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"132":"HLWBL CF1"},"GPIO14":{"133":"HLW8012 CF"},"GPIO15":{"52":"Led1"},"GPIO16":{"0":"None"}}```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
15:15:39 MQT: /smarthome/garten/photovoltaik/stat/STATUS = {"Status":{"Module":6,"DeviceName":"Sonoff POW Garten Photovoltaik","FriendlyName":["Sonoff POW Garten Photovoltaik"],"Topic":"photovoltaik","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:15:39 MQT: /smarthome/garten/photovoltaik/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:39:20","StartupUTC":"2020-09-08T13:36:19","Sleep":50,"CfgHolder":4617,"BootCount":1004,"BCResetTime":"2020-09-07T22:41:22","SaveCount":3608,"SaveAddress":"F6000"}}
15:15:39 MQT: /smarthome/garten/photovoltaik/stat/STATUS2 = {"StatusFWR":{"Version":"8.5.0(tasmota)","BuildDateTime":"2020-09-07T10:35:04","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"482/699"}}
15:15:39 MQT: /smarthome/garten/photovoltaik/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Meins2,4","indebuurt2"],"TelePeriod":120,"Resolution":"558180C0","SetOption":["540A0009","2805C8000100060000005AFF000000000000","00000201","00006000","00000000"]}}
15:15:39 MQT: /smarthome/garten/photovoltaik/stat/STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
15:15:39 MQT: /smarthome/garten/photovoltaik/stat/STATUS5 = {"StatusNET":{"Hostname":"Sonoff POW Garten Photovoltaik","IPAddress":"192.168.178.82","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.36","Mac":"68:C6:3A:D5:C3:60","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
15:15:39 MQT: /smarthome/garten/photovoltaik/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.36","MqttPort":1883,"MqttClientMask":"garten_photovoltaik","MqttClient":"garten_photovoltaik","MqttUser":"mosquitto","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:15:39 MQT: /smarthome/garten/photovoltaik/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-09-08T14:15:39","Local":"2020-09-08T15:15:39","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:17","Sunset":"19:17"}}
15:15:39 MQT: /smarthome/garten/photovoltaik/stat/STATUS9 = {"StatusPTH":{"PowerDelta":[80,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:15:39 MQT: /smarthome/garten/photovoltaik/stat/STATUS10 = {"StatusSNS":{"Time":"2020-09-08T15:15:39","ENERGY":{"TotalStartTime":"2020-02-28T00:00:00","Total":555.660,"Yesterday":2.249,"Today":0.005,"Power":389,"ApparentPower":389,"ReactivePower":0,"Factor":1.00,"Voltage":225,"Current":1.733}}}
15:15:39 MQT: /smarthome/garten/photovoltaik/stat/STATUS11 = {"StatusSTS":{"Time":"2020-09-08T15:15:39","Uptime":"0T00:39:20","UptimeSec":2360,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","Wifi":{"AP":1,"SSId":"Meins2,4","BSSId":"08:96:D7:FA:2F:9D","Channel":6,"RSSI":100,"Signal":-17,"LinkCount":1,"Downtime":"0T00:00:03"}}}```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
15:16:28 MQT: /smarthome/garten/photovoltaik/tele/SENSOR = {"Time":"2020-09-08T15:16:28","ENERGY":{"TotalStartTime":"2020-02-28T00:00:00","Total":555.660,"Yesterday":2.249,"Today":0.005,"Period":0,"Power":191,"ApparentPower":191,"ReactivePower":0,"Factor":1.00,"Voltage":227,"Current":0.844}}```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
'Energy Today' counts the produced Energy like in Version 7.1.2

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Sab3rRid3r, 2020-09-08:
Version 7.1.2 - "Today":1.879
2020-09-07_16:30:11 SonoffPowSolarGarten energyJson: {"Time":"2020-09-07T15:30:10","ENERGY":{"TotalStartTime":"2020-02-28T00:00:00","Total":555.285,"Yesterday":1.712,"Today":1.879,"Power":190,"ApparentPower":190,"ReactivePower":0,"Factor":1.00,"Voltage":225,"Current":0.844}}

Version 8.5.0 - "Today":0.005
2020-09-08_16:30:28 SonoffPowSolarGarten energyJson: {"Time":"2020-09-08T15:30:28","ENERGY":{"TotalStartTime":"2020-02-28T00:00:00","Total":555.660,"Yesterday":2.249,"Today":0.005,"Period":0,"Power":373,"ApparentPower":373,"ReactivePower":0,"Factor":1.00,"Voltage":228,"Current":1.640}}

arendst, 2020-09-08:
I think it's related to #9160. I will investigate the impact for low values.

arendst, 2020-09-08:
If you çompile yourselves you might want to roolback the change of #9160 and see if this solves your issue.

Sab3rRid3r, 2020-09-08:
> If you çompile yourselves you might want to roolback the change of #9160 and see if this solves your issue.

Thank you! I successfully build a binary (tasmota.bin 609.520 kb) based on 8.5.0 sources with replaced changes from #9160 and uncommented 'tasmota' in platform.ini. That's it. It's the first time I try to flash a self compiled/build version so i'm a little nervous to do something wrong ;-)
Is there something obvious I may have forgot apart from a configuration backup?
Thanks in advance.

arendst, 2020-09-09:
You should be fine with this.

arendst, 2020-09-09:
Just did it myselves and indeed the daily energy counters starts to work again.

I will do some checks to see if I can get the old and new functionality working within the reseved counter space of 32 bits as the issue is caused by register overflow.

arendst, 2020-09-09:
Fix has been integrated in latest release v8.5.0 binary downloads

Sab3rRid3r, 2020-09-09:
Oh, it looks like i was too slow. I installed my modified Version an hour ago and since then it works again. So yes, i can confirm the bugfix.
Thank you very much!

closed 2020-09-09 by ascillato2