arendst/Tasmota #9269: No POWER command for MCP23017 relays

iz3man, 2020-09-08:
### PROBLEM DESCRIPTION
No POWER command for MCP23017 relays

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
18:57:30 MQT: stat/mcp/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
18:57:30 MQT: stat/mcp/RESULT = {"Module":{"18":"Generic"}}
18:57:30 MQT: stat/mcp/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"7":"WS2812"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
18:58:13 CMD: Status 0
18:58:13 MQT: stat/mcp/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"mcp","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:58:13 MQT: stat/mcp/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:01:47","StartupUTC":"2020-09-08T17:56:26","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2020-09-08T18:00:04","SaveCount":35,"SaveAddress":"F9000"}}
18:58:13 MQT: stat/mcp/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-09-08T18:56:03","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"303/699"}}
18:58:13 MQT: stat/mcp/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["izeman",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008008","2805C8000100060000005A00000000000000","00000000","00000000"]}}
18:58:13 MQT: stat/mcp/STATUS4 = {"StatusMEM":{"ProgramSize":584,"Free":416,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","043683A1","000000CD","010013F1","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,29"}}
18:58:13 MQT: stat/mcp/STATUS5 = {"StatusNET":{"Hostname":"mcp-4557","IPAddress":"192.168.1.59","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.11","Mac":"10:52:1C:02:D1:CD","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:58:13 MQT: stat/mcp/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_02D1CD","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:58:13 MQT: stat/mcp/STATUS7 = {"StatusTIM":{"UTC":"2020-09-08T17:58:13","Local":"2020-09-08T18:58:13","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:17","Sunset":"19:17"}}
18:58:13 MQT: stat/mcp/STATUS10 = {"StatusSNS":{"Time":"2020-09-08T18:58:13","MCP230XX":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":0,"D7":1,"D8":1,"D9":1,"D10":1,"D11":1,"D12":1,"D13":1,"D14":1,"D15":1}}}
18:58:13 MQT: stat/mcp/STATUS11 = {"StatusSTS":{"Time":"2020-09-08T18:58:13","Uptime":"0T00:01:47","UptimeSec":107,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":47,"Color":"482378","HSBColor":"266,71,47","Channel":[28,14,47],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"izeman","BSSId":"7A:8A:20:2A:02:73","Channel":6,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

```
#define USE_MCP230xx                 // Enable INPUT mode (pinmode 1 through 4)
#define USE_MCP230xx_OUTPUT          // Enable OUTPUT mode (pinmode 5)
#define USE_MCP230xx_DISPLAYOUTPUT   // Display state of OUTPUT pins on main Tasmota web interface
#define USE_MCP230xx_ADDR 0x20
```

in user_config_override.h, and I can control the 16 new digital pins fine on the console.

The pins are configured as OUTPUT INVERTED. The pin need to be pulled to GND when triggered. This works fine as well.

sensor29 0,6,1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

_Additionally all OUTPUT pins will be exposed as RELAYS and ordered behind the normal GPIO based RELAYS. Instead of the above sensor command you can also use the POWERxx command like for any RELAY. If you define INTERLOCK and/or INTERLOCK groups these will also take care about the out pins. The numbering of the RELAY's is following the standard tasmota behavior. Counting from D0 any defined OUT pin will add a new RELAY. Example: D0, D2, D3, D7 are out pins, then D0=POWER1, D2=POWER2, D3=POWER3 and D7=POWER4. Same behavior you can expect when defining PULSETIME for RELAYS._

So I'd expect to have RELAYS that can be controlled by POWER command.

POWER of the connected LEDstrip on D4 works fine:

`18:26:52 CMD: POWER 18:26:52 MQT: stat/mcp/RESULT = {"POWER":"OFF"} 18:26:52 MQT: stat/mcp/POWER = OFF 18:26:58 CMD: POWER ON 18:26:58 MQT: stat/mcp/RESULT = {"POWER":"ON"} 18:26:58 MQT: stat/mcp/POWER = ON`

But there are NO RELAYS defined that can be controlled by POWER.

`18:27:50 CMD: POWER2 ON 18:27:50 MQT: stat/mcp/RESULT = {"Command":"Error"}`

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-08:
As expected. Tasmota v.8.2.0 does not have this feature.
Changes in driver for the MCP230XX are in latest v.8.5

ascillato2, 2020-09-08:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-08 by ascillato2

iz3man, 2020-09-09:
Thanks guys. Also had to update PlatformIO to compile 8.5.0. Didn't notice that these awesome changes to MCP23017 were just made recently. ALL WORKING FINE NOW!!