arendst/Tasmota #9272: Can't set PULSETIME for Relay D8-D15 for MCP23017

iz3man, 2020-09-09:
### PROBLEM DESCRIPTION
Compiled my own version of 8.5.0 with MCP23017 support. Relays are shown fine, and work fine so far, but there are two problems:

I can't set PULSETIME for any RELAY# of 2nd register (D8-D15) 
and the main web interface won't reflect toggling state of the last two relays.

I guess that those two problems are somehow connected so opened one issue for both, the second issue is cosmetical only, so no "real" problem as such.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 min
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO_____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

08:59:19 CMD: Backlog Template; Module; GPIO 255
08:59:20 MQT: stat/mcp/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
08:59:20 MQT: stat/mcp/RESULT = {"Module":{"18":"Generic"}}
08:59:20 MQT: stat/mcp/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"7":"WS2812"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

09:00:17 CMD: status0
09:00:17 MQT: stat/mcp/STATUS = {"Status":{"Module":18,"DeviceName":"MCP","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4","Tasmota5","Tasmota6","Tasmota7","Tasmota8"],"Topic":"mcp","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

```

### TO REPRODUCE
a) Install 8.5.0, ESP with MCP23107, set all pins to output, toggle them from the main web interface
b) Set all pins to output, then try to set PULSETIME for relay #8 of the MCP (any of the 2nd register - GPB[0-7]), in my case this is relay9 as there is a LEDstrip connected to interal D4 of the Wemos.

```
09:02:43 CMD: pulsetime8
09:02:43 MQT: stat/mcp/RESULT = {"PulseTime8":{"Set":3,"Remaining":0}}
09:02:57 CMD: pulsetime9
09:02:57 MQT: stat/mcp/RESULT = {"Command":"Error"}
09:03:02 CMD: pulsetime9 3
09:03:02 MQT: stat/mcp/RESULT = {"Command":"Error"}
```

### EXPECTED BEHAVIOUR
a) toggling state is reflected 
b) pulsetime can be set

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

As you can see, ALL relays are turned OFF, but D13 and D14 are still shown as ON. This is purely cosmetical, as switching those works fine.

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-09-09:
Max number of pulsetimers is indeed 8.
https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota.h#L60
Changing this would take further changes to avoid breaking the structure where settings is stored, and break compatibility.

stefanbode, 2020-09-09:
This is one of the many limits the default tasmota software has. I'm pretty sure because of rare requirement and the efforts of storing the additional values we cannot change this easily, but I have a cheap idea:

Cant we use MOD MAX_PULSETIME to just copy the Pulstime from 1 to 9, 2 to 10 and so on. I assume this is only one line of code and there ARE side-effects that you cannot operate relay 1 and relay 9 at the same time. also you cannot have relay 1 with pulsetime and relay 9 without. so the idea is just using a side-effect to realize the usecase. I assume all you relays have the same pulsetime.. Please send feedback.

iz3man, 2020-09-09:
> Cant we use MOD MAX_PULSETIME
Indeed that would help. Or at least I'd need a way to release the RELAY after any given time (can really be the same for all relays). If I need to turn in on/off adds a lot of unwanted complexity on the home assistant side.
Another idea to solve this would be a way to turn on relay of if another one is turned on.
Use case here: I hard wire 5 remote controls for shutters to the outputs. Pulling 3 signals to GND makes open/stop/close trigger. Of course only one of them can be active. It won't destroy anything, but makes control impossible.

iz3man, 2020-09-09:
> Changing this would take further changes

What would happen if I changed it for a "personal" version? Could it work? I guess there is not need for me to update this device on a frequent base, so once it works I'll leave it that way.

sfromis, 2020-09-09:
You'd need to ensure that the settings block still has the same size, meaning that an expanded block for pulsetimers would need trimming something away elsewhere, like moving the pulsetimers to an area of free space. Changing settings in such a significant way would mean that you'd not be able to upgrade to or from binaries without this change.

iz3man, 2020-09-09:
> you'd not be able to upgrade to or from binaries without this change.

Ok. That would be no issue, as I have to do my own version because of MCP23017 anyway, as it's not compiled in any pre-compiled version.

stefanbode, 2020-09-11:
@iz3man : you have tested the change. can you please provide some feedback, if this fits for you?

iz3man, 2020-09-11:
@stefanbode : Of course I will. Just compiled yesterday, but didn't have time to test the setup. Will do this evening and report back. Thanks for your support!!!

iz3man, 2020-09-11:
@stefanbode : I can confirm that your code works fine. You first define the RELAYS/outputs and then can set pulsetime and it's set for 0&7, 1&8 etc ...

There's still and issue where the frontend won't reflect the state of the last two relays (they always stay on)

[image]

Don't know it this is related somehow ...

arendst, 2020-09-11:
OMG 16 buttons!

iz3man, 2020-09-11:
> OMG 16 buttons!

I can top that if you wish ;) I personally don't need/use the buttons, but they pop up when you define an OUTPUT.

stefanbode, 2020-09-12:
@arendst your great software can go up to 32. Then it's getting crowdy in the top 😂😂

stefanbode, 2020-09-12:
The issue with the last two relays ON might be a problem of the mcp driver. I haven't seen this on my device. The commands are always the same and I tested with 24 relays without issues.

iz3man, 2020-09-12:
@stefanbode : Would you mind sharing your config file so I can upload it to mine and see if it works here? Or I share mine whatever makes more sense

ascillato2, 2020-09-13:
Closing this issue as this feature has been added in the PR #9279 proposed for review. Thanks.

closed 2020-09-13 by ascillato2