arendst/Tasmota #9275: An automated way to get the latest version number available via OTA

agross, 2020-09-09:
**Have you looked for this feature in other issues and in the docs?**  

Yes, but a search query for "version" yields > 1000 open and closed issues.

**Is your feature request related to a problem? Please describe.**  

As I want to upgrade my 14 devices running Tasmota automatically via Ansible, I need a way to get the latest release version (e.g. 8.5.0.1 as of the time of writing).

**Describe the solution you'd like**  

Preferred would be a JSON file containing nothing but the version number, available via HTTP GET from http://ota.tasmota.com/tasmota/ and http://ota.tasmota.com/tasmota32/.

**Describe alternatives you've considered**  

Querying GitHub releases, but

* these contain an extra `v` in front of the version number (git tag)
* these are not always aligned with the latest release, i.e. as of the time of writing the latest GH release version is 8.5.0 whereas http://ota.tasmota.com/tasmota/ has 8.5.0.1 (as reported by the device after the upgrade).

**Additional context**  

My Ansible updater runs this for every Tasmota device in my apartment:

1. Determine latest release version ([code](https://github.com/agross/ansible-raspi/commit/2094507e2672baee1649f2472f3a5b7b02d5a739#diff-cd013ad32e80045fa84107cf5c1b164f))
1. Check release against device version (i.e. `Status 2`)
1. If the device version is greater or equal than the GH release version, do not run an upgrade
1. Otherwise, run the upgrade (i.e. `Upgrade 1`)

The upgrade process works, but with all my devices running 8.5.0.1 right now, I won't receive new updates until 8.6.0 shows up on GitHub releases and GH releases continue to list only minor versions increments. This is because I omit all devices that have `device version >= GH's latest release` (e.g. `8.5.0.1 >= 8.5.0`).

It this information available already, but I didn't find it in the documentation? Thanks!

s-hadinger, 2020-09-09:
First it all depends if you want only releases (8.5.0) or all the development versions - which adds up to multiple commits per day. 

It's not necessarily wise to update all your devices with each new commit in the development branch. We work hard to not break development versions but it can happen. 

Also subversions in the development branch are merely a convention, there is no specific packaging nor marking of subversions in Github.

agross, 2020-09-09:
OK, I was under the impression that versions listed on http://ota.tasmota.com/tasmota/ were not derived from CI builds. At least the default OTA URL pointed here, IIRC, which left me with a warm and fuzzy feeling towards these versions ;-) At least nothing broke for me!

Am I right to assume that http://ota.tasmota.com/tasmota/releases/ is aligned with GH releases?

ascillato, 2020-09-09:
Yes, exactly

http://ota.tasmota.com/tasmota/ is the latest development version

http://ota.tasmota.com/tasmota/releases/ is the latest stable release (obviously it won't contain all the day to day fixes added to development) 

8.5.0.1 means that you are in latest development version 
Latest stable release is 8.5.0

ascillato, 2020-09-09:
The version numbers are:

Latest stable release: https://github.com/arendst/Tasmota/blob/master/tasmota/tasmota_version.h

Latest development version: https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_version.h

sfromis, 2020-09-09:
For the release binaries, the "Firmware Version" is already shown on http://ota.tasmota.com/tasmota/release/ - for dev binaries, you have the timestamps which is when the current build is from.

If convenient, these JSON files have the version number in the beginning:
http://ota.tasmota.com/tasmota/release/release.php
http://ota.tasmota.com/tasmota/development.php

agross, 2020-09-10:
Thank you very much for the information, that was very helpful!

I changed my Ansible playbook to either read http://ota.tasmota.com/tasmota/release/release.php or http://ota.tasmota.com/tasmota/development.php which makes it convenient to switch between release or development builds.

All other actions are derived from these files, e.g. whether an upgrade is required and setting the OTA URL corresponding to the currently installed variant.

Thanks again! 👍

closed 2020-09-10 by agross