arendst/Tasmota #9276: Switch on RGB and White LEDs at same time with SM2135

ehallein, 2020-09-09:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  

I’ve got a Tasmota 8.5 flashed bulb https://templates.blakadder.com/mirabella_genio_I002608.html which uses SM2135 
Template {"NAME":"MiraBellaGenio","GPIO":[0,0,0,0,0,0,0,0,181,0,180,0,0],"FLAG":0,"BASE":18}

Everything is working fine, except I can’t get the RGB and Warm White LEDs to be all on at the same time.

This gives me the Red LED:
06:35:59 CMD: Color 255,0,0
06:35:59 MQT: stat/loungeroomdoor/RESULT = {"POWER":"ON","Dimmer":100,"Color":"255,0,0,0,0","HSBColor":"0,100,100","White":0,"CT":153,"Channel":[100,0,0,0,0]}

Running this straight after makes no change to the light (still red)
06:36:55 CMD: Color 255,0,0,255
06:36:55 MQT: stat/loungeroomdoor/RESULT = {"POWER":"ON","Dimmer":100,"Color":"255,0,0,255,0","HSBColor":"0,100,100","White":100,"CT":153,"Channel":[100,0,0,100,0]}

Then running this causes the light two switch to the white LED (I can see it switch off RED then switch on white)
06:37:32 CMD: Color 0,0,0,255
06:37:32 MQT: stat/loungeroomdoor/RESULT = {"POWER":"ON","Dimmer":100,"Color":"0,0,0,255,0","HSBColor":"0,0,0","White":100,"CT":153,"Channel":[0,0,0,100,0]}

Using SetOption105 to 1 or SetOption37 to 158 (RGB and White Split) makes no difference

It seems no matter what I try,I must have the RGB set to 0,0,0 in order to use the white LED or use White 100, which turns off the RGB.

**Describe the solution you'd like**  
To have RGB and White LEDs on at the same time so I can use white blend mode.

s-hadinger, 2020-09-10:
@arendst This is a limitation in the driver code. Both White channels are set to 0 if any of RGB channels is not zero.

It seems you have developed this driver, is it on purpose?

arendst, 2020-09-10:
It's on purpose simple because the device gets too hot when both rgb and white are powered on. This according to datasheet and original firmware.

So this is as designed and safekeeping.

ehallein, 2020-09-10:
Thanks I’ll close this then

closed 2020-09-10 by ehallein

coenski, 2020-10-09:
> It's on purpose simple because the device gets too hot when both rgb and white are powered on. This according to datasheet and original firmware.
> 
> So this is as designed and safekeeping.
@arendst 
How can i go around this restriction? i would like to use white blend/white split, not at full brightness .... tuning the lights according to temp, as i have done with my e27 bulbs.
I am using Action LSC GU10 Color bulbs/spots.