arendst/Tasmota #9277: PWM Dimmer SetOption to restore switchmode ability

jwoodard80, 2020-09-10:
**Have you looked for this feature in other issues and in the docs?**  
Yes, the PWM dimmer specifically disabled this ability.

**Is your feature request related to a problem? Please describe.**  
Previously I was able to use the dimmer power button in a switchmode 5/6 to give a longpress option. Now with the PWM dimmer module this option is not available. Everything else works a treat, but I was able to use the longpress option to turn on other lights across the room or turn out all bedroom lights in our child's room. 

**Describe the solution you'd like**  

**From the Docs:** When the power is on, holding the down or up button decreases/increases the brightness. The brightness can also be changed using just the power button. When the power is on, holding the power button alternately increases or decreases the brightness. Initially, holding the power button increases the brightness. Releasing and then holding the power button again decreases the brightness.

I would like this specific function to be disabled with a SetOption so as to allow the button to be set to another switchmode.

Personally, I feel like the function is redundant as, when the power is on, I can hold the up/down toggle to accomplish the same thing. Of course, that is just my opinion and it is offered very humbly. 

**Describe alternatives you've considered**  
Currently I'm running a forked version from DigiBlur to keep the functionality but I would very much like to just move to stock Tasmota rather than keep up with the fork. This problem is the only thing keeping me from changing. My wife has grown accustom to the feature and was quite displeased with me when it stopped functioning. :-(

**Additional context**  
I am happy to help in anyway I can VIA testing. Thank you for your consideration in this matter!

**(Please, remember to close the issue when the problem has been addressed)**

pcdiem, 2020-09-10:
I assume you used a rule to implement the long press functionality you had before. If you have a rule that handles the power button hold state, it should prevent PWM Dimmer from processing it.

jwoodard80, 2020-09-10:
I would have thought that as well but when I did the upgrade the rule was still there and it did not trigger. 

However, for verification, I will re-attempt everything tonight and report back.

jwoodard80, 2020-09-11:
I was able to get it to work.... I realized that it was now a button rather than a switch. 

For anyone else searching this, I did the following (Change publish topic to suit needs)

Backlog ButtonTopic 0; SetOption1 1; SetOption11 0; SetOption32 20  
Rule1  ON button1#state=3 DO publish cmnd/topicHOLD/power 2 ENDON
Rule1 1

Thank you for the response!

closed 2020-09-11 by jwoodard80

jwoodard80, 2020-09-11:
Since it is not clear in the docs that this option can be overridden, I feel like it would be a good idea to add this information to the PWM Dimmer information. 

Example: When the power is on, holding the power button alternately increases or decreases the brightness. Initially, holding the power button increases the brightness. Releasing and then holding the power button again decreases the brightness. **_This action can be overridden by rules if one wishes to have the long press do something different._** 

If anyone agrees I'm happy to edit it, however, I have never done it before so I'd have to learn.

pcdiem, 2020-09-11:
From https://tasmota.github.io/docs/PWM-dimmer-switch

SetOption32 defines the button hold time. When the PWM Dimmer module is initially selected, SetOption32 is set to 5 (1/2 second). Button presses and holds execute the normal ButtonTopic and Rule processing. If ButtonTopic is set and SetOption61 is 0 or a the button press/hold matches a rule, the button press/hold is ignored by PWM Dimmer.

jwoodard80, 2020-09-11:
Well, my sincere apologies, I stand corrected. I have no idea how I missed that. I bet I read that page 15 times and only now that you point it out does it slap me in the face. HA! 

Thank you for entertaining my ignorance with your kind help.