arendst/Tasmota #9292: when LedLink(i) is defined Wifi down is not signalled

tig987, 2020-09-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP8266 generic (ESP-12S)
- [x] Tasmota binary firmware version number used: 8.5.0
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Arduino
- [x] Flashing tools used: Arduino
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Configure GPIO2 as LedLink
When the connection to Wifi is up but MQTT server is down the LED starts blinking correctly.
But when the Wifi is not available the LED does not start blinking.
Also when booting with Wifi not available the LED does not start blinking.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The LedLink LED should start blinking when Wifi is down OR if MQTT is down.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

tig987, 2020-09-13:
I would not call this a feature request but a bug.
The documentation says:

> Link status LED shows the network state, more specifically the Wi-Fi and MQTT connection status.
> It blinks if the device is not connected to your Wi-Fi AP and MQTT broker (if MQTT is enabled).

For my understanding this means the Link status LED should blink if the device is not connected to the MQTT broker
or if it is not connected to the Wifi AP (and therefore also not connected to the MQTT broker).

But the LED only blinks when the Wifi connection is good but the device is not connected to the MQTT broker.
It doesn't blink when there is no connection to the Wifi. Neither when the Wifi is not available at boot nor when the Wifi connection fails later.

MulOnPomm, 2020-09-27:
I can confirm that the Wifi status led (1s blink interval) is not working in version 8.4 and after. It works in 8.2, 8.3 and 8.3.1. MQTT status led (2s blink interval) is working in all mentioned versions.

closed 2020-10-09 by ascillato2