arendst/Tasmota #9296: THERMOSTAT feature Negative numbers

elik745i, 2020-09-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I'm building smart fridge controller and require minimum 2 relays and 2 sensors to be controlled independently. So I have managed to include thermostat feature, compile new firmware and upload it into ESP12f module. I have managed to activate features and set temperatures, set multiple outputs, I'm using Celsius to measure temperature, however when trying to set temperature to negative degrees over console it does not work. Also I do not know how to assign two sensors so each of them would control their own relays (one is for fridge main compressor, second for defrosting heater control).

So far my user_config_override.h file looks like this:
``
`#ifndef USE_THERMOSTAT
#define USE_THERMOSTAT
#define THERMOSTAT_CONTROLLER_OUTPUTS         2           // Number of outputs to be controlled independently
#define THERMOSTAT_SENSOR_NAME                "DS18B20-1" // Name of the local sensor to be used
#define THERMOSTAT_RELAY_NUMBER               1         // Default output relay number
#endif

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [  ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [  ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
-  [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [  ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [  ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [  ] Device used (e.g., Sonoff Basic): _____
- [  ] Tasmota binary firmware version number used: _____
  - [  ] Pre-compiled
  - [  ] Self-compiled
    - [  ] IDE / Compiler used: _____Arduino IDE
- [  ] Flashing tools used: _____Arduino IDE
- [  ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"157":"LedLink"},"GPIO3":{"0":"None"},"GPIO4":{"4":"DS18x20"},"GPIO5":{"21":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"22":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

rule1 ON Power1#boot DO Backlog sensorinputset 1;controllermodeset 2;thermostatmodeset 1;climatemodeset 1;temptargetset %mem1% ENDON
```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

TrDA-hab, 2020-09-20:
@arijav 
I am also interested in this problem. You can help us, pliz!

arijav, 2020-09-21:
Understood, I will look into it this week and soon fix it into the driver

elik745i, 2020-09-21:
yes please add feature to be able to control negative temperatures for 2 and more separate relays independently. Thanks for help, will be waiting for updated feature!

TrDA-hab, 2020-09-21:
I'm also really looking forward to adding this new functionality!

arijav, 2020-09-21:
The negative temperatures should be something I can fix fast. The topic of the sensors is more tricky, because it is linked to how the sensors are handled out of the driver and it might be a big general rework. I could give you however some ideas to implement into the code locally. If the sensors are of the same kind, f.i. DS18B20, they might be reported as a single sensor giving periodically different temperature values. If this is the case (need to check) I see no fix w/o radical changes and therefore on my driver I could not offer a solution.

TrDA-hab, 2020-09-21:
@arijav I think we can all wait to add some good new functionality. And thanks for your great work!

arijav, 2020-09-22:
I have today no way to test, need to check on the weekend. I have implemented a possible fix for the negative temperatures. Could somebody test if it works for you? -> https://github.com/arijav/Tasmota/tree/new_branch_dev_3
You have to set the temperature frost limit to a negative value

elik745i, 2020-09-23:
ok thanks I will test now!

elik745i, 2020-09-23:
Nop does not work, setting temperature crashed at all:
[image]

arijav, 2020-09-23:
Ok, thanks for the feedback I will have to test myself. Will build up the setup hopefully within this week

elik745i, 2020-09-23:
> Ok, thanks for the feedback I will have to test myself. Will build up the setup hopefully within this week

Holdon looks like I have made mistake building thermostat custom firmware... maybe it does work, give me half an hour...

elik745i, 2020-09-23:
Sorry my mistake, copied wrong user_override file (blank one). Negative numbers working now:
[image]

Have you managed to implement 2 separate channels of control yet? I would integrate controller into fridge for testing. 
I have 2 DS18B20 sensors hookedup to the ESP and two relays. I need to be able independantly set temperatures for each relay so each of them listen to their own sensor and act based on temptargetset

elik745i, 2020-09-23:
I think need to implement new command set to be able to assign each relay to it's sensor name? then when for instance I send command like settargettemp1 assigned to relay1 sensor will act accordingly, and if I send command settargettemp2 relay2 and so on. just an idea...
Thanks again for your help bud!

elik745i, 2020-09-23:
settargettem1 and settargettem2 works as I defined in user_config_override file:

#define THERMOSTAT_CONTROLLER_OUTPUTS         2        // Number of outputs to be controlled independently

however I thing they both outputs (relays) listen to sensor1, not sure, still testing now

arijav, 2020-09-23:
I am happy to hear that the negative temperatures work. The topic of the sensors is still not in, I am evaluating the most consistent way to implement it. Hope to solve it soon

elik745i, 2020-09-23:
Thanks for your support bud!

arijav, 2020-09-23:
I have just added a possible fix for the temp sensor. The probability that it works is low since I've changed a couple of things and not tested. Could you give it a try anyway?. You have to modify in my_user_config.h THERMOSTAT_SENSOR_NUMBER to the number of sensors you used. It will just work if all sensor have the same name but differ on the last 2 characters -> "_ + sensor number" (f.i. DS18B20_1, DS18B20_2). I would appreciate if you test it as well with just one sensor and therefore THERMOSTAT_SENSOR_NUMBER to 1

elik745i, 2020-09-23:
> I have just added a possible fix for the temp sensor. The probability that it works is low since I've changed a couple of things and not tested. Could you give it a try anyway?. You have to modify [...]

Thanks! I have to integrate device into the fridge first, currently it is on prototyping board on my desk at home, It is about 120km far from me now, I'm on remote location till next Tuesday. I can only connect to home workstation through remote desktop connection for manipulation. Once I'm back home will certainly give it a try and get back to you!

arijav, 2020-09-23:
Great. I will try in the meantime to test by myself if the changes at least do not affect working with one sensor. I will not be able to test more than one unfortunately because this is what I have, so I would need your feedback to check if it really works.

I did not fully get the topic of independently control 2 or more thermostats on one device. This feature is existing, you just have to adapt the THERMOSTAT_CONTROLLER_OUTPUTS and configure each Thermostat via MQTT using CommandN, being N the number of the thermostat you want to configure and Command the command you would like to set.

elik745i, 2020-09-23:
> Great. I will try in the meantime to test by myself if the changes at least do not affect working with one sensor. I will not be able to test more than one unfortunately because this is what I have, [...]

Logic is: Two sensors set up in the fridge: Sensor 1 in the colling box, where all the food stored (idealy 2 sensors as most of the fridges has two compartments deep frost and cool box) another themperature sensor is installed on the radiator behind the fridge, from where cold is transferred into the fridge itself by set of fans. The purpose of the temp sensor on the radiator is - when temperature below certain degree on it, for each radiator is different as shapes and forms are different, it means it is completely blocked with formed ice and need to be defrosen. Controller reads let say -18*C from radiator's sensor and stops main compressor (Relay 1) and starts relay 2, which is hooked to the heater. Heater warms up radiator, ice melts down and when radiators temperature raises, lets say above 5*C it means all ice gone and system is ready for next cycle of cooling. Relay 2 switches off and relay 1 switches on and listens to sensor 1 and set temperature 1, cools fridges compartment 1 and 2. and switches off the compressor.
Sorry for my bad English maybe I wrote the whole assay not really sound, however cutting long story short - currently I do not know how to set up such logics on the controller without using MQTT requests as MQTT is not reliable (communication with server could be lost e.t.c.) I need each relay, yes I can set 2 or more relays as you said, to be controlled by each sensor. Relay 1 with sensor1, relay2 with sensor2 e.t.c....

ascillato2, 2020-12-07:
@arijav 

Hi, any news on this?

arijav, 2020-12-08:
Yes, the negative temperature issue is corrected in my branch, I will merge it soon. I have been quite busy with some topics of Opensprinkler, but I am back. The description of the wished functionality from elik745i is too specific for an application, I do not foresee to change the driver for the time being to address this, other than negative temperatures

closed 2020-12-09 by ascillato2