arendst/Tasmota #9301: Setting Mem2 to value 0 in a rule does not work

tig987, 2020-09-13:
### PROBLEM DESCRIPTION
When setting variable mem2 to value 0 in a rule and accessing it afterwards mem2 is infact not 0 but empty.
The same procedure is working fine with variable mem1.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP8266 generic (ESP-12s)
- [ ] Tasmota binary firmware version number used: 8.5.0
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Arduino
- [x] Flashing tools used: Arduino
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rule2
ON Time#Minute|5 DO mem1 0 ENDON
ON Time#Minute|5 DO mem2 0 ENDON

Rule1
ON tele-counter#c1 DO mem1=%mem1%+%value% ENDON
ON tele-counter#c2 DO mem2=%mem2%+%value% ENDON

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  11:40:00 RUL: TIME#MINUTE|5 performs "mem1 0"
11:40:00 MQT: sensors/cellar/stat/RESULT = {"Mem1":"0"}
11:40:00 RUL: TIME#MINUTE|5 performs "mem2 0"
11:40:00 MQT: sensors/cellar/stat/RESULT = {"Mem2":"0"}
11:40:33 RUL: TELE-COUNTER#C1 performs "mem1=0+11"
11:40:33 MQT: sensors/cellar/stat/RESULT = {"Mem1":"11.000"}
11:40:33 RUL: TELE-COUNTER#C2 performs "mem2=+21"
11:40:33 MQT: sensors/cellar/stat/RESULT = {"Mem2":"0.000"}

```

### TO REPRODUCE
set the rules as shown above.
you can then see that mem1=%mem1%+%value%
is calulated correcty as "mem1=0+11"
but mem2 calcuation mem2=%mem2%+%value%
is wrong; "mem2=+21"
Where the value 0 should be it is empty.

The same happens when using the rule syntax
```
ON Time#Minute|5 DO mem1=0 ENDON
ON Time#Minute|5 DO mem2=0 ENDON

```
then the output is 

> 11:30:00 RUL: TIME#MINUTE|5 performs "mem1=0"
> 11:30:00 MQT: sensors/cellar/stat/RESULT = {"Mem1":"0.000"}
> 11:30:00 RUL: TIME#MINUTE|5 performs "mem2=0"
> 11:30:00 MQT: sensors/cellar/stat/RESULT = {"Mem2":"0.000"}
> 11:30:33 RUL: TELE-COUNTER#C1 performs "mem1=0.000+11"
> 11:30:33 MQT: sensors/cellar/stat/RESULT = {"Mem1":"11.000"}
> 11:30:33 RUL: TELE-COUNTER#C2 performs "mem2=.000+21"
> 11:30:34 MQT: sensors/cellar/stat/RESULT = {"Mem2":"0.000"}

mem2 is caculated as "mem2=.000+21" - the digit before the . is missing

### EXPECTED BEHAVIOUR
set mem2 correctly to 0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-09-18:
Please, can you share your `status 0` output?

tig987, 2020-09-18:
The status 0 output is:
`22:09:02 MQT: sitea/sensors/cellar/stat/STATUS4 = {"StatusMEM":{"ProgramSize":584,"Free":352,"Heap":21,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F8027D7","00100001","00000405","000000C0","40000800","00000000","00000000"],"Drivers":"1,2,3,4,8,9,10,20,21,29","Sensors":"1,2,3,9"}}
22:09:02 MQT: sitea/sensors/cellar/stat/STATUS5 = {"StatusNET":{"Hostname":"xyz-4380","IPAddress":"192.168.178.253","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"A0:20:A6:31:B1:1C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:09:02 MQT: sitea/sensors/cellar/stat/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.xyz.xyz","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_31B11C","MqttUser":"xyz","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:09:02 MQT: sitea/sensors/cellar/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-09-18T20:09:02","Local":"2020-09-18T22:09:02","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:31","Sunset":"19:56"}}
22:09:02 MQT: sitea/sensors/cellar/stat/STATUS10 = {"StatusSNS":{"Time":"2020-09-18T22:09:02+02:00","COUNTER":{"C1":2,"C2":0},"BME280":{"Temperature":21.1,"Humidity":48.7,"DewPoint":9.9,"Pressure":970.7},"PressureUnit":"hPa","TempUnit":"C"}}
22:09:02 MQT: sitea/sensors/cellar/stat/STATUS11 = {"StatusSTS":{"Time":"2020-09-18T22:09:02+02:00","Uptime":"0T03:00:40","UptimeSec":10840,"Heap":21,"SleepMode":"Dynamic","Sleep":20,"LoadAvg":49,"MqttCount":2,"Wifi":{"AP":1,"SSId":"xyz","BSSId":"AB:AB:AB:BC:BC:BC","Channel":1,"RSSI":34,"Signal":-83,"LinkCount":2,"Downtime":"0T00:01:55"}}}
22:09:23 MQT: sitea/sensors/cellar/tele/STATE = {"Time":"2020-09-18T22:09:23+02:00","Uptime":"0T03:01:01","UptimeSec":10861,"Heap":21,"SleepMode":"Dynamic","Sleep":20,"LoadAvg":50,"MqttCount":2,"Wifi":{"AP":1,"SSId":"xyz","BSSId":"AB:AB:AB:BC:BC:BC","Channel":1,"RSSI":32,"Signal":-84,"LinkCount":2,"Downtime":"0T00:01:55"}}
22:09:24 MQT: sitea/sensors/cellar/tele/SENSOR = {"Time":"2020-09-18T22:09:23+02:00","COUNTER":{"C1":3,"C2":0},"BME280":{"Temperature":21.1,"Humidity":48.7,"DewPoint":9.9,"Pressure":970.7},"PressureUnit":"hPa","TempUnit":"C"}
`

arendst, 2020-10-26:
This was a serious error leading to character pool corruption

Thx for reporting and sorry it took so long to solve.

closed 2020-10-26 by ascillato2

tig987, 2020-10-26:
my pleasure - glad I could help.