arendst/Tasmota #9305: status 0 no longer outputs STATUS, STATUS1, STATUS2 in 8.5.0 when compiled with USE_MQTT_TLS

Flo-Kra, 2020-09-14:
### PROBLEM DESCRIPTION
When using a self-compiled version of Tasmota 8.5.0 with option USE_MQTT_TLS enabled, command status 0 does not output information STATUS, STATUS1, STATUS2 and on some devices also STATUS3. 
This breaks compatibility with TasmoAdmin as it relies on that command. 
When compiled without that USE_MQTT_TLS option and when using the precombiled binaries the behavior is normal. 
I recently used a USE_MQTT_TLS enabled version of 8.1.0 and didn´t experience that issue with it. 
Compiling was done using VSCode + PlatformIO, all updates installed as of 14.09.2020. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff S20
- [x] Tasmota binary firmware version number used: 8.5.0
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VSCode 1.49.0 / PlatformIO 5.0.1
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
13:47:06 CMD: Backlog Template; Module; GPIO 255
13:47:06 MQT: stat/Tasmota10/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
13:47:06 MQT: stat/Tasmota10/RESULT = {"Module":{"8":"Sonoff S2X"}}
13:47:07 MQT: stat/Tasmota10/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
no rules used

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
13:52:47 MQT: stat/Tasmota10/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Kinf","Kconf"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["5402200B","2805C8000100060000005AFF000000000000","00000601","00206000","00000000"]}}
13:52:47 MQT: stat/Tasmota10/STATUS4 = {"StatusMEM":{"ProgramSize":629,"Free":372,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE7D7","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
13:52:47 MQT: stat/Tasmota10/STATUS5 = {"StatusNET":{"Hostname":"sdo-t5vz-stlamp","IPAddress":"10.1.5.38","Gateway":"10.1.5.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.11","Mac":"5C:CF:7F:F9:17:7F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:52:47 MQT: stat/Tasmota10/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.local.lan","MqttPort":8883,"MqttClientMask":"Tasmota10-SonS20","MqttClient":"Tasmota10-SonS20","MqttUser":"tasmota4","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:52:47 MQT: stat/Tasmota10/STATUS7 = {"StatusTIM":{"UTC":"2020-09-14T11:52:47","Local":"2020-09-14T13:52:47","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:26","Sunset":"20:04"}}
13:52:47 MQT: stat/Tasmota10/STATUS10 = {"StatusSNS":{"Time":"2020-09-14T13:52:47"}}
13:52:47 MQT: stat/Tasmota10/STATUS11 = {"StatusSTS":{"Time":"2020-09-14T13:52:47","Uptime":"0T00:00:21","UptimeSec":21,"Heap":21,"SleepMode":"Normal","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"Kinf","BSSId":"F2:9F:C2:FA:BA:F5","Channel":13,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
shows no difference to weblog 2 mode.

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash Tasmota 8.5.0 compiled with USE_MQTT_TLS and check output of "status 0" on the console. 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-09-14:
As designed.

``Status 0`` still works as expected BUT it doesn't work for http requests like used by TasmoAdmin simply because the result buffer is too small due to lack of resources when TLS is enabled.

As can be seen in file tasmota_globals.h:
```
#ifdef USE_MQTT_TLS
  const uint16_t WEB_LOG_SIZE = 2000;          // Max number of characters in weblog
#else
  const uint16_t WEB_LOG_SIZE = 4000;          // Max number of characters in weblog
#endif
```
you might want to change this and see if you're still able to run your TLS device with less free memory.

Flo-Kra, 2020-09-14:
Thank you for the quick response and clarification! 
You´re right - that was causing it. I now changed the buffer size to 4000 also for TLS and it works as I expected now. 
Free memory now is 21kB - will see if it runs stable for a longer time.

closed 2020-09-14 by Flo-Kra

arendst, 2020-09-14:
Thx. Keep me posted regarding stability. I'm glad free mem is still 21kB. I might change buffersize to 4000  for TLS if stable.

Flo-Kra, 2020-09-21:
No issues so far with ~30 devices on 8.5.0 with MQTT TLS and rules with
expressions enabled, and web buffersize set to 4000. So after 7 days of
testing I consider it to be safe to increase the buffersize. Didn´t see
less than 18kb of free mem, no unintended reboot (all devices were up for
at least 5 days until they power cycled because my RCD triggered because of
another reason :-) ) and no unintended MQTT disconnection occured.
2 more devices are running the IR firmware, also compiled with MQTT TLS and
increased buffersize - same here, no issues.
best regards,
Florian

arendst, 2020-09-22:
Thx! I will change the default to 4000 today.