arendst/Tasmota #9306: tasmota32.bin @ ESP32 NodeMCU doesn't run

GrendelWWU, 2020-09-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I tried to run tasmota32.bin on an ESP32. After flashing successfully, I'm retrieving errors.

First, I erased the flash mem:
_gerbil@RedHeat:~$ esptool.py --port /dev/ttyUSB0 --chip esp32 erase_flash
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting......
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:97:1e:e4
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.6s
Hard resetting via RTS pin..._

Second I flashed tha tasmota32.bin:
_gerbil@RedHeat:~$ esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x1000 /home/gerbil/Downloads/tasmota32.bin 
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:97:1e:e4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1073312 bytes to 631673...
Wrote 1073312 bytes (631673 compressed) at 0x00001000 in 55.9 seconds (effective 153.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin..._

Then, one LED (TX) is flashing. On the serial console I'm getting these error messages again and again:

_rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3f400020,len:254204
ets Jun  8 2016 00:22:57_

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [-] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [-] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [-] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [-] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [-] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] asking Google
- [x] Device used (e.g., Sonoff Basic): ESP32 NodeMCU Module from AZ Delivery
- [x] Tasmota binary firmware version number used: 8.50
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: esptool.py @ Ubuntu 20.04
- [-] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [-] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [-] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3f400020,len:254204
ets Jun  8 2016 00:22:57

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

described above

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expected, that I could connect with my smartphone to an access point tasmota-xyz123 (or similar) and then, configure it...
(like I do with my Wemos modules with 8266)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-09-14:
Read the docs. Follow the correct flashing procedure: https://tasmota.github.io/docs/ESP32

GrendelWWU, 2020-09-14:
Hello.
Ok, Google didn't point me to that link. :-/
Thanks a lot. I got some overlapping messages, when following this procedure, cause first I took the files from
https://github.com/arendst/Tasmota/blob/firmware/firmware/tasmota32/ESP32_needed_files
But I think, these files are for developing purposes and too large. So I got smaller files from here and there....not the latest versions, but small enough not to get these overlapping memory messages:
_-rw-r--r-- 1 gerbil gerbil  8192 Sep 14 20:43 ba0.bin
-rw-r--r-- 1 gerbil gerbil 15872 Sep 14 20:40 bl.bin
-rw-rw-r-- 1 gerbil gerbil  3072 Sep 14 20:47 part.bin_

Et voilà... it works.

Maybe it will be a good idea, to put the link (docs/ESP32) to the releases-page and add a link with matching additional files.
Best regards,
Thomas

sfromis, 2020-09-14:
The files on the Github link are *exacly* the size you found elsewhere. If you got "too large" files, you probably did not ensure a binary download, but got html files downloaded instead, which indeed fails badly.

ascillato2, 2020-09-14:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-14 by ascillato2

danielw1990, 2023-12-03:
https://github.com/arendst/Tasmota/blob/firmware/firmware/tasmota32/ESP32_needed_files
is 404 now and
https://tasmota.github.io/docs/ESP32

Does not mention anything I could connect to "ESP32D0WDQ6" (anymore?)

So what was the solution? I have tried holding down the only button before connecting or at the start of the flashing.