arendst/Tasmota #9309: WS2812 Eco

Ilocans, 2020-09-14:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
Hi,

Thank you for the great work you've done. I have a question regarding ledstrip.
I am a beginner but I was able to use a WS2812B ledstrip, as described in the documentation and it works great. 
I also have WS2812B Eco and because it has WS2812B in its name, I was hoping it will work also. Maybe, I am missing something but it does not seem to be compatible. The leds turn on but there are all white, I cannot turn them off, change color,....

My configuration is simple, I am using a Wemos D1 and the led strip.

 I was wondering if there anything that could be done to make it work ?

Thanks for your help

**Describe the solution you'd like**  
I would like to be able to use my ws2812b Eco ledstrip with tasmota.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
N/A
**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

TrDA-hab, 2020-09-17:
According to this document, the difference is only in the construction / materials of the LED.
https://ae01.alicdn.com/kf/HTB1NfDLPjTpK1RjSZKPq6y3UpXao.jpg

ECO is an economical version, cheaper, with less no-load power consumption, slightly lower voltage drop during long runs, better suited for small battery-powered projects.

Judging by the datasheet WS2812 and WS2812ECO, they are fully compatible. I could not find any differences in LED control.

WS2812ECO:
https://datasheet.lcsc.com/szlcsc/1811151230_Worldsemi-WS2812E_C139127.pdf
WS2812:
https://datasheet.lcsc.com/szlcsc/2006151006_Worldsemi-WS2812B-B_C114586.pdf

Compatibility is also listed on the manufacturer's page:
http://www.btf-lighting.cn/ProductCenter/info_itemid_128.html
https://aliexpress.ru/item/2036819167.html

Ilocans, 2020-09-18:
Yes I was expecting it to work with no problem.
I tested the ECO version with my arduino and the fastled library just to check if it has any problem but it works fine.

To check my configuration, I test my WS2812 with tasmota and it works. I replace my WS2812 by my WS2812ECO : the leds turn on but there are all white, I cannot turn them off, change color,....

ascillato, 2020-09-18:
> the leds turn on but there are all white, I cannot turn them off, change color,....

May be, it is a damaged led strip. Sorry.

Ilocans, 2020-09-18:
Why did you closed.
The WS2812ECO ledstrip is working with my arduino but not with tasmota so it is not damaged.

TrDA-hab, 2020-09-19:
I see we are back to this question:
1.Does your arduino use 5v or 3.3v?
2. Do you have a wiring diagram for Arduino and Wemos?
3. There may be an error when connecting, perhaps there are nuances not specified in the datasheet for this new equipment. Real testing is needed to understand compatibility.
4. Tasmota has not announced support for WS2812ECO, you must understand that!

Ilocans, 2020-09-19:
Hi,

1 I was using 5V with my arduino
2. I don't have a siagram but it is simple and almost the same for both: Arduino/Wemos and also a capacitor of 1000uF connected to 5V . WS2812ECO/WS2812 connected to 5V. Ledstrip data pin connected directly to a GPIO4/GPIO7 (WEMOS/Arduino)

I understand that WS2812ECO support is not announced, if I can do anything to help to make it work.

Jason2866, 2020-09-20:
Have you tried a sample arduino sketch on the wemos mini?
Does this work?

Ilocans, 2020-09-20:
Good question.
Actually, I did not try with the wemos mini. I tested this morning the blinking example and it is not working. So I guess the problem comes from the Fastled library ?
I created an issue on their Github.

Jason2866, 2020-09-20:
Tasmota does use https://github.com/Makuna/NeoPixelBus

TrDA-hab, 2020-09-20:
> Good question.
> Actually, I did not try with the wemos mini.

Please try this, we need more information!

Ilocans, 2020-09-21:
> > Good question.
> > Actually, I did not try with the wemos mini.
> 
> Please try this, we need more information!
I thought I read the library used was FastLed, my mistake.

So I tested an example sketch (NeoPixeTest) of NeoPixelBus on my wemos.  Unfortunately,  it does not seem to work the led does not turn on at all.

ascillato, 2020-09-21:
Have you tried a different gpio?

If that doesn't work, please try a level shifter to 5v.

ascillato2, 2020-09-22:
Hi, any news on this?

Ilocans, 2020-09-23:
According the comments in the example, 
>  For Esp8266, the Pin is omitted and it uses GPIO3 due to DMA hardware use

So it should only work with GPIO3. I tried any with GPIO3 and another, just to be sure. The LEdStrip turn on, full white but no color or turn off as programmed.

After more reading I found there are several method that can be used with the constructor of the NeoPixel Object. I tried several but none of them seems working with my led strip.

> If that doesn't work, please try a level shifter to 5v.

I don't have a level shifter but my Wemos is connected to my PC through usb and my ledstrip is connected to 5V Power Supply so that should not be a problem.

ascillato, 2020-09-23:
May be your strip requires that the DATA line to be managed using 5V instead of the 3.3v that the ESP8266 can provide. So, may be a logic level shifter from 3.3v to 5v is necessary in order to make this led strip to work. May be it does not support 3.3v on the data line.

ESP8266 GPIO 3.3v ---> LEVEL SHIFTER 3.3v to 5V ---> DATA LINE

Ilocans, 2020-09-24:
Unfortunately, I don't have a Level Shifter, so it is not something I can test, at least for now. I could buy one on ebay but it is not sure it will sole the problem and it will take a few weeks before I get one.
The WS2812B does not require a Level Shifter for the Data Line so In my opinion, it should be the same for the Eco version. I was wondering if there other alternative we could explore ?

ascillato, 2020-09-24:
At hardware level for the serial communication, the only difference between your arduino board and the esp8266 is that the esp8266 transmits in 3.3v and the arduino in 5v. So, that seems to be the root of the problem, but a confirmation is this is needed. Is this is the problem, there is nothing we can do from the software side.

Ilocans, 2020-09-25:
I found a not too expensive level shifter on amazon. I should receive it in a few days.

ascillato, 2020-09-25:
Those are great news! Thanks.

arendst, 2020-09-25:
Or simply use a resistor of 470 or 330 ohm between esp8266 GPIO and WS2812Eco data pin.

Ilocans, 2020-09-25:
I tried another Library Adafruit_NeoPixe just for fun.
I tested the simple example they provide. It works, so I guess the level shifter is not the problem.

ascillato, 2020-09-25:
You tested that in the esp8266?

Ilocans, 2020-09-25:
> You tested that in the esp8266?

Yes and it works well.

Ilocans, 2020-09-25:
Guess no need to test with level shifter anymore

Ilocans, 2020-09-29:
Hello,
So what's the next step ? Is there anything else I can do to help ?

Jason2866, 2020-10-02:
You can try to define a different model type in my_user_config.h
https://github.com/arendst/Tasmota/blob/development/tasmota/my_user_config.h#L478

Jason2866, 2020-10-11:
Any news? Tried latest development version? The library for WSxxxx support has been updated.

Ilocans, 2020-10-13:
Hi I tried to update the firmware with no change. Is there any special binary I could use for the latest develpment version ?

Ilocans, 2020-10-21:
Hi ! Good news,
After receiving my bc560c transistor for testing like describe in [Tasmota docs]( https://tasmota.github.io/docs/WS2812B-and-WS2813/), it seems to work with this configuration. 
Also bc560c does not seems necessary to work. I made it work using only a 1k resistor  between esp8266 GPIO and WS2812Eco data pin. It did not work with  470 or 330 ohm like suggested above.

ascillato2, 2020-10-21:
Great News!!!

Thanks for sharing this information.

Please, can you add it to the docs? Thanks!!!!!

closed 2020-10-21 by ascillato2