arendst/Tasmota #9310: Shutter with stepper motor calibration based on steps (not time)

Kepete, 2020-09-14:
**Have you looked for this feature in other issues and in the docs?**  
Yes.
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No. 
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Currently shutter calibration is done based on time. According to the [calibration doc](https://tasmota.github.io/docs/Blinds-and-Shutters/#calibration), you need to set the open and close durations in seconds. Also for accuracy set the 50% mark and even use the Increasing Calibration Granularity possibility. 

This solution is designed for Pluse Motors and not actually suitable for Stepper Motors. 
With a stepper motor you can assign the amount of steps it takes to go from 0% to 100%. Based on this exactly calculate the amount of steps for each %.  
This will make it at first easier to setup any shutters, but more importantly it will be way more accurate in usage.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
This feature request is already an alternative solution.
**Additional context**  
_Add any other context or screenshots about the feature request here._
It would be ideal if Tasmota could also calculate the steps itself, based on a new stepper motor calibration process.
A function to say this is 0% and then ask the motor to step slowly until user clicks this is 100%. Then Tasmota knows and can also display the step count. 
**(Please, remember to close the issue when the problem has been addressed)**

dr3men, 2020-09-14:
Hello,
I think my problem can be resolved in this thread.
I have a problem calibrating roller blinds.
When the calibration passes about 70-80 percent of the distance, it immediately starts sending an infinite result. The motor continues to run indefinitely until you stop supplying power to the driver. Tasmota does not freeze.
I use: Wemos d1 mini + A4988 driver and Nema17 motor.

At the same time, if I calibrate not to the full height of the window, everything works fine.

How can I solve this problem? Can I manually specify the number of steps to open and close?

`18:06:24 CMD: ShutterCloseDuration 70
18:06:24 MQT: stat/blinde/RESULT = {"ShutterCloseDuration1":"70.0"}
18:06:29 CMD: ShutterOpenDuration 70
18:06:29 MQT: stat/blinde/RESULT = {"ShutterOpenDuration1":"70.0"}
18:06:37 CMD: ShutterClose1
18:06:37 MQT: stat/blinde/RESULT = {"POWER1":"OFF"}
18:06:37 MQT: stat/blinde/POWER1 = OFF
18:06:37 MQT: stat/blinde/RESULT = {"POWER2":"ON"}
18:06:37 MQT: stat/blinde/POWER2 = ON
18:06:37 MQT: stat/blinde/RESULT = {"POWER1":"ON"}
18:06:37 MQT: stat/blinde/POWER1 = ON
18:06:37 MQT: stat/blinde/RESULT = {"POWER3":"ON"}
18:06:37 MQT: stat/blinde/POWER3 = ON
18:06:37 MQT: stat/blinde/RESULT = {"ShutterClose1":0}
18:06:37 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0}}
18:06:38 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0}}
18:06:39 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":99,"Direction":-1,"Target":0}}
18:06:40 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":97,"Direction":-1,"Target":0}}
18:06:41 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":96,"Direction":-1,"Target":0}}
18:06:42 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":95,"Direction":-1,"Target":0}}
18:06:43 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":93,"Direction":-1,"Target":0}}
18:06:44 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":92,"Direction":-1,"Target":0}}
18:06:45 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":90,"Direction":-1,"Target":0}}
18:06:46 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":89,"Direction":-1,"Target":0}}
18:06:47 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":87,"Direction":-1,"Target":0}}
18:06:48 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":86,"Direction":-1,"Target":0}}
18:06:49 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":85,"Direction":-1,"Target":0}}
18:06:50 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":83,"Direction":-1,"Target":0}}
18:06:51 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":82,"Direction":-1,"Target":0}}
18:06:52 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":80,"Direction":-1,"Target":0}}
18:06:53 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":79,"Direction":-1,"Target":0}}
18:06:54 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":77,"Direction":-1,"Target":0}}
18:06:55 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":76,"Direction":-1,"Target":0}}
18:06:56 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":75,"Direction":-1,"Target":0}}
18:06:57 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":73,"Direction":-1,"Target":0}}
18:06:58 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":72,"Direction":-1,"Target":0}}
18:06:59 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":70,"Direction":-1,"Target":0}}
18:07:00 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":69,"Direction":-1,"Target":0}}
18:07:01 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":67,"Direction":-1,"Target":0}}
18:07:02 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":66,"Direction":-1,"Target":0}}
18:07:03 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":65,"Direction":-1,"Target":0}}
18:07:04 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":63,"Direction":-1,"Target":0}}
18:07:05 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":62,"Direction":-1,"Target":0}}
18:07:06 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":60,"Direction":-1,"Target":0}}
18:07:07 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":59,"Direction":-1,"Target":0}}
18:07:08 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":57,"Direction":-1,"Target":0}}
18:07:09 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":56,"Direction":-1,"Target":0}}
18:07:10 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":55,"Direction":-1,"Target":0}}
18:07:11 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":53,"Direction":-1,"Target":0}}
18:07:12 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":52,"Direction":-1,"Target":0}}
18:07:13 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":50,"Direction":-1,"Target":0}}
18:07:14 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":49,"Direction":-1,"Target":0}}
18:07:15 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":47,"Direction":-1,"Target":0}}
18:07:16 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":46,"Direction":-1,"Target":0}}
18:07:17 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":45,"Direction":-1,"Target":0}}
18:07:18 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":43,"Direction":-1,"Target":0}}
18:07:19 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":42,"Direction":-1,"Target":0}}
18:07:20 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":40,"Direction":-1,"Target":0}}
18:07:21 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":39,"Direction":-1,"Target":0}}
18:07:22 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":37,"Direction":-1,"Target":0}}
18:07:23 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":36,"Direction":-1,"Target":0}}
[... cut ...]
18:07:39 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":15}}
18:07:39 MQT: stat/blinde/RESULT = {"POWER3":"OFF"}
18:07:39 MQT: stat/blinde/POWER3 = OFF
18:07:40 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":15}}
18:07:41 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":15}}
18:07:42 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":15}}
18:07:43 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":15}}`

Jason2866, 2020-09-14:
@dr3men  please do not hijack this feature request with a different issue / question

stefanbode, 2020-09-15:
This is already working. Just use the calibration command and put in your desired steps. Feedback welcome. Regarding open duration and closeduration I am sorry. We have to stay on a common level. But steps is easily to calculate backwards.

dr3men, 2020-09-16:
> This is already working. Just use the calibration command and put in your desired steps. Feedback welcome. Regarding open duration and closeduration I am sorry. We have to stay on a common level. But steps is easily to calculate backwards.

Are you talking about this? 
ShutterCalibration<x> | Granular shutter position calibration. The measured opening position of the shutter at the 30, 50, 70, 90, and 100 percent opened locations. For example: ShutterCalibration<x> 23 38 56 74 82
-- | --
But judging by the instructions from the calibration, this is the distance, in centimeters for example. How does this relate to steps?

Kepete, 2020-09-16:
> This is already working. Just use the calibration command and put in your desired steps. Feedback welcome. Regarding open duration and closeduration I am sorry. We have to stay on a common level. But steps is easily to calculate backwards.

According to Tasmota docs: 
_ShutterCalibration<x> takes position measurements (not the time it takes to move). During calibration you position the shutter to an indicated percentage (e.g., 30%) of opening and measure the shutter position (e.g., 15). Use the same unit of measure for all your measurements (e.g., centimeters, inches, etc.). After calibration ShutterPosition<x> 30 will move to 30% opening which will correspond to the position you provided (15)._

So according to this, no, ShutterCalibration does not take steps into account.
So this feature request is still up :)

dr3men, 2020-09-16:
> > This is already working. Just use the calibration command and put in your desired steps. Feedback welcome. Regarding open duration and closeduration I am sorry. We have to stay on a common level. But steps is easily to calculate backwards.
> 
> According to Tasmota docs:
> _ShutterCalibration takes position measurements (not the time it takes to move). During calibration you position the shutter to an indicated percentage (e.g., 30%) of opening and measure the shutter position (e.g., 15). Use the same unit of measure for all your measurements (e.g., centimeters, inches, etc.). After calibration ShutterPosition 30 will move to 30% opening which will correspond to the position you provided (15)._
> 
> So according to this, no, ShutterCalibration does not take steps into account.
> So this feature request is still up :)

Ok, it turns out that you first need to set an approximate maximum open time level. And then calibrate according to the established position, bringing the Shutter to this position.
But I have a problem. I can set the opening to 50 seconds, which is only 80% of the position. If I set more, Tasmota starts sending an endless signal to Shutter.

18:07:28 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":29,"Direction":-1,"Target":0}} 
18:07:29 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":27,"Direction":-1,"Target":0}} 
18:07:30 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":26,"Direction":-1,"Target":0}} 
18:07:31 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":25,"Direction":-1,"Target":0}} 
18:07:32 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0}} 
18:07:33 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0}}

Kepete, 2020-09-16:
@dr3men as said before, please do not use this git feature request for your troubleshooting case.
This is a feature request thread.

stefanbode, 2020-09-18:
@Kepete : all fine we use the issues for issues AND feature requests. the 50 seconds looks for me like an overflow of variables. I have to check this. Regarding the calibration process: you first must configure the shutter to open 100% and close 100%. So everything must be up and running. Then in the calibration process, you match the %-Values to the real opening %. must suitable is using the cm measure for opening. But you can also use steps. internally it is normalized to the last values that is 100%. Therefore really flexible. I will now take a look on the 50second issue. What if your max_pwm_frequency you use for the stepper?

stefanbode, 2020-09-18:
I checked it with shutteropenduration 60 and the same to close. Works all fine on my side. shutterfrequency is 1000 (default).

stefanbode, 2020-09-18:
regarding: "I want to set the number of steps". This is easily calculated_ shutteropenduration = steps / shutterfrequency. If you set shuttercloseduration to a different value this will not influence the number of steps but the speed to close in relation to open.

stefanbode, 2020-09-18:
@dr3men : Please can you open a NEW issue for the problem and also attach the logs. Most interesting is the start of the shutter. I know it runs endless, therefor please only the first 10 seconds. Best with maximum loglevel (4) and also recommended to use the latest build here. Fixes will be made against this build anyway.

stefanbode, 2020-09-18:
@Kepete : Regarding your request to start /stop and read the steps and then have the 100%... Here is how you can make it:
- set "shutterfrequency 200" (now it moves very slow)
- set "shutteropenduration 300" and "shuttercloseduration 300" (now the stepper can run for 5 minutes)
open the webinterface and start from a closed shutter and hit the open button. Now the shutter moves and you can see the counter above. These are your steps. If you hit the open button again the shutter stop and you can see the exact count at this position. if not already open just hit open again.
If you know what speed you can use (shutterfrequency) then you can calculate the final openingtime with the expression in the comment above steps/frequency. As written in the description: adding shuttermotordelay to soft start/stop will not change the max steps.

stefanbode, 2020-09-18:
@dr3men : issue found and fixed. Please wait for the new pull request integrated and then test. Pretty sure it addresses your problem.

closed 2020-09-18 by ascillato2