arendst/Tasmota #9311: PZEM-004t v3 Energy Value jump

leranp, 2020-09-14:
### PROBLEM DESCRIPTION
I have 3 PZEM-004t v3 connected to NodeMcu .
Sometimes the voltage is down to 0 for 1 second ( it can be 1 pzem or 3 of them) and the Energy is jumping to enormous value, the voltage is the only value that make this problem, i do not see anything else that making this jump, after this jump the counter stop and the Energy value is not changing at all ( Energy Today & Energy Total), the others values ( Frequency\Current etc are running normal)
i tried:
- re flash
- update from 8.3.1 to 8.5.0
- changing SetOption72 

only restart can make the counter back again , but the energy value still high

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ V] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ V] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ V] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [V ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ V] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [V ] Device used (e.g., Sonoff Basic): NodeMcu 
- [ V] Tasmota binary firmware version number used: 8.5.0 (SENSORS)
  - [ V] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [V ] Flashing tools used: tasmotizer-1.1a
- [ V] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here: No Rules

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
20:59:02 MQT: tasmota_Pzem/stat/STATUS = {"Status":{"Module":18,"DeviceName":"PZEM HUB","FriendlyName":["PZEM HUB"],"Topic":"tasmota_Pzem","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:59:02 MQT: tasmota_Pzem/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T22:40:11","StartupUTC":"2020-09-13T19:18:51","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2020-06-01T12:35:59","SaveCount":28,"SaveAddress":"F8000"}}
20:59:02 MQT: tasmota_Pzem/stat/STATUS2 = {"StatusFWR":{"Version":"8.5.0(sensors)","BuildDateTime":"2020-09-09T11:42:52","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"379/699"}}
20:59:02 MQT: tasmota_Pzem/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["LH",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00CA8008","2805C8000100060000005A00000000000000","40008200","00006000","00000000"]}}
20:59:02 MQT: tasmota_Pzem/stat/STATUS4 = {"StatusMEM":{"ProgramSize":627,"Free":376,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCD","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
20:59:02 MQT: tasmota_Pzem/stat/STATUS5 = {"StatusNET":{"Hostname":"tasmota_Pzem-7500","IPAddress":"192.168.31.249","Gateway":"192.168.31.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.31.1","Mac":"18:FE:34:F1:FD:4C","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
20:59:02 MQT: tasmota_Pzem/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.31.100","MqttPort":1883,"MqttClientMask":"tasmota_Pzem","MqttClient":"tasmota_Pzem","MqttUser":"pi","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:59:02 MQT: tasmota_Pzem/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-09-14T17:59:02","Local":"2020-09-14T20:59:02","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+03:00","Sunrise":"08:26","Sunset":"21:04"}}
20:59:02 MQT: tasmota_Pzem/stat/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:59:02 MQT: tasmota_Pzem/stat/STATUS10 = {"StatusSNS":{"Time":"2020-09-14T20:59:02","ENERGY":{"TotalStartTime":"1970-01-01T00:19:50","Total":15198.079,"Yesterday":25434.854,"Today":32712.156,"Period":0,"Power":[276,119,271],"ApparentPower":[304,202,392],"ReactivePower":[127,164,283],"Factor":[0.91,0.59,0.69],"Frequency":[50,50,50],"Voltage":[229,229,228],"Current":[1.330,0.884,1.716]},"AM2301":{"Temperature":37.2,"Humidity":45.3,"DewPoint":23.4},"TempUnit":"C"}}
20:59:02 MQT: tasmota_Pzem/stat/STATUS11 = {"StatusSTS":{"Time":"2020-09-14T20:59:02","Uptime":"0T22:40:11","UptimeSec":81611,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"LH","BSSId":"24:05:88:38:E1:47","Channel":6,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
 21:01:09 CFG: CR 379/699, Busy 0
21:01:09 LOG: SerialLog 0, WebLog 4, MqttLog 2, SysLog 0, LogHost , LogPort 514, TelePeriod 60
21:01:09 HTP: Configuration
21:01:10 CFG: Saved to flash at F7, Count 29, Bytes 4096
21:01:10 DMP: 01 04 14 08 F1 05 1A 00 00 0A AF 00 00 21 31 00 04 01 F4 00 5B 00 00 12 06
21:01:10 HTP: Main Menu
21:01:11 DMP: 03 04 14 08 F0 06 B0 00 00 0A 8F 00 00 51 D4 00 03 01 F4 00 45 00 00 85 5E
21:01:12 DMP: 02 04 14 08 F5 03 71 00 00 04 9A 00 00 72 EF 00 02 01 F4 00 3A 00 00 5F 78
21:01:13 DMP: 01 04 14 08 F1 05 1E 00 00 0A B3 00 00 21 31 00 04 01 F4 00 5B 00 00 7F 9B
21:01:13 HTP: Console
21:01:14 DMP: 03 04 14 08 F1 06 A0 00 00 0A 84 00 00 51 D5 00 03 01 F4 00 45 00 00 A2 7C
21:01:15 DMP: 02 04 14 08 F5 03 6F 00 00 04 AA 00 00 72 EF 00 02 01 F4 00 3B 00 00 26 EE
21:01:15 WIF: Checking connection...
21:01:16 DMP: 01 04 14 08 F1 05 1C 00 00 0A B1 00 00 21 31 00 04 01 F4 00 5B 00 00 D9 B9
21:01:17 DMP: 03 04 14 08 F0 06 9C 00 00 0A 7D 00 00 51 D5 00 03 01 F4 00 45 00 00 13 B4
21:01:18 DMP: 02 04 14 08 F5 03 66 00 00 04 B3 00 00 72 EF 00 02 01 F4 00 3C 00 00 62 24
21:01:19 DMP: 01 04 14 08 F1 05 1D 00 00 0A AF 00 00 21 32 00 04 01 F4 00 5B 00 00 B4 C7
21:01:20 DMP: 03 04 14 08 EF 06 97 00 00 0A 70 00 00 51 D5 00 03 01 F4 00 45 00 00 E4 1D
21:01:21 DMP: 02 04 14 08 F4 03 66 00 00 04 AC 00 00 72 EF 00 02 01 F4 00 3C 00 00 75 47
21:01:22 DMP: 01 04 14 08 F0 05 1D 00 00 0A AE 00 00 21 32 00 04 01 F4 00 5B 00 00 8B BA
21:01:23 DMP: 03 04 14 08 EE 06 90 00 00 0A 66 00 00 51 D5 00 03 01 F4 00 45 00 00 5C 86
21:01:24 DMP: 02 04 14 08 F3 03 65 00 00 04 AC 00 00 72 F0 00 02 01 F4 00 3C 00 00 BC F3
21:01:25 DMP: 01 04 14 08 EE 05 1C 00 00 0A AC 00 00 21 32 00 04 01 F4 00 5B 00 00 73 42

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

TrDA-hab, 2020-09-18:
I have spare (unused) equipment for one phase. I can try to reproduce your error.
How often does the error occur - every 10 minutes, or after 1 hour, or after 3 hours, or 12 hours ... or completely by accident?
Need more information, what kind of load are you measuring (in detail)?

leranp, 2020-09-18:
1. It can be once a month, and can be every week, it is randomly, but not every hour/minute
2. Every phase is connected to my 3pshase in my home, they are connected to the same power line (230v).
3. It can be one or 3 phase that show 0 voltage for 1 second

TrDA-hab, 2020-09-19:
Good.
I started testing. We will see the results in 48 hours.
I will be using Tasmota 8.4.0.1

leranp, 2020-09-19:
Thanks for your help,
Maybe try to disconnect the pzem power for a moment, i didn't tried and in my case the pzems are in a lock closet and connected to UPS ,so i believe that disconnecting is not the problem

TrDA-hab, 2020-09-20:
1. My testing revealed no problems. Here is my schedule, loading my 3D printer. I used 8.4.1.
[image]
2. Possibly the error occurs when your UPS increases the voltage (when the voltage in the mains is low).
3. In 1-3 days I will install another PZEM for long testing to monitor my city apartment (one phase) and use 8.5.1

leranp, 2020-09-20:
It looks like you didn't have a voultge drop to 0v ,
I am thinking of changing the NodeMCU to Wemos D1 Mini to see if this is a board issue, and maybe the power supply isn't good for the NodeMCU (i plug it into the raspberry pi, i didn't have any choice because i don't have a free socket)

It will help if i will open a log server to see what happened every time?

TrDA-hab, 2020-09-20:
> я думаю о замене NodeMCU на Wemos D1 Mini

Я много тестировал NodeMCU и Wemos D1 Mini и ESP-01S . Все работают стабильно. Нет смысла в замене. 

Я думаю ваша проблема в УПС, я попробую это протестировать. Вы установили ПЗЕМ перед UPS или после UPS ? У вас есть схема подключения?

leranp, 2020-09-20:
> > I am thinking of changing the NodeMCU to Wemos D1 Mini
> 
> I have tested a lot of NodeMCU and Wemos D1 Mini and ESP-01S. All work stably. There is no point in replacing.
> 
> I think your problem is in the UPS, I will try to test it. Did you install PZEM before UPS or after UPS? Do you have a connection diagram?

I can see the UPS voltage/Load/Status , doesn't see any problem, if the voltage was drop down the other devices was also had problem ( the UPS is connected to me google wifi, raspberry pi, modem, LAN switch etc)
But the connection of the NodeMCU to the raspberry pi as power supply can make this problem

TrDA-hab, 2020-09-20:
The output waveforms can be different from the UPS.
Depending on the model, UPS systems have either a sine wave or a simulated sine wave. If your UPS emits a simulated signal (for example, when it is powered by batteries), PZEM does not understand the simulated sinusoidal signal and goes crazy.
[image]
I think 90% I found your mistake.

TrDA-hab, 2020-09-20:
I have tested now my APS 750 Samart UPS and PZEM in different combinations and everything works fine and without errors.

TrDA-hab, 2020-09-20:
For example only.
I actually tethered it for now and looked for your mistake!
[image]

You can read about my PZEM-BOX here:
https://www.thingiverse.com/thing:4583058
... I made it specifically for tests.

leranp, 2020-09-20:
> I have tested now my APC 750 Samart UPS and PZEM in different combinations and everything works fine and without errors.

> 
I have tested now my APC 750 Samart UPS and PZEM in different combinations and everything works fine and without errors.

It can happening once a week , testing it for couples of hours will probably not going to make the difference

TrDA-hab, 2020-09-20:
> It can happening once a week , testing it for couples of hours will probably not going to make the difference

What UPS are you using?
Model or manufacturer reference?

leranp, 2020-09-21:
> > It can happening once a week , testing it for couples of hours will probably not going to make the difference.
> 
> What UPS are you using?
> Model or manufacturer reference?
> 
> > It can happening once a week , testing it for couples of hours will probably not going to make the difference.
> 
> I understand. I have specially created different emergency situations for my UPS and did not receive any errors. I am not saying that I am 100% right, I am only talking about a possible problem with the UPS.

This is my UPS model : Advice AIN650

TrDA-hab, 2020-09-21:
Here are my thoughts, my 95% opinion is a UPS problem:

1. You have a UPS from a low price segment. Your UPS is generating a bad quality "square" sine wave at the output (when running on battery).

2. PZEM has a simple internal AC-DC converter that is powered by a "square" sinusoid of poor quality and periodically produces errors.

Possible solutions:
1. Buy and install a good smart-UPS. Or rent a smart-UPS from your friends (for tests).
2. Ask the developers to add to Tasmota the processing of the "Alarm status" command that sends the PZEM. Perhaps this will make it easier for you to find the problem.
https://innovatorsguru.com/wp-content/uploads/2019/06/PZEM-004T-V3.0-Datasheet-User-Manual.pdf

Good luck!

leranp, 2020-09-21:
> Here are my thoughts, my 95% opinion is a UPS problem:
>
> 1. You have a UPS from a low price segment. Your UPS is generating a bad quality "square" sine wave at the output (when running on battery). [...]

Thank you for your help,
I will make an request for this Alarm status of the PZEM, 
meanwhile i will try to switch the power of the PZEMs from UPS to one of my phase line, and also switching the power supply of the NodeMCU from raspberry pi to stand alone charger.
I thought ti calculate the KWH with a sensor template in the home assistant, because all the values from the PZEM are good, and in the template i can add and "if" statement that if the voultge is 0 he will ignore this value , what do you think?

Thanks again

TrDA-hab, 2020-09-21:
Thank you, you sped me up, and that one is very lazy )))

**I did a lot of work today:**
1. I connected my city apartment via PZEM (Tasmota 8.5.1). This is the first long-term test.
[image]
2. I connected my home server from UPS and PZEM (Tasmota 8.5.1). This is the second long-term test.
[image]

> I will make an request for this Alarm status of the PZEM,

This is great, it will help a lot of people.

> I thought ti calculate the KWH with a sensor template in the home assistant, because all the values from the PZEM are good, and in the template i can add and "if" statement that if the voultge is 0 he will ignore this value , what do you think?

I think this is a good solution to the problem. Maybe I would have added more to send a command to soft reset Tasmota.

I think now you should close this problem. If I have news I will write here. Keep in touch.
Good luck!

closed 2020-09-21 by leranp

sirLeone, 2021-01-25:
I can confirm that an issue exists even without UPS. I have a boiler for heating water and i a monitoring electricity usage with sonoff 4ch and PZEM-004T V3 connected RX/TX pins. Today morning, there were power failure on every 3 phases in my neighbourhood. After some time, when electricity went back, power usage jesterday,today and total has jumped to the sky.
[image]
Here is a graph from my Grafana instance, showing power usage from exactly this PZEM from last 10 days :
[image]

Tasmota version : 8.5.0.1(tasmota)

TrDA-hab, 2021-01-25:
> Я могу подтвердить, что проблема существует даже без ИБП. У меня есть бойлер для нагрева воды и мониторинга потребления электроэнергии с помощью sonoff 4ch и PZEM-004T V3, подключенных к контактам RX / TX. Вчера утром в моем районе было отключено электричество на всех трех фазах. Спустя какое-то время, когда электричество вернулось, потребление энергии в шутный день, сегодня и в целом подскочило до небес. Вот график из моего экземпляра Grafana, показывающий использование энергии именно этим PZEM за последние 10 дней:
> [image]
> 
> [image]
> 
> Версия тасмота: 8.5.0.1 (тасмота)

1. "PZEM-004t v3" is a stupid and cheap calculator that reacts to everything that happens nearby. You should NOT expect "PZEM-004t v3" to be magically accurate and error-free!
2. Tasmota - only transfers the data received from "PZEM-004t v3" to you and nothing more !!! Do not demand much from Tasmota (!) She cannot do this and should not do it !!!!
3. You must understand that "PZEM-004t v3" only monitors energy, and if there is a power error, it does NOT know how to respond to it correctly!

skupu, 2023-01-10:
Hi All,

I also face some bad results (mainly 0 V) happening with different timing (unfortunately quite often - even few times a day), no rule and cannot reproduce.
I have 12 pieces working in 4 buildings (3-phases each) and I noticed that there are problems most cases just with two pieces.
So this seems like a hardware bug.

I use it for voltage level alarmimg, so my automation has timeout and actually is based on all three phases, as power loss is most cases on all, not selectively on one phase. Of course if after the timeout the power supply is not restored, then the alarm is triggered.

Nevertheless I would like to have no false alarms and therefore I will probably ask supplier for a replacement.

I hope my case shows to you all that this is not a sinusoid-caused issue.

Regards,
Piotr