arendst/Tasmota #9314: PowerSet X for Devices without Power Monitoring

danielr1996, 2020-09-14:
**Have you looked for this feature in other issues and in the docs?**  
I've searched the issues and found #3208 which suggests that the feature is currently not available.
The [Documention for Power Calibration](https://tasmota.github.io/docs/Power-Monitoring-Calibration/) has no info on the supported devices.

**Is your feature request related to a problem? Please describe.**  
I cannot validate if this is already possible because i don't have the required devices and want to check the features before buying them. 

Power Monitoring with devices like the Sonoff Pow R2 is nice for devices with _varying_ loads such as computers, washing machines, tvs, etc. but quite unneccesary for devices with _static_ loads like light bulbs because the sonoff pow would always display roughly the same value or zero if turned off.

**Describe the solution you'd like**  
**Devices with Power Measurement**
Devices that already measure the power consumption would not be affected at all.

**Devices with on/off switch**
On Devices with an on/off switch but no power measurement i want to set the power consumption manually (e.g. by measuring with kill-a-watt, another sonoff pow, reading the spec for a light bulb,...) with `PowerSet X`.
Now if the device is turned on i'd like the device to report a power usage of X, if the device is turned off a power usage of 0.

**Devices with a dimmer**
In addition to the on/off behaviour (on meaning 100% dimmer position) the device would report X * dimmer position as power consumption (Assuming that dimmer position correlates linearly to the power usage). 

**Describe alternatives you've considered**  
Alternativly one could just monitor the on/off state and use that in custom software or ioBroker to calculate the power consumption afterwards. Currently i'm using a custom tasmota build with the prometheus extension, a prometheus instance that scrapes the sonoff pow and a grafana dashboard that displays the scraped values. Calculating the power usage afterwards would require either
1) setting up another software that modifies the scraped values (this might be easier if already using iobroker but im currently not planning on doing so.
2) using the [Transformations Feature](https://grafana.com/docs/grafana/latest/panels/transformations/) in Grafana and calculating the value inside grafana, but this has the drawback that the desired values are only correct in grafana but not other software that might use prometheus as a datasource.

**Additional context**  
I know this is rather an edge case but i hope there are other people that are facing the same problem.

I'm not experienced with programming of embedded software but i would get my hands dirty and try to and submit a pull request implementing this feature. 

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-09-15:
As Wh may change with voltage fluctuations, anything close to real numbers could not work. Dimmers are not linear, closer to logarithmic. Every dimmer would have to be calibrated, including whatever happens to be connected at any time.

schrej-zz, 2020-09-15:
I agree with dimmers, but for devices with constant power this might be an interesting option.

danielr1996, 2020-09-15:
> 
> 
> As Wh may change with voltage fluctuations, anything close to real numbers could not work. Dimmers are not linear, closer to logarithmic. Every dimmer would have to be calibrated, including whatever happens to be connected at any time.

Of course the voltage may fluctuate but i think this is neglectable depending on use case (at least for my use case). 
For the dimmers i have updated the issue.

malloy139, 2020-11-25:
+1 for this. I use a Shelly 1 for my electric floor heating.
Another posibillity would be to include a total power on time in seconds in the MQTT status messages.

sfromis, 2020-12-23:
What would make most sense to me would be to have a counter for on time, leaving any guesstimates about watt-hours up to the user. Note that this would not work for people expecting actual on-time when turning power off with a switch before the device running Tasmota, as on-time cannot be counted then (due to not saving constantly to flash), and Tasmota would not even be able to see when it was turned off.

I'd instead do any such calculations backend, reacting to the MQTT messages from Tasmota.

closed 2021-01-20 by ascillato2