arendst/Tasmota #9315: Add a white toggle for RGBW lights in Web UI

tduffy83, 2020-09-15:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
Currently there is no way to turn on the white channel from the Web UI for a RGBW bulb.  This led me to much confusion as I believed the template I was using was not functioning correctly.  I dove into the docs and learned this is by design.

**Describe the solution you'd like**  
Add a toggle button to the Web UI that turns the white channel on.  Could also be achieved with a white value slider a la home assistant.

**Describe alternatives you've considered**  
The alternative would be to just forget about the WEB UI as 99% of the time I'll be using HA to change the bulb.  However, the main reason I believe this should be changed as it will avoid a ton of confusion from new users like myself who believe their templates are not working when they can only control the RGB channels from the Web UI.

**Additional context**  
N/A

**(Please, remember to close the issue when the problem has been addressed)**

Kepete, 2020-09-15:
One thought(although not a solution)- it is possible to set each channel as a separate PWM channel. Then you could control R G B W all separately. 
SetOption68 ON
https://tasmota.github.io/docs/Lights/#independent-pwm-channels

But then you can't give tasmota any color values anymore, so it's not a solution really.

tduffy83, 2020-09-15:
Yea there are a bunch of workarounds, but none solve the fundamental issue of confusing new users into thinking their template is not working.  I was able to work around this by adding another PWM to an unused GPIO.  This tricked tasmota into thinking the bulb is a 5 channel bulb and giving you a white balance slider that will turn the white light on.  This does make HA think the bulb is a 5 channel bulb, but then upon adding into HA I just customized the supported_features flag to remove the white balance slider.  Again, it's not really

closed 2020-10-24 by stale[bot]