arendst/Tasmota #9318: Invalid UdpListener Library found

Programmierfreund, 2020-09-15:
1. I just have downloaded the Tasmota-8.5.0.zip that was published yesterday and have tried to compile and upload it to a new Wemos D1 via Arduino IDE 1.8.13. The upload completes but the following error message occurs during the compilation process:

Ungültige Bibliothek C:\Users\.....\Arduino\libraries\UdpListener in Missing 'url' from library gefunden (invalid library found).

2. Unfortunately the new 8.5.0 flashed Wemos D1 is self resetting and rebooting continuously. So I changed to a previous of Tasmota - and everything is fine.

Most likely the second issue is independent from the invalid library problem.

closed 2020-09-15 by close-issue-app[bot]

Jason2866, 2020-09-15:
@Programmierfreund  Arduino IDE is not a supported platform to build Tasmota

Programmierfreund, 2020-09-15:
@Jason2866 That't it - thank you very much! I will change to Platform.io.