arendst/Tasmota #9319: Shelly i3 support

craigueh, 2020-09-15:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Can support be added for the Shelly i3?
This is a 3 input device without any relays or other features and ideal for placing behind light switches etc. bacause of its small size.
The 3 mains switch inputs are tied to GPIO 12,13 and 14 and should change between on/off when mains is detected on the input.
However, I've tested with Esphome and the inputs continually bounce between on/off. If I set the input_pullup mode the inputs don't detect anything so I'm not sure where to go from here. 

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

vergoossen, 2020-09-15:
**Have you looked for this feature in other issues and in the docs?**
No

https://templates.blakadder.com/shelly_i3.html

;)

ascillato2, 2020-09-15:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-15 by ascillato2