arendst/Tasmota #9320: tuya dimmer range off

Poerk, 2020-09-15:
### PROBLEM DESCRIPTION
when dimming a tuya divice 'dimmer 100' result in "TYA: Send dim value=720"
while 'dimmer 50' result in: TYA: Send dim value=**110** (id=2)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): MS-105 (Tuya MCU)
- [x] Tasmota binary firmware version number used: 8.4.0.3
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: platformio
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
17:08:10 CMD: Backlog Template; Module; GPIO 255
17:08:10 SRC: WebConsole from 192.168.68.114
17:08:10 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
17:08:11 SRC: Backlog
17:08:11 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
17:08:11 RSL: RESULT = {"NAME":"","GPIO":[0,107,0,108,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":54}
17:08:11 SRC: Backlog
17:08:11 CMD: Group 0, Index 1, Command "MODULE", Data ""
17:08:11 RSL: RESULT = {"Module":{"0":""}}
17:08:11 SRC: Backlog
17:08:11 CMD: Group 0, Index 1, Command "GPIO", Data "255"
17:08:11 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
17:08:12 TYA: Send "55aa00000000ff"
17:08:12 RSL: RESULT = {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
17:08:12 TYA: Heartbeat
17:08:16 WIF: Checking connection...```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
17:09:15 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"TasDimmerEettafel","FriendlyName":["TasDimmerEettafel"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:09:15 RSL: STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:25:45","StartupUTC":"2020-09-15T14:43:30","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2020-09-09T16:12:25","SaveCount":116,"SaveAddress":"F5000"}}
17:09:15 RSL: STATUS2 = {"StatusFWR":{"Version":"8.4.0.3(tasmota)","BuildDateTime":"2020-08-30T22:13:57","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"418/699"}}
17:09:15 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MiNieHome","NielsAP"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00010000","00006000","00000000"]}}
17:09:15 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":511,"Free":492,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","07D02792","04108001","00000005","010000C0","C0000800","00000000","00000000"],"Drivers":"1,2,3,4,7,8,9,12,16,20,21,29,35","Sensors":"1,2,3"}}
17:09:15 RSL: STATUS5 = {"StatusNET":{"Hostname":"tas-dim-eetkamer","IPAddress":"192.168.68.126","Gateway":"192.168.68.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"24:62:AB:56:1C:D8","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
17:09:15 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-09-15T16:09:15","Local":"2020-09-15T17:09:15","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:27","Sunset":"19:02"}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
17:10:35 CMD: **dimmer 50**
17:10:35 SRC: WebConsole from 192.168.68.114
17:10:35 CMD: Group 0, Index 1, Command "DIMMER", Data "50"
17:10:35 RSL: RESULT = {"POWER":"ON","Dimmer":50}
17:10:35 TYA: Send dim **value=110** (id=2)
17:10:35 TYA: Send "55aa00060008020200040000006e83"
17:10:35 RSL: RESULT = {"TuyaReceived":{"Data":"55AA00070008020200040000006E84","Cmnd":7,"CmndData":"020200040000006E","DpType2Id2":110,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000006E"}}}
17:10:35 TYA: fnId=21 is set for dpId=2
17:10:35 TYA: RX Dim State=110
17:10:36 WIF: Checking connection...
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
DimmerRange 10, 720
dimmer 50

### EXPECTED BEHAVIOUR
Wheb issuing the command 'dimmer 50' i expect : TYA: Send dim **value=361** (id=2)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Poerk, 2020-09-15:
on tasmota version 8.1.0.0 it works like expected:

16:56:44 CMD: dimmer 50
16:56:44 SRC: WebConsole from 192.168.68.114
16:56:44 CMD: Group 0, Index 1, Command "DIMMER", Data "50"
16:56:44 RSL: RESULT = {"POWER":"ON","Dimmer":50}
16:56:45 TYA: Send dim value=**361** (id=2)
16:56:45 TYA: Send "55aa0006000802020004000001697f"
16:56:45 UDP: Packet (173)
16:56:45 RSL: RESULT = {"TuyaReceived":{"Data":"55AA00070008020200040000016980","Cmnd":7,"CmndData":"0202000400000169","DpId":2,"DpIdType":2,"DpIdData":"00000169"}}
16:56:45 TYA: fnId=21 is set for dpId=2
16:56:45 TYA: RX Dim State=361
16:56:47 TYA: Send "55aa00000000ff"
16:56:47 RSL: RESULT = {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
16:56:47 TYA: Heartbeat
16:56:48 UDP: Packet (123)

Poerk, 2020-09-15:
after some more testing i found this:
Using ```TuyaSend2 2 361``` then the light will dimm to a correct 50% level. 
it results in:
```"DIMMER", Data "50"``` and  ```dim skipped value=110```

But when using ```dimmer 50``` 
it results in: ```TYA: Send dim value=110 (id=2)``` 

mayby this 110 value is a coincident, but might it be possible that the skipped value and dim value is swapped?

log files for both commands:

20:39:57 CMD:``` tuyasend2 2,361```
20:39:57 SRC: WebConsole from 192.168.68.114
20:39:57 CMD: Group 0, Index 2, Command "TUYASEND", Data "2,361"
20:39:57 TYA: Send "55aa0006000802020004000001697f"
20:39:57 RSL: RESULT = {"TuyaSend":"Done"}
20:39:57 {"TuyaReceived":{"Data":"55AA00070008020200040000016980","Cmnd":7,"CmndData":"0202000400000169","DpType2Id2":361,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000169"}}}
20:39:57 TYA: fnId=21 is set for dpId=2
20:39:57 TYA: RX Dim State=361
20:39:57 SRC: Switch
20:39:57 CMD: Group 0, Index 3, Command "DIMMER", Data "50"
20:39:57 TYA: Send ```dim skipped value=110```
20:39:57 RSL: RESULT = ```{"POWER":"ON","Dimmer":50}```
20:39:58 CFG: Saved to flash at F6, Count 206, Bytes 4096

------

20:41:17 CMD: ```dimmer 50```
20:41:17 SRC: WebConsole from 192.168.68.114
20:41:17 CMD: Group 0, Index 1, Command "DIMMER", Data "50"
20:41:17 RSL: RESULT = {"POWER":"ON","Dimmer":50}
20:41:17 TYA: Send``` dim value=110 ```(id=2)
20:41:17 TYA: Send "55aa00060008020200040000006e83"
20:41:17 {"TuyaReceived":{"Data":"55AA00070008020200040000006E84","Cmnd":7,"CmndData":"020200040000006E","DpType2Id2":110,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000006E"}}}
20:41:17 TYA: fnId=21 is set for dpId=2
20:41:17 TYA: RX Dim State=110
20:41:19 TYA: Send "55aa00000000ff"
20:41:19 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}

effelle, 2020-09-15:
Disable `ledtable` .

Poerk, 2020-09-16:
@effelle  that seems to fix it. thanks

closed 2020-09-16 by Poerk