arendst/Tasmota #9325: POWER resets PULSETIME

btro1404, 2020-09-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Value of PULSETIME is set to zero after receive POWER without argument

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Garden Water Timer BQ05 Valve 
- [x] Tasmota binary firmware version number used: 8.5.0 (8.4.0 also)
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tasmotizer 1.0
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
12:02:58 CMD: Backlog Template; Module; GPIO 255
12:02:59 MQT: stat/Wasser96/RESULT = {"NAME":"BQ05","GPIO":[17,0,0,0,0,0,0,0,21,157,0,0,0],"FLAG":0,"BASE":18}
12:02:59 MQT: stat/Wasser96/RESULT = {"Module":{"0":"BQ05"}}
12:02:59 MQT: stat/Wasser96/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"157":"LedLink"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:

```
  Rules output here:
12:04:04 CMD: Backlog Rule1; Rule2; Rule3
12:04:04 MQT: stat/Wasser96/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
12:04:04 MQT: stat/Wasser96/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
12:04:04 MQT: stat/Wasser96/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:04:59 CMD: Status 0
12:04:59 MQT: stat/Wasser96/STATUS = {"Status":{"Module":0,"DeviceName":"TasmotaWasser96","FriendlyName":["Tasmota Wasser 96"],"Topic":"%hostname%","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:04:59 MQT: stat/Wasser96/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-DE.bin","RestartReason":"Software/System restart","Uptime":"0T00:41:27","StartupUTC":"2020-09-16T09:23:32","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2020-08-17T19:18:44","SaveCount":271,"SaveAddress":"FB000"}}
12:04:59 MQT: stat/Wasser96/STATUS2 = {"StatusFWR":{"Version":"8.5.0(tasmota)","BuildDateTime":"2020.09.09 11:47:14","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"418/699"}}
12:04:59 MQT: stat/Wasser96/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
12:04:59 MQT: stat/Wasser96/STATUS4 = {"StatusMEM":{"ProgramSize":596,"Free":404,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
12:04:59 MQT: stat/Wasser96/STATUS5 = {"StatusNET":{"Hostname":"Wasser96","IPAddress":"192.168.64.96","Gateway":"192.168.64.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.64.254","Mac":"C8:2B:96:D0:0B:E8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:04:59 MQT: stat/Wasser96/STATUS6 = {"StatusMQT":{"MqttHost":"ccu3.fritz.box","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D00BE8","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:04:59 MQT: stat/Wasser96/STATUS7 = {"StatusTIM":{"UTC":"2020-09-16T10:04:59","Local":"2020-09-16T12:04:59","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:55","Sunset":"19:32"}}
12:04:59 MQT: stat/Wasser96/STATUS10 = {"StatusSNS":{"Time":"2020-09-16T12:04:59"}}
12:04:59 MQT: stat/Wasser96/STATUS11 = {"StatusSTS":{"Time":"2020-09-16T12:04:59","Uptime":"0T00:41:27","UptimeSec":2487,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxxxx","BSSId":"xx:xx:xx:xx:D5:AD","Channel":6,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
11:47:21 CMD: power on
11:47:21 SRC: WebConsole from 192.168.64.14
11:47:21 CMD: Gruppe 0, Index 1, Befehl "POWER", Daten "on"
11:47:21 MQT: stat/Wasser96/RESULT = {"POWER":"ON"}
11:47:21 MQT: stat/Wasser96/POWER = ON
11:47:21 CFG: in Flash gespeichert am F4, zählen 270, Bytes 4096
11:47:24 CMD: pulsetime1
11:47:24 SRC: WebConsole from 192.168.64.14
11:47:24 CMD: Gruppe 0, Index 1, Befehl "PULSETIME", Daten ""
11:47:24 MQT: stat/Wasser96/RESULT = {"PulseTime1":{"Set":7300,"Remaining":7297}}
11:47:28 CMD: power
11:47:28 SRC: WebConsole from 192.168.64.14
11:47:28 CMD: Gruppe 0, Index 1, Befehl "POWER", Daten ""
11:47:28 MQT: stat/Wasser96/RESULT = {"POWER":"ON"}
11:47:28 MQT: stat/Wasser96/POWER = ON
11:47:32 CMD: pulsetime1
11:47:32 SRC: WebConsole from 192.168.64.14
11:47:32 CMD: Gruppe 0, Index 1, Befehl "PULSETIME", Daten ""
11:47:32 MQT: stat/Wasser96/RESULT = {"PulseTime1":{"Set":7300,"Remaining":0}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. POWER ON
2. PULSETIME1
3. POWER
4. PULSETIME1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
After I ask the device in step 3. about its status (answer "POWER":"ON")  I expect in step 4. an answer like in step 2 ("Remaining":7289). But the device sends "Remaining":0 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The device will not automatically power off after the configured PULSETIME value

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-09-16:
As designed. That is an intended behavior.

closed 2020-09-16 by ascillato2

btro1404, 2020-09-16:
From the Docs:
_Every command used without a parameter (payload) returns the current setting._ 
Which means: A command without payload will change nothing - it will only report the current setting. But the command `POWER` will change the "remaining PulseTime".

If this issue is really not a bug: Please change the docs in the sections "Power<x>" and "PulseTime<x>" to the correct behaviour like: `POWER` (without parameter) will set the "Remaining PulseTime" to zero. Because of that the device will not automatically switch off after receiving this command although `PULSETIME` is set.

Or - please - re-open this issue.

BTW: 
In section _Power<x>" the doc says "Control the corresponding Relay (enumerated) power state and also restart PulseTime<x>._ That is not right. `Power on` restarts the remaining PulseTime, `Power off` set it to zero.

ascillato, 2020-09-17:
Please, help us and do a pull request in the docs with this information. Thanks.

btro1404, 2020-09-20:
I made a draft (for myself). But also if I think twice about the devices behavior, I can't see an advantage of this design. It makes no sense to cancel a started countdown only because the device was asked about its status.

Where can I get informations about the sense behind this behavior?

And: How can I request a change of the design?

ascillato, 2020-09-20:
> Where can I get informations about the sense behind this behavior?

It was an old feature request. Couldn't find the exact issue where it was requested and explained. Sorry. But this is an intended behaviour. Changing that, will be a breaking change for other users setups that are using this for very long time.

> How can I request a change of the design?

Open a Feature Request or what is way better and more efective, make a Pull Request with the code change in order to review it.

btro1404, 2020-09-20:
> But this is an intended behaviour. Changing that, will be a breaking change for other users setups that are using this for very long time.

Sorry about my renewed intervention, but your answer is not logical for me because in the docs is a warning for unnecessary updates:

> Unless your Tasmota powered device exhibits a problem or you need to make use of a feature that is not available in the Tasmota version currently installed on your device, leave your device alone - it works so don't make unnecessary changes! 

Other users will only affected if they update their devices anyway.

> make a Pull Request with the code change in order to review it.

I'm sorry, unfortunately I'm not a programmer. 

Thanks for your help. I will think about my future steps.