arendst/Tasmota #9326: Tasmota + HM10 + MiFlora, wrong data

phoxy84, 2020-09-16:
### PROBLEM DESCRIPTION
I have a board with Tasmota and HM10 (v708) connected to it. Sometimes the readings from my Xiaomi Flora sensors are very strange, and incorrect.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): ESP32 with HM10 (v708)
- [ X] Tasmota binary firmware version number used: 8.5.0.1 (sensors)
  - [ X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [X ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
14:36:41 MQT: stat/HM10/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
14:36:41 MQT: stat/HM10/RESULT = {"Module":{"0":"Generic"}}
14:36:41 MQT: stat/HM10/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"194":"HM10 RX"},"GPIO3":{"0":"None"},"GPIO4":{"195":"HM10 TX"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [X ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
14:37:14 MQT: stat/HM10/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":35,"Free":476,"Rules":"on system#boot do hm10auto 60 endon"}
14:37:14 MQT: stat/HM10/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":67,"Free":444,"Rules":"ON HM10#Found<5 DO Add1 1 ENDON ON Var1#State<=3 DO HM10Scan ENDON "}
14:37:14 MQT: stat/HM10/RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":34,"Free":477,"Rules":"on system#boot do hm10page 5 endon"}

```
- [ X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
14:37:33 MQT: stat/HM10/STATUS = {"Status":{"Module":0,"DeviceName":"HM10","FriendlyName":["HM10"],"Topic":"HM10","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:37:33 MQT: stat/HM10/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-sensors.bin.gz","RestartReason":"Software Watchdog","Uptime":"0T02:14:47","StartupUTC":"2020-09-16T10:22:46","Sleep":50,"CfgHolder":4617,"BootCount":2784,"BCResetTime":"2020-05-20T13:37:47","SaveCount":2864,"SaveAddress":"FB000"}}
14:37:33 MQT: stat/HM10/STATUS2 = {"StatusFWR":{"Version":"8.5.0.1(sensors)","BuildDateTime":"2020-09-12T19:05:15","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"371/699"}}
14:37:33 MQT: stat/HM10/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VFNL-C9D1E0_ub",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
14:37:33 MQT: stat/HM10/STATUS4 = {"StatusMEM":{"ProgramSize":627,"Free":376,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCD","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
14:37:33 MQT: stat/HM10/STATUS5 = {"StatusNET":{"Hostname":"HM10-6674","IPAddress":"10.0.0.55","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.1","Mac":"EC:FA:BC:5E:DA:12","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:37:33 MQT: stat/HM10/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.15","MqttPort":1883,"MqttClientMask":"HM10","MqttClient":"HM10","MqttUser":"hassio","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
14:37:33 MQT: stat/HM10/STATUS7 = {"StatusTIM":{"UTC":"2020-09-16T12:37:33","Local":"2020-09-16T14:37:33","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:29","Sunset":"20:00"}}
14:37:33 MQT: stat/HM10/STATUS10 = {"StatusSNS":{"Time":"2020-09-16T14:37:33","MJ_HT_V1-d8a69d":{"Temperature":25.7,"Humidity":56.0,"DewPoint":16.3,"Battery":97},"TempUnit":"C"}}
14:37:33 MQT: stat/HM10/STATUS11 = {"StatusSTS":{"Time":"2020-09-16T14:37:33","Uptime":"0T02:14:47","UptimeSec":8087,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"VFNL-C9D1E0_ub","BSSId":"80:2A:A8:94:69:1E","Channel":11,"RSSI":100,"Signal":-15,"LinkCount":1,"Downtime":"0T00:00:03"}}}
14:37:33 MQT: stat/HM10/STATUS12 = {"StatusSTK":{"Exception":4,"Reason":"Software Watchdog","EPC":["4022df0d","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4022df1c","4023baa2","40214822","40236e7d","4024dde8","40101f15"]}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Just wait for an incorrect reading.

### EXPECTED BEHAVIOUR
No incorrect readings.

### SCREENSHOTS
The C4:7c:8d:66:08:99 moisture data is 0% and fertility 46382, light 264960. 
[image]

This should be (and is most of the times)
light 984, moisture 46% and fertility 1187.
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Staars, 2020-09-16:
My initial guess is, that the HM-10 sends an unexpected (error) message beginning with an „f“ (ASCII code 46).
I will take a look.

phoxy84, 2020-09-17:
I looked into the data, the issue is with moisture, fertility and light. The temperature sensor seems to be ok, see attachment

Temperature:
[image]

Illuminance:
[image]

Moisture:
[image]

Staars, 2020-09-17:
Thanks for the Info.
Are you able to compile? I could eventually prepare a version with more debug output for you.

phoxy84, 2020-09-17:
No, I don’t know how that works, sorry

Staars, 2020-09-20:
No problem, I am already working on a new version.
The problem seems to be, that the timing of the response of the sensor is quite variable and I need to add more error checking.

Staars, 2020-09-23:
The new development version is online.
I assume that your bad readings came from unexpected disconnections and the misinterpretation of the return value `CONNF`.

Anyway, after endless looking at the returned scan data, I am pretty sure now, that no scan responses do occur (passive scan) and that means, it is perfectly fine to scan all the time to read the data via advertisements. This is way more battery efficient for the sensors and the recommended way.

Example for the Flora sensors:
`HM10PERIOD 86400` (for battery once a day)
`HM10AUTO 1` scan all the time

More updates will come later.

phoxy84, 2020-09-24:
I will give it a try and will let you know

Staars, 2020-09-24:
Please post console output (seriallog 3) from that moments. I have activated more debug messages for the Flora sensors in the recent driver to see the buffer.

phoxy84, 2020-09-24:
I found one error now:

`
11:12:04 MQT: tele/HM10/SENSOR = {"Time":"2020-09-24T11:12:04","Flora-61b3d5":{"Temperature":22.5,"Illuminance":583,"Moisture":36,"Fertility":600},"Flora-660899":{"Temperature":23.4,"Illuminance":**286720**,"Moisture":28,"Fertility":**60444**,"Battery":38},"Flora-6a48b3":{"Temperature":23.4,"Illuminance":596,"Moisture":66,"Fertility":392,"Battery":95},"MJ_HT_V1-d8b304":{"Temperature":22.8,"Humidity":50.7,"DewPoint":12.0,"Battery":83},"MJ_HT_V1-d8a69d":{"Temperature":22.8,"Humidity":49.3,"DewPoint":11.6,"Battery":94},"TempUnit":"C"}`

Staars, 2020-09-24:
To narrow it down, I need the log data at the time of the faulty measurement. You can use `seriallog 3` and then you should get something like that:
```
15:06:14 HM10: Flora connect AT+CONc47c8d6ab577
15:06:14 HM10: response: OK+CONNA
15:06:14 HM10: get response
15:06:14 HM10: get response
15:06:15 HM10: response: OK+CONN

15:06:15 HM10: read handle 0038
15:06:15 HM10: Battery: 99
15:06:15 HM10: get response
15:06:18 HM10: response: 
15:06:18 HM10: write to handle 0033
15:06:19 HM10: read handle 0035
15:06:19 DMP: E0 00 00 00 00 00 00 00 00 00
15:06:19 HM10: T * 10: 224, L: 0, M: 0, F: 0
```

The line with DMP contains the data, that the HM-10 passes to the ESP8266. I must see, if the error happens there or later. (No problems in my setup)

phoxy84, 2020-09-24:
My output is different:

`16:29:12 CMD: Seriallog 3
16:29:12 MQT: stat/HM10/RESULT = {"SerialLog":{"3":{"Active":"3"}}} `

Empty after this line

Staars, 2020-09-24:
Yes, we need the log from the time of measurement. But the loglevel is definitely correct.

You should always be able to see a log output similiar to mine some seconds after reboot (and a lot more).

phoxy84, 2020-09-24:
Thuis is after reboot:

`00:00:00 CFG: Loaded from flash at F8, Count 3025
00:00:00 QPC: Count 1
00:00:00 Project tasmota HM10 Version 8.5.0.1(sensors)-2_7_4_1
00:00:00 WIF: Connecting to AP1 VFNL-C9D1E0_ub Channel 11 BSSId 80:2A:A8:94:69:1E in mode 11N as HM10-6674...
00:00:01 WIF: Connected
16:38:37 HTP: Web server active on HM10-6674 with IP address 10.0.0.55
16:38:38 MQT: Attempting connection...
16:38:39 MQT: Connected
16:38:39 MQT: tele/HM10/LWT = Online (retained)
16:38:39 MQT: cmnd/HM10/POWER = 
16:38:39 MQT: tele/HM10/INFO1 = {"Module":"Generic","Version":"8.5.0.1(sensors)","FallbackTopic":"cmnd/HM10_fb/","GroupTopic":"cmnd/tasmotas/"}
16:38:39 MQT: tele/HM10/INFO2 = {"WebServerMode":"Admin","Hostname":"HM10-6674","IPAddress":"10.0.0.55"}
16:38:39 MQT: tele/HM10/INFO3 = {"RestartReason":"Software/System restart"}
16:38:39 RUL: SYSTEM#BOOT performs "hm10auto 60"
16:38:39 MQT: stat/HM10/RESULT = {"HM10Auto":60}
16:38:39 RUL: SYSTEM#BOOT performs "hm10page 5"
16:38:39 MQT: stat/HM10/RESULT = {"HM10Page":5}
16:38:39 MQT: homeassistant/sensor/5EDA12_status/config = {"name":"HM10 status","stat_t":"tele/HM10/HASS_STATE","avty_t":"tele/HM10/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"tele/HM10/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"5EDA12_status","dev":{"ids":["5EDA12"],"name":"HM10","mdl":"Generic","sw":"8.5.0.1(sensors)","mf":"Tasmota"}} (retained)
16:38:41 QPC: Reset
16:38:42 MQT: tele/HM10/STATE = {"Time":"2020-09-24T16:38:42","Uptime":"0T00:00:09","UptimeSec":9,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"Wifi":{"AP":1,"SSId":"VFNL-C9D1E0_ub","BSSId":"80:2A:A8:94:69:1E","Channel":11,"RSSI":100,"Signal":-16,"LinkCount":1,"Downtime":"0T00:00:03"}} `

I don’t understanding what i am doing wrong

ascillato, 2020-09-24:
Instead of seriallog, please use weblog.

Seriallog is when you are reading the data from the serial port using an ftdi or serial adapter. For logging level on the web interface, the command is `weblog`.

phoxy84, 2020-09-24:
That works. I keep you posted

Staars, 2020-09-24:
... the later messages.

The driver initializes, the does 1 or more scans and after that connects to every sensor to read data.
I need the log messages from the last part.
(Connecting repeats every n seconds, where n is HM10period n or - if not set - is equal to Teleperiod).

phoxy84, 2020-09-24:
An 08:03:00 log should have a problem with the sensor.hm10_flora_660899_illuminance sensor, sensor.hm10_flora_660899_fertility and sensor.hm10_flora_660899_moisture

Hope it is in the log files

`
20:02:10 HM10: get response
20:02:14 HM10: write to handle 0033
20:02:14 HM10: response: 
20:02:14 HM10: read handle 0035
20:02:14 DMP: E4 00 00 97 00 00 00 35 79 08
20:02:14 HM10: T * 10: 228, L: 151, M: 53, F: 2169
20:02:14 HM10: get response
20:02:15 HM10: disconnect
20:02:15 HM10: response: OK+LOST

20:02:15 HM10: get response
20:02:16 HM10: disconnect
20:02:16 HM10: response: OK
20:02:17 HM10: get response
20:02:17 HM10: Flora connect AT+CONc47c8d6a48b3
20:02:17 HM10: response: OK+CONNA
20:02:17 HM10: get response
20:02:17 HM10: response: OK+CONN

20:02:17 HM10: get response
20:02:19 HM10: read handle 0038
20:02:19 HM10: Battery: 95
20:02:19 HM10: get response
20:02:20 HM10: response: 
20:02:22 HM10: write to handle 0033
20:02:23 HM10: read handle 0035
20:02:23 DMP: E8 00 00 23 00 00 00 42 96 01
20:02:23 HM10: T * 10: 232, L: 35, M: 66, F: 406
20:02:23 HM10: get response
20:02:24 HM10: disconnect
20:02:24 HM10: response: OK+LOST

20:02:24 HM10: get response
20:02:25 HM10: MJ_HT_V1 connect AT+CON4c65a8d8a69d
20:02:25 HM10: response: OK+CONNA
20:02:25 HM10: get response
20:02:28 HM10: response: OK+CONN

20:02:28 HM10: get response
20:02:30 HM10: read handle 0x18
20:02:30 HM10: Battery: 94
20:02:30 HM10: get response
20:02:31 HM10: subscribe to 0x0f
20:02:31 HM10: T * 10: 231, H * 10: 478
20:02:31 HM10: get response
20:02:32 HM10: disconnect
20:02:32 HM10: response: OK+LOST

20:02:32 HM10: get response
20:02:33 HM10: MJ_HT_V1 connect AT+CON4c65a8d8b304
20:02:33 HM10: response: OK+CONNA
20:02:33 HM10: get response
20:02:37 HM10: get response
20:02:39 HM10: read handle 0x18
20:02:39 HM10: get response
20:02:40 HM10: subscribe to 0x0f
20:02:41 HM10: get response
20:02:42 HM10: disconnect
20:02:43 HM10: get response
20:02:43 HM10: disconnect
20:02:44 HM10: get response
20:02:44 HM10: Flora connect AT+CONc47c8d660899
20:02:44 HM10: response: OK+CONNA
20:02:44 HM10: get response
20:02:44 HM10: response: OK+CONN

20:02:44 HM10: get response
20:02:46 HM10: read handle 0038
20:02:46 HM10: Battery: 37
20:02:47 HM10: get response
20:02:47 HM10: response: 
20:02:50 HM10: write to handle 0033
20:02:50 HM10: read handle 0035
20:02:50 DMP: 00 E8 00 00 56 00 00 00 2C BE
20:02:50 HM10: T * 10: 59392, L: 22016, M: 0, F: 48684
20:02:50 HM10: get response
20:02:51 HM10: disconnect
20:02:51 HM10: response: OK+LOST

20:02:52 HM10: get response
20:03:35 MQT: tele/HM10/HASS_STATE = {"Version":"8.5.0.1(sensors)","BuildDateTime":"2020-09-23T18:05:11","Module or Template":"Generic","RestartReason":"Software/System restart","Uptime":"0T03:25:02","Hostname":"HM10-6674","IPAddress":"10.0.0.55","RSSI":"100","Signal (dBm)":"-19","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:03","MqttCount":1,"LoadAvg":19}
20:03:42 MQT: tele/HM10/STATE = {"Time":"2020-09-24T20:03:42","Uptime":"0T03:25:09","UptimeSec":12309,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"VFNL-C9D1E0_ub","BSSId":"80:2A:A8:94:69:1E","Channel":11,"RSSI":100,"Signal":-19,"LinkCount":1,"Downtime":"0T00:00:03"}}
20:03:42 MQT: tele/HM10/SENSOR = {"Time":"2020-09-24T20:03:42","Flora-61b3d5":{"Temperature":22.8,"Illuminance":151,"Moisture":53,"Fertility":2169},"Flora-6a48b3":{"Temperature":23.2,"Illuminance":35,"Moisture":66,"Fertility":406,"Battery":95},"MJ_HT_V1-d8a69d":{"Temperature":23.1,"Humidity":47.8,"DewPoint":11.4,"Battery":94},"MJ_HT_V1-d8b304":{"Temperature":22.8,"Humidity":45.2,"DewPoint":10.3,"Battery":83},"Flora-660899":{"Temperature":23.0,"Illuminance":22016,"Moisture":0,"Fertility":48684,"Battery":37},"TempUnit":"C"}
20:03:52 HM10: disconnect
20:03:52 HM10: response: OK
20:03:53 HM10: get response
20:03:53 HM10: start discovery
20:04:02 HM10: get response
20:04:02 HM10: show status
20:04:02 RUL: Event too long (18)
20:04:02 RUL: Event too long (18)
20:04:02 RUL: Event too long (18)
20:04:02 RUL: Event too long (18)
20:04:47 HTP: Console
20:05:04 HM10: disconnect
20:05:05 HM10: get response
20:05:05 HM10: start discovery
20:05:14 HM10: get response
20:05:14 HM10: show status
20:05:14 RUL: Event too long (18)
20:05:14 RUL: Event too long (18)
20:05:14 RUL: Event too long (18)
20:05:14 RUL: Event too long (18)`

Staars, 2020-09-24:
This helps:
20:08:29 DMP: 00 E7 00 00 44 00 00 00 2C B8
20:08:29 HM10: T * 10: 59136, L: 17408, M: 0, F: 47148

... somehow :)

This means that there is a leading 0 in the receive buffer, that shifts the whole buffer to the right one byte.

I will have to do some tests and hope to find a way to prevent this reliably. Hopefully this is just a matter of the timing of the various commands.

Please test, if reading a advertisements only works for you. This particular error should not happen that way and I recommend this anyway.
Like that:

`HM10PERIOD 86400` (for battery once a day)
`HM10AUTO 1` scan all the time (if you can live with more missed packets, you can change 1 second to a higher value, but 1 should work fine)

phoxy84, 2020-09-24:
Ok, I will try and let you know.

Staars, 2020-09-26:
A new development version is out for testing.
Please test both options (current standard=active connections and permanent scanning=passive readings) as I want to develop the driver away from periodically active connections (probably leaving the possibility as a fallback).
In the end I want to behave all 3 BLE-drivers as similar as technically possible and working primarily as bridge.

ascillato2, 2020-09-26:
https://github.com/arendst/Tasmota/pull/9388

closed 2020-09-26 by ascillato2