arendst/Tasmota #9327: support for multiple  Hiking DDS238-2 ZN/S on the same modbus wiring

capputeam, 2020-09-16:
**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
A single Hiking DDS238-2 ZN/S with modbus id=1 works perfectly wired with a sonoff basic with tasmota.sensor loaded.
Changing the modbus id != 1 tasmota is not able to fetch the power data anymore.
Even sending serial command the response is not recognized
i.e. with modbus id=2
13:37:59 CMD: Group 0, Index 5, Command "SERIALSEND", Data "020300000032c42c"
13:37:59 MQT: stat/ProvaRS485/RESULT = {"SerialSend":"Done"}
13:38:00 DMP: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
13:38:00 DBG: DDS2382 response error 9
13:38:01 DMP: 
13:38:01 DBG: DDS2382 response error 7

**Describe the solution you'd like**  
It would be nice if it could be possible to set the modbus id of the device.
It would be more than nice to have the possibility to configure a set of modbus id in order to manage multiple devices on RS485 modbus.
The optimal solution would be to let the device able to autodiscover the devices present in the same modbus wiring and collect data from all automatically.

**Describe alternatives you've considered**  
The only alternative I found is use ESPEASY instead of Tasmota, but personally I'd prefer not doint this.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-09-16:
have a Look at https://tasmota.github.io/docs/Smart-Meter-Interface/#2-sbc-ale3-modbus

there is an example of 2 modbus devices on the same bus

ascillato2, 2020-09-18:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-18 by ascillato2

capputeam, 2020-09-28:
Hi again

I tried many times to use scripts but nothing happened.
I built a tasmota sensor version 8.5.0.1 with script enabled (following the procedure  reported in the page linked above).
I'm pretty sure that wiring is fine (with modscan on pc I can read all the values from all the devices).
I tried to use the following script but nothing happens, no values are returned at all.
I dont know how to debug the script, in the console (weblog 4 and seriallog 4) nothing is reported.

>D
>B
=>sensor53 r
>M 1	  
	  
+1,3,m,1,9600,Hiking,1,1,0104000c,01040010,01040011,0304000c,03040010,03040011,0404000c,04040010,04040011,0504000c,05040010,05040011

1,=h Contatore 1
1,010304UUuu@i0:10,C1_Voltage,V,C1Voltage,0
1,010304UUuu@i1:1000,C1_CosPhi,,C1CosPhi,2
1,010304UUuu@i2:100,C1_Frequency,Hz,C1Frequency,0
1,=h Contatore 3
1,030304UUuu@i3:10,C3_Voltage,V,C3Voltage,0
1,030304UUuu@i4:1000,C3_CosPhi,,C3CosPhi,2
1,030304UUuu@i5:100,C3_Frequency,Hz,C3Frequency,0
1,=h Contatore 4
1,040304UUuu@i6:10,C4_Voltage,V,C1Voltage,0
1,040304UUuu@i7:1000,C4_CosPhi,,C4CosPhi,0
1,040304UUuu@i8:100,C4_Frequency,Hz,C4Frequency,0	
1,=h Contatore 5
1,050304UUuu@i0:10,C5_Voltage,V,C5Voltage,0
1,050304UUuu@i1:1000,C5_CosPhi,,C5CosPhi,0
1,050304UUuu@i2:100,C5_Frequency,Hz,C5Frequency,0  
#

gemu2015, 2020-09-28:
did you enable SML driver ?
#define USE_SML_M

// these are useful too, also not used currently in your script
#define USE_SML_SCRIPT_CMD
#define SML_REPLACE_VARS

capputeam, 2020-09-28:
This is what I added to the user_config_override.h file:
#ifndef USE_SCRIPT
#define USE_SCRIPT  # adds about 17k flash size, variable ram size
#endif
#ifndef USE_SML_M
#define USE_SML_M
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif
// define max number of decoder entries (defaults to 20 if not defined)
#define SML_MAX_VARS 20

gemu2015, 2020-09-28:
ok, then you should at least see the web interface with your decoder values ?

debugging

i would suggest to first set repeat transmission rate to at least 2 (200 ms)

+1,3,m,1,9600,Hiking,1,**2**,0104000c,01040010,01040011,0304000c,03040010,03040011,0404000c,04040010,04040011,0504000c,05040010,05040011

now go to console and enable sml debug by typing
sensor53 d1

you now should see the hex values returned from your modbus device (if it responds)

for resuming normal operation type
sensor53 d0 

do you see the transmission led of you modbus transceiver blinking ?

normally your transmission and receiving leds should blink every 0,2 seconds

if only the transmission led blinks something is wrong with address, checksum, baud rate or parity.

capputeam, 2020-09-28:
both tx and rx led binks, but with every 4 seconds or more.

In the console nothing is reported even after enabling sml debug mode.

11:30:23 WIF: Checking connection...
11:30:40 CMD: sensor53 d1
11:30:40 SRC: WebConsole from 192.168.1.86
11:30:40 CMD: Group 0, Index 53, Command "SENSOR", Data "d1"
11:30:40 MQT: stat/ProvaRS485/RESULT = {"Time":"2020-09-28T11:30:40","SML":{"CMD":"dump: 1"}}
11:30:43 WIF: Checking connection...
11:31:03 WIF: Checking connection...
11:31:23 WIF: Checking connection...
11:31:43 WIF: Checking connection...
11:32:03 WIF: Checking connection...

I'm using this configuration.

[image]

gemu2015, 2020-09-28:
oh that's the bug!

You should NOT define anything in the Tasmota module parameters .

this causes SML to stop directly because of double define of GPIOs

GPIOs are solely defined in the SML script

capputeam, 2020-09-28:
Many thanks, I was missing that,

Removing the configuration, now only the tx led blinks (no answer from the devices)

is there a way to see also the message sent in the console?

gemu2015, 2020-09-28:
no currently no debug for the sent characters

no answer, means you send the wrong codes to the device
looking at the dds2328 driver source shows that you should send
e.g. 010**3**000c not 010**4**000c

however the device normally should send an error code which also would blink the receive led

the baud rate should be correct with 9600 baud is assume

capputeam, 2020-09-30:
Now it works fine, many many thanks for supporting me on that.
[image]

gemu2015, 2020-09-30:
nice to hear!

you may consider publishing this solution in the Tasmota SML example docs.

capputeam, 2020-09-30:
why not, how?

gemu2015, 2020-09-30:
this is the tasmota doc page where you can post your example
https://github.com/tasmota/docs/blob/master/docs/Smart-Meter-Interface.md

or i may do that for you if you post your final descriptor here

capputeam, 2020-10-01:
Here the descripion of my configuration.

**Wiring**
[image]
**Registry table for Hiking DDS238-2 ZN/S**

Register(hex) | Meaning | Scale | Unit | Data format | R/W | Display
-- | -- | -- | -- | -- | -- | --
0000-0001 | total energy | 1/100 | kWh | unsigned dword | R | T00   00000.0
0002-0003 | reserved |   |   | unsigned dword |   |  
0004-0005 | reserved |   |   | unsigned dword |   |  
0006-0007 | reserved |   |   | unsigned dword |   |  
0008-0009 | export energy | 1/100 | kWh | unsigned dword | R | T01   00000.0
000A-000B | import energy | 1/100 | kWh | unsigned dword | R | T02   00000.0
000C | voltage | 1/1000 | V | unsigned word | R | U         000.00
000D | current | 1/100 | A | unsigned word | R | I           000.00
000E | active power | 1 | W | signed word | R | P        00000.0
000F | reactive power | 1 | VAr | unsigned word | R | q        00000.0
0010 | power factor | 1/1000 |   | unsigned word | R | PF             0.00
0011 | frequency | 1/100 | Hz | unsigned word | R | F                 0.00
0012 | reserved |   |   | unsigned word |   |  
0013 | reserved |   |   | unsigned word |   |  
0014 | reserved |   |   | unsigned word |   |  
0015:high | station address 1-247 |   |   | unsigned char | R/W | ID              000
0015:low | baud rate |   | 1-4² | unsigned char | R/W | b              0000

**Script:**
>D
>B
=>sensor53 r
>M 1	  
	  
+1,3,m,1,9600,Hiking,1,10,0103000c,0103000e,0303000c,0303000e,0403000c,0403000e,0503000c,0503000e
_---> two groups of registrers for each device --> default 2 registers returned ---> 4 values per device_ 
1,=h Contatore 1
1,010304UUuu@i0:10,C1_Voltage,V,C1Voltage,1     _---> decoder for the first registry returned for the first group_
1,010304xxxxUUuu@i0:1000,C1_Current,A,C1Current,3       _---> decoder for the second registry returned for the first group_
1,010304SSss@i1:1,C1_ActivePower,W,C1ActivePower,0
1,010304xxxxUUuu@i1:1,C1_ReactivePower,Var,C1ReactivePower,0
1,=h Contatore 3
1,030304UUuu@i2:10,C3_Voltage,V,C3Voltage,1
1,030304xxxxUUuu@i2:1000,C3_Current,A,C3Current,3
1,030304SSss@i3:1,C3_ActivePower,W,C3ActivePower,0
1,030304xxxxUUuu@i3:1,C3_ReactivePower,Var,C3ReactivePower,0
1,=h Contatore 4
1,040304UUuu@i4:10,C4_Voltage,V,C4Voltage,1
1,040304xxxxUUuu@i4:1000,C4_Current,A,C4Current,3
1,040304SSss@i5:1,C4_ActivePower,W,C4ActivePower,0
1,040304xxxxUUuu@i5:1,C4_ReactivePower,Var,C4ReactivePower,0
1,=h Contatore 5
1,050304UUuu@i6:10,C5_Voltage,V,C5Voltage,1
1,050304xxxxUUuu@i6:1000,C5_Current,A,C5Current,3
1,050304SSss@i7:1,C5_ActivePower,W,C5ActivePower,0
1,050304xxxxUUuu@i7:1,C5_ReactivePower,Var,C5ReactivePower,0
#

**Console output (sensor53 d1)**
10:16:03 : 01 03 04 09 1f 00 0e 49 ad    _---> device 01 registry 000c (09 1f) and 000d (00 0e)_ 
10:16:04 : 01 03 04 00 20 00 19 3a 33   _---> device 01 registry 000e (00 20) and 000f (00 19)_
10:16:05 : 03 03 04 09 1a 00 00 fb a8 
10:16:06 : 03 03 04 00 00 00 00 d9 f3 
10:16:07 : 04 03 04 09 21 00 01 3d 65 
10:16:08 : 04 03 04 00 02 00 03 4e f2 
10:16:09 : 05 03 04 09 0b 00 00 cd ad 
10:16:10 : 05 03 04 00 00 00 00 bf f3 
10:16:11 : 01 03 04 09 09 00 09 e9 ab 
10:16:12 : 01 03 04 00 0e 00 0e 1a 34 
10:16:13 : 03 03 04 09 08 00 00 5b ad 
10:16:14 : 03 03 04 00 00 00 00 d9 f3 
10:16:15 : 04 03 04 09 0b 00 01 1c ad 
10:16:16 : 04 03 04 00 02 00 03 4e f2 
10:16:17 : 05 03 04 09 08 00 00 3d ad 
10:16:18 : 05 03 04 00 00 00 00 bf f3 

**Browser page**
[image]

**MQTT** 

11:17:16 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:16","Hiking":{"C1Voltage":235.3}}
11:17:16 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:16","Hiking":{"C1Current":0.024}}
11:17:17 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:17","Hiking":{"C1ActivePower":56.0}}
11:17:17 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:17","Hiking":{"C1ReactivePower":26.0}}
11:17:18 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:18","Hiking":{"C3Voltage":235.3}}
11:17:18 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:18","Hiking":{"C3Current":0.000}}
11:17:19 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:19","Hiking":{"C3ActivePower":0.0}}
11:17:19 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:19","Hiking":{"C3ReactivePower":0.0}}
11:17:20 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:20","Hiking":{"C4Voltage":235.0}}
11:17:20 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:20","Hiking":{"C4Current":0.001}}
11:17:21 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:21","Hiking":{"C4ActivePower":2.0}}
11:17:21 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:21","Hiking":{"C4ReactivePower":3.0}}
11:17:22 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:22","Hiking":{"C5Voltage":235.5}}
11:17:22 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:22","Hiking":{"C5Current":0.000}}
11:17:23 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:23","Hiking":{"C5ActivePower":0.0}}
11:17:23 MQT: tele/ProvaRS485/SENSOR = {"Time":"2020-10-01T11:17:23","Hiking":{"C5ReactivePower":0.0}}

gemu2015, 2020-10-01:
ok, thanks

added to docs