arendst/Tasmota #9332: Missing battery value for the Sonoff Zigbee motion sensor

jantielens, 2020-09-17:
### PROBLEM DESCRIPTION
I have a Sonoff Zigbee motion sensor ([SONOFF SNZB-03](https://www.itead.cc/sonoff-snzb-03-zigbee-motion-sensor.html)). The sensor is successfully paired and sends motion alerts on MQTT, but it doesn't report battery information. 

Raw messages:
```
ESP-ZIG: {"ZbEZSPReceived":"4500000401000501FF0001000044A0C41516FFFF0919400001000000000002"}

ESP-MQT: tele/sonoffbridge2/SENSOR = {"ZbReceived":{"0x1615":{"Device":"0x1615","Name":"somtionboven","0500<00":"010000000000","ZoneStatusChange":1,"Endpoint":1,"LinkQuality":71}}} (retained)
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge & Sonoff Zigbee Motion sensor
- [X] Tasmota binary firmware version number used: 8.5.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
13:04:17 MQT: stat/sonoffbridge2/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
13:04:17 MQT: stat/sonoffbridge2/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
13:04:18 MQT: stat/sonoffbridge2/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"165":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"166":"Zigbee Rx"},"GPIO4":{"215":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"158":"LedLinki"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"17":"Button1"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
none

- [X] Provide the output of this command: `Status 0`:
```
13:10:58 MQT: stat/sonoffbridge2/STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"sonoffbridge2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0}}
13:10:58 MQT: stat/sonoffbridge2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"1T17:05:16","StartupUTC":"2020-09-15T19:05:42","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2020-09-15T12:54:11","SaveCount":22,"SaveAddress":"F8000"}}
13:10:58 MQT: stat/sonoffbridge2/STATUS2 = {"StatusFWR":{"Version":"8.5.0(zbbridge)","BuildDateTime":"2020-09-09T11:44:49","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"364/699"}}
13:10:58 MQT: stat/sonoffbridge2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"192.168.1.10","LogPort":8514,"SSId":["jnfl",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008209","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
13:10:58 MQT: stat/sonoffbridge2/STATUS4 = {"StatusMEM":{"ProgramSize":528,"Free":472,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F100786","04500001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,21,23,41","Sensors":""}}
13:10:58 MQT: stat/sonoffbridge2/STATUS5 = {"StatusNET":{"Hostname":"sonoffzbridge2","IPAddress":"192.168.1.84","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"A4:CF:12:DB:FB:D5","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:10:58 MQT: stat/sonoffbridge2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.30","MqttPort":1883,"MqttClientMask":"sonoffbridge2","MqttClient":"sonoffbridge2","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:10:58 MQT: stat/sonoffbridge2/STATUS7 = {"StatusTIM":{"UTC":"2020-09-17T12:10:58","Local":"2020-09-17T13:10:58","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:30","Sunset":"18:58"}}
13:10:58 MQT: stat/sonoffbridge2/STATUS10 = {"StatusSNS":{"Time":"2020-09-17T13:10:58"}}
13:10:58 MQT: stat/sonoffbridge2/STATUS11 = {"StatusSTS":{"Time":"2020-09-17T13:10:58","Uptime":"1T17:05:16","UptimeSec":147916,"Vcc":3.475,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"jnfl","BSSId":"42:37:86:0F:AF:72","Channel":5,"RSSI":100,"Signal":-20,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
na
```

### TO REPRODUCE
- Install Tasmota to the Sonoff Zigbee Bridge
- Configure MQTT
- Pair the Tasmota Motion sensor
- Observe the MQTT messages

### EXPECTED BEHAVIOUR
The MQTT message should contain BatteryPercentage as an attribute.
```
ESP-ZIG: {"ZbEZSPReceived":"4500000401000501FF0001000044A0C41516FFFF0919400001000000000002"}

ESP-MQT: tele/sonoffbridge2/SENSOR = {"ZbReceived":{"0x1615":{"Device":"0x1615","Name":"somtionboven","0500<00":"010000000000","ZoneStatusChange":1,"Endpoint":1,"LinkQuality":71, "BatteryPercentage":100}}} (retained)
```

### SCREENSHOTS
na

### ADDITIONAL CONTEXT
na

schrej-zz, 2020-09-18:
Hi,
have you been patient enough?
Device sends battery status only once per day.

Jason2866, 2020-09-18:
Not all sonoff zigbee sensors do send battery infos

coldfire84, 2020-09-22:
I also have this issue, from issue #9216 (specifically [this ](https://github.com/arendst/Tasmota/issues/9216#issuecomment-692322535)comment) it would appear that the battery status _is_ sent, but I too cannot get this to display with the latest dev compiled build (timestamp 20200922 13:17) and a re-pairing of the motion sensor.

[image]

Have had the sensor paired for a number of weeks with various Tasmota releases tested, only change has been to the GUI in that the Link Quality is now a icon/ scale (used to be a number) plus the last time since trigger is shown (new as of a couple of days ago).

ascillato2, 2020-09-24:
@jantielens 

Hi, any news on this?

s-hadinger, 2020-09-25:
It is still unclear which Sonoff devices report or don't report battery status. There are maybe different firmwares out there.

There is an easy workaround for now, create a rule that is triggered when you receive a message from those devices, and explicitly ask for BatteryLevel before it gets back to sleep.

For example, getting Sonoff SNZB-02:
`Rule2 on ?#temperature do zbsend {\"Device\":\"%ZBDEVICE%\",\"Read\":{\"BatteryPercentage\":true}} endon`
`Rule2 1`

ascillato2, 2020-09-25:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-25 by ascillato2

coldfire84, 2020-09-25:
> It is still unclear which Sonoff devices report or don't report battery status. There are maybe different firmwares out there.
> 
> There is an easy workaround for now, create a rule that is triggered when you receive a message from those devices, and explicitly ask for BatteryLevel before it gets back to sleep.
> 
> For example, getting Sonoff SNZB-02:
> `Rule2 on ?#temperature do zbsend {"Device":"%ZBDEVICE%","Read":{"BatteryPercentage":true}} endon`
> `Rule2 1`
> 
> The above rule will request battery percentage for any device reporting temperature.

Thanks @s-hadinger , I've tried as suggested and whilst I get a response I don't think it is being interpreted/ translated:

```
14:44:41 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"DS_Bathroom_Motion":{"Device":"0x84EC","Name":"DS_Bathroom_Motion","0500<00":"010000000000","ZoneStatusChange":1,"Occupancy":1,"Endpoint":1,"LinkQuality":45}}}
14:44:41 RUL: ZBRECEIVED#DS_BATHROOM_MOTION#OCCUPANCY=1 performs "zbsend {"Device":"0x84EC","Read":{"BatteryPercentage":true}}"
14:44:41 SRC: Rule
14:44:41 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x84EC","Read":{"BatteryPercentage":true}}"
14:44:41 ZIG: guessing endpoint 1
14:44:41 ZIG: ZbEZSPSend 340000EC84040101000101400100000A0105000A002100
14:44:41 MQT: stat/zbbridge-1/RESULT = {"ZbSend":"Done"}
14:44:41 ZIG: {"ZbEZSPReceived":"340000E7"}
14:45:09 ZIG: {"ZbEZSPReceived":"3F0000EC8404010100010140000000E7016600"}
14:46:01 ZIG: {"ZbEZSPReceived":"4500000401000501FF00010000E180BCEC84FFFF0919D70000000000000002"}
14:46:01 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0x84EC","srcendpoint":1,"dstendpoint":255,"wasbroadcast":0,"LinkQuality":50,"securityuse":0,"seqnumber":225,"fc":"0x19","manuf":"0x0000","transact":215,"cmdid":"0x00","payload":"000000000000"}}
14:46:01 ZIG: ZbZCLRawReceived: {"0x84EC":{"0500<00":"000000000000","ZoneStatusChange":0,"0406/0000":0,"Endpoint":1,"LinkQuality":50}}
14:46:01 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"DS_Bathroom_Motion":{"Device":"0x84EC","Name":"DS_Bathroom_Motion","0500<00":"000000000000","ZoneStatusChange":0,"Occupancy":0,"Endpoint":1,"LinkQuality":50}}}
```

Incidentally, the Rule I have configured:

```
Rule2 on zbreceived#DS_Bathroom_Motion#Occupancy=1 do zbsend {"Device":"0x84EC","Read":{"BatteryPercentage":true}} endon
Rule2 1
```

duczz, 2020-12-22:
is a solution still being worked on or will this workaround be retained?
My SNZB-02 does only not send the Battery informations to the mqtt broker

i think aqara is the better choice :)

s-hadinger, 2020-12-23:
I don't understand why the device does not send the battery level. My device regularly reports it's battery percentage without needing explicit reads.

coldfire84, 2020-12-23:
I tested my SNZB-02 with zigbee2mqtt and the battery status is shown, so I believe it is being sent.

duczz, 2020-12-23:
@coldfire84 he's showing in tastmota webgui yes, but not sent it to the mqtt-broker
@s-hadinger i send later a paring log.

s-hadinger, 2020-12-23:
If it is on the webui it means it was sent via MQTT. Maybe a long time ago.

duczz, 2020-12-24:
@s-hadinger here is the log
`14:16:05 ZIG: {"ZbEZSPReceived":"23000000D16B5195D51C004B120004"}
14:16:05 ZIG: {"ZbEZSPReceived":"2400D16B5195D51C004B12000203FFFF"}
14:16:06 ZIG: {"ZbEZSPReceived":"23000001D16B5195D51C004B120004"}
14:16:07 ZIG: {"ZbEZSPReceived":"2400D16B5195D51C004B120001000000"}
14:16:07 MQT: tele/tasmota_05DF17/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x00124B001CD59551","ShortAddr":"0x6BD1","ParentNetwork":"0x0000","Status":1,"Decision":0}}
14:16:08 MQT: tele/tasmota_05DF17/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x00124B001CD59551","ShortAddr":"0x6BD1","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
14:16:09 MQT: tele/tasmota_05DF17/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
14:16:09 ZIG: Zigbee Devices Data saved in EEPROM (18 bytes)
14:16:09 MQT: tele/tasmota_05DF17/6BD1/SENSOR = {"ZbReceived":{"0x6BD1":{"Device":"0x6BD1","Manufacturer":"eWeLink","ModelId":"TH01","Endpoint":1,"LinkQuality":18}}}
14:16:11 MQT: tele/tasmota_05DF17/RESULT = {"ZbState":{"Status":33,"Device":"0x6BD1","Endpoint":"0x01","ProfileId":"0x0104","DeviceId":"0x0302","DeviceVersion":0,"InClusters":["0x0000","0x0003","0x0402","0x0405","0x0001"],"OutClusters":["0x0003"]}}
14:16:12 ZIG: Zigbee Devices Data saved in EEPROM (29 bytes)
14:16:13 ZIG: auto-bind `ZbBind {"Device":"0x6BD1","Endpoint":1,"Cluster":"0x0001"}`
14:16:14 MQT: tele/tasmota_05DF17/RESULT = {"ZbBind":{"Device":"0x6BD1","Status":0,"StatusMessage":"SUCCESS"}}
14:16:15 ZIG: auto-bind `ZbBind {"Device":"0x6BD1","Endpoint":1,"Cluster":"0x0402"}`
14:16:15 MQT: tele/tasmota_05DF17/RESULT = {"ZbBind":{"Device":"0x6BD1","Status":0,"StatusMessage":"SUCCESS"}}
14:16:17 ZIG: auto-bind `ZbBind {"Device":"0x6BD1","Endpoint":1,"Cluster":"0x0405"}`
14:16:18 MQT: tele/tasmota_05DF17/RESULT = {"ZbBind":{"Device":"0x6BD1","Status":0,"StatusMessage":"SUCCESS"}}
14:16:19 ZIG: auto-bind `ZbSend {"Device":"0x6BD1","Config":{"BatteryVoltage":{"MinInterval":3600,"MaxInterval":14400,"ReportableChange":0.1},"BatteryPercentage":{"MinInterval":3600,"MaxInterval":14400,"ReportableChange":1}}}`
14:16:20 MQT: tele/tasmota_05DF17/6BD1/SENSOR = {"ZbReceived":{"0x6BD1":{"Device":"0x6BD1","ConfigResponse":{},"Endpoint":1,"LinkQuality":29}}}
14:16:21 ZIG: auto-bind `ZbSend {"Device":"0x6BD1","Config":{"Temperature":{"MinInterval":30,"MaxInterval":3600,"ReportableChange":0.2}}}`
14:16:22 MQT: tele/tasmota_05DF17/6BD1/SENSOR = {"ZbReceived":{"0x6BD1":{"Device":"0x6BD1","ConfigResponse":{},"Endpoint":1,"LinkQuality":26}}}
14:16:23 ZIG: auto-bind `ZbSend {"Device":"0x6BD1","Config":{"Humidity":{"MinInterval":30,"MaxInterval":3600,"ReportableChange":1}}}`
14:16:24 MQT: tele/tasmota_05DF17/6BD1/SENSOR = {"ZbReceived":{"0x6BD1":{"Device":"0x6BD1","ConfigResponse":{},"Endpoint":1,"LinkQuality":31}}}
14:16:29 MQT: tele/tasmota_05DF17/6BD1/SENSOR = {"ZbReceived":{"0x6BD1":{"Device":"0x6BD1","Temperature":23.83,"Humidity":48.21,"Endpoint":1,"LinkQuality":26}}}`

batoria, 2020-12-30:
I had same problem, but I've found a working solution.
After pairing I used the following command:

ZbSend {"Device":"0xE5D3","Config":{"BatteryPercentage":{"MinInterval":600,"MaxInterval":3600,"ReportableChange":1}}}

duczz, 2021-01-16:
nope, does not work for me, get no mqtt value from the battery, in the webinterface it exists

duczz, 2021-01-24:
i found my problem with the battery, it is a Adapter problem from iobroker.sonoff