arendst/Tasmota #9334: LWT is posted different depending on online/offline

cconde, 2020-09-17:
### PROBLEM DESCRIPTION

When device comes online MQTT is posted to /tele/MYDEVICE/LWT

But when device goes offline, then it's posted to tele/tasmota_FFFFFF/LWT

I don't know how to fix this.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff S20
- [X] Tasmota binary firmware version number used: 8.5.0(sensors)
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
22:18:49 MQT: stat/MYDEVICE/RESULT = {"NAME":"MY-DEVICE","GPIO":[17,0,255,255,255,0,0,0,21,56,255,0,0],"FLAG":0,"BASE":8}
22:18:49 MQT: stat/MYDEVICE/RESULT = {"Module":{"8":"Sonoff S2X"}}
22:18:49 MQT: stat/MYDEVICE/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"4":"DS18x20"},"GPIO2":{"0":"None"},"GPIO3":{"4":"DS18x20"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}

Sorry, trying to search by LWT is hard because any issue that copies mqtt logs tend to pop up as unwanted result...

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

No rules used

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

22:20:50 MQT: stat/MYDEVICE/STATUS = {"Status":{"Module":8,"DeviceName":"MY-DEVICE","FriendlyName":["S20 Luz Nocturna"],"Topic":"MYDEVICE","ButtonTopic":"","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[5,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:20:50 MQT: stat/MYDEVICE/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"2T07:39:02","StartupUTC":"2020-09-15T13:41:48","Sleep":50,"CfgHolder":4617,"BootCount":28,"BCResetTime":"2020-09-15T09:10:07","SaveCount":118,"SaveAddress":"F6000"}}
22:20:50 MQT: stat/MYDEVICE/STATUS2 = {"StatusFWR":{"Version":"8.5.0(sensors)","BuildDateTime":"2020-09-09T11:42:52","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"402/699"}}
22:20:50 MQT: stat/MYDEVICE/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ARGOS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","0305C8000100060000005A00000000000000","01000001","00006000","00000000"]}}
22:20:50 MQT: stat/MYDEVICE/STATUS4 = {"StatusMEM":{"ProgramSize":627,"Free":376,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCD","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
22:20:50 MQT: stat/MYDEVICE/STATUS5 = {"StatusNET":{"Hostname":"MY-DEVICE","IPAddress":"192.168.14.161","Gateway":"192.168.14.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.14.1","Mac":"DC:4F:22:11:46:9C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:20:50 MQT: stat/MYDEVICE/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.14.1","MqttPort":1883,"MqttClientMask":"MYDEVICE","MqttClient":"MYDEVICE","MqttUser":"MYDEVICE","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:20:50 MQT: stat/MYDEVICE/STATUS7 = {"StatusTIM":{"UTC":"2020-09-17T21:20:50","Local":"2020-09-17T22:20:50","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:30","Sunset":"18:58"}}
22:20:50 MQT: stat/MYDEVICE/STATUS10 = {"StatusSNS":{"Time":"2020-09-17T22:20:50","DS18B20":{"Id":"01162E3605EE","Temperature":34.3},"TempUnit":"C"}}
22:20:50 MQT: stat/MYDEVICE/STATUS11 = {"StatusSTS":{"Time":"2020-09-17T22:20:50","Uptime":"2T07:39:02","UptimeSec":200342,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"ARGOS","BSSId":"F0:B4:29:58:DB:1D","Channel":11,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

No console output as it's when loses power

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Upgrade OTA to sensors firmware, configure, check LWT Online published, disconnect

LWT Offline is published to a wrong topic

BTW, using standard mosquitto browser

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Same LWT topic for offline and online

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-09-18:
Please, can you post the logs from your MQTT broker? Thanks.

ascillato2, 2020-09-20:
Please, when you have time, please post the logs. Thanks

closed 2020-09-20 by ascillato2