arendst/Tasmota #9335: Remote Configuration build

tichris0, 2020-09-18:
**Have you looked for this feature in other issues and in the docs?**  
Yup

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

The Sonoff only has 1M memory chip and Tasmota maxes that out on some builds

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

Given that most of the Sonoffs are connected to the internet and that the configuration code is only used very rarely (if not only once), it seems logical to move most of the web/JS code from the firmware to offline.  That would enable Tasmota to have a much more complex and richer UI without compromising the size of the build.  It would also enable Tasmota to fix some bugs without a firmware update.  In addition, it would make the code a lot more readable.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Keeping things as-is, but it has the problems mentioned above.

**Additional context**  
_Add any other context or screenshots about the feature request here._

If there's an appetite for this; I'd would be happy to investigate and slowly migrate the code over to such a format.  I would expect a first pass to keep both paths working and to add an additional build version of Tasmota that supports online configuration.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-18:
Thx for your idea. The goal and design of Tasmota is NOT to use the Interrnet. So this will not be done from the project owner and the crew contributing code to Tasmota. If you want such a feature you have to fork.

ascillato2, 2020-09-18:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-18 by ascillato2