arendst/Tasmota #9340: Spontaneous reboot shutter

dr3men, 2020-09-18:
### PROBLEM DESCRIPTION
I already wrote about the problem with Shater, when he starts giving an endless command:
https://github.com/arendst/Tasmota/issues/9310#issuecomment-693249119
I have updated to the latest firmware with full uninstall. The problem I described was solved, BUT now I started to have another problem.
I send the tent up (Close) and down () several times and, for example, the third time the ESP reboots itself.
First, it gives a pulse that does not correspond to the set speed and after a second I reboot

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): **Wemos D1mini + A4988 motor driver + nema17**
- [x ] Tasmota binary firmware version number used: **8.5.0.1(tasmota)**
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: **NodeMCU PyFlasher**
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
Backlog PulseTime1 0; PulseTime2 0
Interlock OFF
SetOption80 1
ShutterRelay1 1
ShutterFrequency 15000
ShutterMotorDelay1 1.0
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
17:48:58 CMD: Group 0, Index 1, Command "STATUS", Data "0"
17:48:58 MQT: stat/blinde/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3"],"Topic":"blinde","ButtonTopic":"0","Power":4,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:48:58 MQT: stat/blinde/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:16:38","StartupUTC":"2020-09-18T16:32:20","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2020-09-18T16:58:43","SaveCount":89,"SaveAddress":"FB000"}}
17:48:58 MQT: stat/blinde/STATUS2 = {"StatusFWR":{"Version":"8.5.0.1(tasmota)","BuildDateTime":"2020-09-18T07:23:42","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"329/699"}}
17:48:58 MQT: stat/blinde/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AuthMe",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","40000000","00006000","00000000"]}}
17:48:58 MQT: stat/blinde/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":2,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
17:48:58 MQT: stat/blinde/STATUS5 = {"StatusNET":{"Hostname":"blinde-6833","IPAddress":"192.168.1.66","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"8C:AA:B5:1B:9A:B1","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:48:58 MQT: stat/blinde/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.59","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1B9AB1","MqttUser":"dr3men","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:48:58 MQT: stat/blinde/STATUS7 = {"StatusTIM":{"UTC":"2020-09-18T16:48:58","Local":"2020-09-18T17:48:58","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:31","Sunset":"18:56"}}
17:48:58 MQT: stat/blinde/STATUS10 = {"StatusSNS":{"Time":"2020-09-18T17:48:58","COUNTER":{"C1":1215168},"Shutter1":{"Position":100,"Direction":0,"Target":100}}}
17:48:58 MQT: stat/blinde/STATUS11 = {"StatusSTS":{"Time":"2020-09-18T17:48:58","Uptime":"0T00:16:38","UptimeSec":998,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"AuthMe","BSSId":"50:FF:20:22:1E:54","Channel":9,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:03"}}}
17:48:58 MQT: stat/blinde/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":810,"Close":810,"50perc":50,"Delay":20,"Opt":"0000","Calib":"300:500:700:900:1000","Mode":"0"}}

```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
18:05:27 SHT: time: 214, toBeAcc 0, current_stop_way -9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1407000, next_stop 1397500, target: 0,  velocity_change_per_step_max 750
18:05:27 SHT: time: 215, toBeAcc 0, current_stop_way -9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1406000, next_stop 1396500, target: 0,  velocity_change_per_step_max 750
18:05:27 SHT: time: 217, toBeAcc 0, current_stop_way -9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1404000, next_stop 1394500, target: 0,  velocity_change_per_step_max 750
18:05:27 SHT: time: 217, toBeAcc 0, current_stop_way -9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1404000, next_stop 1394500, target: 0,  velocity_change_per_step_max 750
18:05:27 SHT: time: 218, toBeAcc 0, current_stop_way -9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1403000, next_stop 1393500, target: 0,  velocity_change_per_step_max 750
18:05:27 SHT: Shutter 1 Real 1403000, Start 1620000, Stop 0, Dir -1, Delay 20, Rtc 10.90 [s], Freq 15000, PWM 0
18:05:27 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":87,"Direction":-1,"Target":0}}
18:05:27 SHT: time: 219, toBeAcc 0, current_stop_way -9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1402000, next_stop 1392500, target: 0,  velocity_change_per_step_max 750
18:05:27 SHT: time: 220, toBeAcc 0, current_stop_way -9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1401000, next_stop 1391500, target: 0,  velocity_change_per_step_max 750
18:05:27 SHT: time: 221, toBeAcc 0, current_stop_way -9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1400000, next_stop 1390500, target: 0,  velocity_change_per_step_max 750
18:05:27 SHT: time: 222, toBeAcc 0, current_stop_way -9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1399000, next_stop 1389500, target: 0,  velocity_change_per_step_max 750
18:05:27 SHT: time: 227, toBeAcc 0, current_stop_way -9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1394000, next_stop 1384500, target: 0,  velocity_change_per_step_max 750
18:05:27 SHT: time: 228, toBeAcc 0, current_stop_way -9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1393000, next_stop 1383500, target: 0,  velocity_change_per_step_max 750
[... 74 lines cut ...]
18:13:23 SHT: Shutter 1 Real 1553000, Start 1620000, Stop 0, Dir -1, Delay 20, Rtc 3.40 [s], Freq 15000, PWM 0
18:13:23 MQT: stat/blinde/RESULT = {"Shutter1":{"Position":96,"Direction":-1,"Target":0}}
18:13:46 SHT: time: 68, toBeAcc 0, current_stop_way 9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1272000, next_stop 1281500, target: 1620000,  velocity_change_per_step_max 750
18:13:46 SHT: time: 68, toBeAcc 0, current_stop_way 9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1272000, next_stop 1281500, target: 1620000,  velocity_change_per_step_max 750
[... cut ...]
18:13:47 SHT: time: 74, toBeAcc 0, current_stop_way 9500,vel_vur 15000, vel_max 15000, act_vel_change 0, min_runtime_ms 1000, act.pos 1278000, next_stop 1287500, target: 1620000,  velocity_change_per_step_max 750
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

TrDA-hab, 2020-09-18:
The first thing I see is that the speed of the stepper motor should be optimally in the range of 400 - 4500 Hertz (1 Hertz == 1 step).

p.s.
1. ShutterFrequency 15000 - just kills the stepper motor!
2. Huge self-induction drives the stepper motor controller crazy.
3. If you need high speed you must use an overdrive gearbox and this is no option.

TrDA-hab, 2020-09-18:
Please test the operation of the stepper motor using ShutterFrequency 2500. And tell us if there is an error.

TrDA-hab, 2020-09-19:
I will check your information and use 8.5.0.1. Just wait 1 day.

stefanbode, 2020-09-20:
Hi friends, yes 15000 hz is quite a lot. Anyway I see a bug.high frequency and/or like long runtime cause sometime to hit the int32 bit calculation range. I can see after the stop thatcher remaining steps are totally wrong. Before this everything is fine. This also might responsible for the reboot. Thanks for reporting.

stefanbode, 2020-09-20:
I checked my old documentation and even with 24v on the driver it was impossible to get higher then 8000hz. Even with the softest ramp. Now I see a one second ramp an 15khz. Just wondering about this super stepper motor.

dr3men, 2020-09-20:
Previously, I had a half step 1/16 when it worked at 20,000. The motor and driver did not warm up.
Thanks for the recommendations. I reduced the frequency to 2000 and increased the step to 1/2. The reboot problems I previously described were gone, but the following appeared:
- when controlled from the web interface, it works fine for some time, but for about the tenth time the device gives incorrect impulses to the driver (the motor starts to make noise, but does not spin), the firmware and config hang and reset. When turned off and on, the motor makes noise again. At the same time, Wemos is seen in the network as an access point in the network: ESP-1ZB23.
- when controlled through MQTT, this happens immediately.

good: https://youtu.be/7UgXSU9yFt0
bad: https://youtu.be/AM2KIDrtB4M

Program Version | 8.5.0.1(tasmota)
-- | --
2020-09-19T14:22:41
2_7_4_1/2.2.2-dev(38a443e)

lvl 4 log is not written, because the device freezes.
I reflash Wemos again, set up, it resumes work.

TrDA-hab, 2020-09-20:
I tested 8.5.0.1 and 8.2.0 The behavior is the same.

1. If you do not wait for the end of the command execution and give a new command, then after 3-7 commands the ESP freezes (buffer overflow is possible).
2. If you wait for the end of the command execution and give a new command after 1 second, then everything works fine.

sergios100, 2020-09-20:
I think I have a similar issue:

I start the shutter by pressing the "up" arrow button in the Webinterface, Stepper-Motor starts to move as expected. To stop the upwards movement I press once again the "up" arrow button, Stepper-Motor starts to ramp down. So far so good. But instead of staying still there it starts to move move again. According to the weblog 4 output, somethings seems to go wrong.

Here is my setup:

[x ] Provide the output of command: Backlog Template; Module; GPIO 255:

`13:05:56 SRC: Serial
13:05:56 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
13:05:56 SRC: Backlog
13:05:56 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
13:05:56 RSL: stat/tasmota_FCB308/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
13:05:56 SRC: Backlog
13:05:56 CMD: Group 0, Index 1, Command "MODULE", Data ""
13:05:56 RSL: stat/tasmota_FCB308/RESULT = {"Module":{"1":"ESP32-DevKit"}}
13:05:56 SRC: Backlog
13:05:56 CMD: Group 0, Index 1, Command "GPIO", Data "255"
13:05:56 RSL: stat/tasmota_FCB308/RESULT = {"GPIO0":{"416":"PWM1"},"GPIO1":{"0":"None"},"GPIO2":{"352":"Counter1"},"GPIO3":{"0":"None"},"GPIO4":{"225":"Relay2"},"GPIO5":{"256":"Relay_i1"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO39":{"0":"None"}}
`

Backlog Rule1; Rule2; Rule3

`13:16:31 CMD: Backlog Rule1; Rule2; Rule3
13:16:31 SRC: Serial
13:16:31 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
13:16:32 SRC: Backlog
13:16:32 CMD: Group 0, Index 1, Command "RULE", Data ""
13:16:32 RSL: stat/tasmota_FCB308/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
13:16:32 SRC: Backlog
13:16:32 CMD: Group 0, Index 2, Command "RULE", Data ""
13:16:32 RSL: stat/tasmota_FCB308/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
13:16:32 SRC: Backlog
13:16:32 CMD: Group 0, Index 3, Command "RULE", Data ""
13:16:32 RSL: stat/tasmota_FCB308/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
`

[x ] Provide the output of this command: Status 0:
`
13:07:46 SRC: Serial
13:07:46 CMD: Group 0, Index 1, Command "STATUS", Data "0"
13:07:46 RSL: stat/tasmota_FCB308/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3"],"Topic":"tasmota_FCB308","ButtonTopic":"0","Power":4,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:07:46 RSL: stat/tasmota_FCB308/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin.gz","RestartReason":"Vbat power on reset","Uptime":"0T00:08:54","StartupUTC":"2020-09-20T11:58:52","Sleep":50,"CfgHolder":4617,"BootCount":50,"BCResetTime":"2020-09-14T11:50:34","SaveCount":359}}
13:07:46 RSL: stat/tasmota_FCB308/STATUS2 = {"StatusFWR":{"Version":"8.5.0.1(tasmota)","BuildDateTime":"2020-09-18T07:24:00","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","CpuFrequency":80,"Hardware":"ESP32","CR":"385/699"}}
13:07:46 RSL: stat/tasmota_FCB308/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FaryUltra",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","40000000","00006000","00000000"]}}
13:07:46 RSL: stat/tasmota_FCB308/STATUS4 = {"StatusMEM":{"ProgramSize":1049,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","00140001","000000CF","010013C0","40000981","00004000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29","Sensors":"1,2,3,5,6"}}
13:07:46 RSL: stat/tasmota_FCB308/STATUS5 = {"StatusNET":{"Hostname":"tasmota_FCB308-4872","IPAddress":"192.168.1.121","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"24:62:AB:FC:B3:08","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:07:46 RSL: stat/tasmota_FCB308/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FCB308","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:07:46 RSL: stat/tasmota_FCB308/STATUS7 = {"StatusTIM":{"UTC":"2020-09-20T12:07:46","Local":"2020-09-20T13:07:46","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:34","Sunset":"18:51"}}
13:07:46 RSL: stat/tasmota_FCB308/STATUS10 = {"StatusSNS":{"Time":"2020-09-20T13:07:46","COUNTER":{"C1":5057},"Shutter1":{"Position":8,"Direction":0,"Target":8}}}
13:07:46 RSL: stat/tasmota_FCB308/STATUS11 = {"StatusSTS":{"Time":"2020-09-20T13:07:46","Uptime":"0T00:08:54","UptimeSec":534,"Heap":111,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":" [... cut ...]
13:03:18 SHT: Shutter 1 Real 16000, Start 16000, Stop 16000, Dir 0, Delay 20, Rtc 4.30 [s], Freq 0, PWM 0
13:03:18 RSL: stat/tasmota_FCB308/SHUTTER1 = 8
13:03:18 RSL: stat/tasmota_FCB308/RESULT = {"Shutter1":{"Position":8,"Direction":0,"Target":8}}
13:03:19 CFG: Saved, Count 359, Bytes 4096
13:03:23 WIF: Checking connection...
`

Uups, my posted code looks like a big mess. I tried to attach it as a text file:
[Shutter-Issues.txt](https://github.com/arendst/Tasmota/files/5251656/Shutter-Issues.txt)

stefanbode, 2020-09-22:
Hi gents. Thanks for sharing. I will take a look this evening. So far I have not seen this but as always the customer is right. Let me try to reproduce. Look like a bug

stefanbode, 2020-09-23:
Ok. I am able to reproduce the error. The stop function needs rework for the stepper motors. I still have now an exception after the stop command without any idea why and were. Working......

TrDA-hab, 2020-09-23:
This is good news. I'm ready to run new tests.

sergios100, 2020-09-23:
Hi Stefan,

I did some further investigation. Besides the Issue that we have on the Stepper stopfunction, there seems to be (the same?) issue also in Shutter mode 1. I took a freshly flashed ESP8266, set 
`Backlog Template; Module; GPIO 255`

`10:47:33 RSL: stat/tasmota_C2C9BC/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18} `

`10:47:34 RSL: stat/tasmota_C2C9BC/RESULT = {"Module":{"18":"Generic"}} `

`10:47:34 RSL: stat/tasmota_C2C9BC/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"22":"Relay2"},"GPIO5":{"21":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}`

_in short: Relay1 = GPIO5 and Relay2 = GPIO4_

`SetOption80 1` 
and `Interlock 1,2` 
as well as `Interlock ON`
so pretty stock. 

Now when I press the down button on the Webinterface, shutter starts to move downwards and if I press the down button on the Web Interface once more, it stops, as expected, see Log here:

10:57:46 SRC: WebGui from 192.168.1.120
10:57:46 SRC: WebGui from 192.168.1.120
10:57:46 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data "StopClose"
10:57:46 SRC: Shutter
10:57:46 RSL: stat/tasmota_C2C9BC/RESULT = {"POWER2":"ON"}
10:57:46 RSL: stat/tasmota_C2C9BC/POWER2 = ON
10:57:46 RSL: stat/tasmota_C2C9BC/RESULT = {"ShutterPosition1":0}
10:57:46 SHT: Shutter 1 Real 101000, Start 101000, Stop 0, Dir -1, Delay 0, Rtc 0.05 [s], Freq 0, PWM 0
10:57:46 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":51,"Direction":-1,"Target":0}}
10:57:46 CFG: Saved to flash at FB, Count 41, Bytes 4096
10:57:47 SHT: Shutter 1 Real 85000, Start 101000, Stop 0, Dir -1, Delay 0, Rtc 0.85 [s], Freq 0, PWM 0
10:57:47 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":43,"Direction":-1,"Target":0}}
10:57:48 SHT: Shutter 1 Real 65000, Start 101000, Stop 0, Dir -1, Delay 0, Rtc 1.85 [s], Freq 0, PWM 0
10:57:48 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":33,"Direction":-1,"Target":0}}
10:57:49 SHT: Shutter 1 Real 45000, Start 101000, Stop 0, Dir -1, Delay 0, Rtc 2.85 [s], Freq 0, PWM 0
10:57:49 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":23,"Direction":-1,"Target":0}}

**Here the "down button" is pressed again:**

10:57:50 SRC: WebGui from 192.168.1.120
10:57:50 SRC: WebGui from 192.168.1.120
10:57:50 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data "StopClose"
10:57:50 SHT: Stop moving 1: dir: -1
10:57:50 RSL: stat/tasmota_C2C9BC/RESULT = {"ShutterPosition1":9}
10:57:50 SHT: Shutter 1 Real 25000, Start 101000, Stop 18000, Dir -1, Delay 0, Rtc 3.85 [s], Freq 0, PWM 0
10:57:50 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":13,"Direction":-1,"Target":9}}
10:57:50 SHT: Stop Shutter 0 ..
10:57:51 SRC: Shutter
10:57:51 RSL: stat/tasmota_C2C9BC/RESULT = {"POWER2":"OFF"}
10:57:51 RSL: stat/tasmota_C2C9BC/POWER2 = OFF
10:57:51 SHT: Shutter 1 Real 18000, Start 18000, Stop 18000, Dir 0, Delay 0, Rtc 4.20 [s], Freq 0, PWM 0
10:57:51 RSL: stat/tasmota_C2C9BC/SHUTTER1 = 9
10:57:51 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":9,"Direction":0,"Target":9}}
10:57:51 CFG: Saved to flash at FA, Count 42, Bytes 4096

**Shutter stops as expected**

Lets do it now with the "up button":

11:01:45 SRC: WebGui from 192.168.1.120
11:01:45 SRC: WebGui from 192.168.1.120
11:01:45 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data "StopOpen"
11:01:45 SRC: Shutter
11:01:45 RSL: stat/tasmota_C2C9BC/RESULT = {"POWER1":"ON"}
11:01:45 RSL: stat/tasmota_C2C9BC/POWER1 = ON
11:01:45 RSL: stat/tasmota_C2C9BC/RESULT = {"ShutterPosition1":100}
11:01:45 SHT: Shutter 1 Real 101000, Start 100000, Stop 200000, Dir 1, Delay 0, Rtc 0.10 [s], Freq 0, PWM 0
11:01:45 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":51,"Direction":1,"Target":100}}
11:01:45 SHT: Shutter 1 Real 106000, Start 100000, Stop 200000, Dir 1, Delay 0, Rtc 0.35 [s], Freq 0, PWM 0
11:01:45 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":53,"Direction":1,"Target":100}}
[... 18 lines cut ...]
11:01:48 SRC: Shutter
11:01:48 RSL: stat/tasmota_C2C9BC/RESULT = {"POWER2":"ON"}
11:01:48 RSL: stat/tasmota_C2C9BC/POWER2 = ON
11:01:48 RSL: stat/tasmota_C2C9BC/RESULT = {"ShutterPosition1":54}
11:01:49 SHT: Shutter 1 Real 161000, Start 166000, Stop 108000, Dir -1, Delay 0, Rtc 0.30 [s], Freq 0, PWM 0
11:01:49 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":81,"Direction":-1,"Target":54}}
11:01:49 SHT: Shutter 1 Real 161000, Start 166000, Stop 108000, Dir -1, Delay 0, Rtc 0.30 [s], Freq 0, PWM 0
11:01:49 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":81,"Direction":-1,"Target":54}}
11:01:49 SHT: Shutter 1 Real 157000, Start 166000, Stop 108000, Dir -1, Delay 0, Rtc 0.50 [s], Freq 0, PWM 0
11:01:49 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":79,"Direction":-1,"Target":54}}
11:01:49 CFG: Saved to flash at F6, Count 46, Bytes 4096
11:01:50 SHT: Shutter 1 Real 137000, Start 166000, Stop 108000, Dir -1, Delay 0, Rtc 1.50 [s], Freq 0, PWM 0
11:01:50 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":69,"Direction":-1,"Target":54}}
11:01:51 SHT: Shutter 1 Real 117000, Start 166000, Stop 108000, Dir -1, Delay 0, Rtc 2.50 [s], Freq 0, PWM 0
11:01:51 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":59,"Direction":-1,"Target":54}}
11:01:51 SHT: Stop Shutter 0 ..
11:01:52 SRC: Shutter
11:01:52 RSL: stat/tasmota_C2C9BC/RESULT = {"POWER2":"OFF"}
11:01:52 RSL: stat/tasmota_C2C9BC/POWER2 = OFF
11:01:52 SHT: Shutter 1 Real 108000, Start 108000, Stop 108000, Dir 0, Delay 0, Rtc 2.95 [s], Freq 0, PWM 0
11:01:52 RSL: stat/tasmota_C2C9BC/SHUTTER1 = 54
11:01:52 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":54,"Direction":0,"Target":54}}
11:01:52 CFG: Saved to flash at F5, Count 47, Bytes 4096

Hope this helps and thanks a lot Stefan for your investigation!

sergios100, 2020-09-23:
One more thing:
It seems to be related to the Webinterface button! If i assign Hardware-inputs and use them, they behave as expected.

Buttons on GPIO 2 and GPIO 12:

`11:35:14 CMD: Group 0, Index 1, Command "GPIO", Data "255"`
`11:35:14 RSL: stat/tasmota_C2C9BC/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"17":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"22":"Relay2"},"GPIO5":{"21":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"18":"Button2"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}`

**Hardware "UP" button pressed, Shutter moves upwards:**
11:31:15 APP: Button1 multi-press 1
11:31:15 SRC: Button
11:31:15 RSL: stat/tasmota_C2C9BC/RESULT = {"POWER1":"ON"}
11:31:15 RSL: stat/tasmota_C2C9BC/POWER1 = ON
11:31:15 SHT: Shutter 1 Real 100000, Start 100000, Stop 200000, Dir 1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0
11:31:15 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":50,"Direction":1,"Target":100}}
11:31:15 CFG: Saved to flash at F7, Count 181, Bytes 4096
11:31:16 SHT: Shutter 1 Real 109000, Start 100000, Stop 200000, Dir 1, Delay 0, Rtc 0.50 [s], Freq 0, PWM 0
11:31:16 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":55,"Direction":1,"Target":100}}
11:31:17 SHT: Shutter 1 Real 130000, Start 100000, Stop 200000, Dir 1, Delay 0, Rtc 1.55 [s], Freq 0, PWM 0
11:31:17 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":65,"Direction":1,"Target":100}}
11:31:18 SHT: Shutter 1 Real 150000, Start 100000, Stop 200000, Dir 1, Delay 0, Rtc 2.55 [s], Freq 0, PWM 0
11:31:18 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":75,"Direction":1,"Target":100}}
**Hardware "UP" button pressed again, to stop motion:**
11:31:18 APP: Button1 multi-press 1
11:31:19 SRC: Button
11:31:19 SHT: Stop Shutter 0 ..
11:31:19 RSL: stat/tasmota_C2C9BC/RESULT = {"POWER1":"OFF"}
11:31:19 RSL: stat/tasmota_C2C9BC/POWER1 = OFF
11:31:20 CFG: Saved to flash at F6, Count 182, Bytes 4096

all fine! Same as well for the "down" hardware button:

11:40:15 APP: Button2 multi-press 1
11:40:15 SRC: Button
11:40:15 RSL: stat/tasmota_C2C9BC/RESULT = {"POWER2":"ON"}
11:40:15 RSL: stat/tasmota_C2C9BC/POWER2 = ON
11:40:15 SHT: Shutter 1 Real 100000, Start 100000, Stop 0, Dir -1, Delay 0, Rtc 0.00 [s], Freq 0, PWM 0
11:40:15 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":50,"Direction":-1,"Target":0}}
11:40:16 SHT: Shutter 1 Real 87000, Start 100000, Stop 0, Dir -1, Delay 0, Rtc 0.70 [s], Freq 0, PWM 0
11:40:16 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":44,"Direction":-1,"Target":0}}
11:40:16 CFG: Saved to flash at FB, Count 185, Bytes 4096
11:40:17 SHT: Shutter 1 Real 67000, Start 100000, Stop 0, Dir -1, Delay 0, Rtc 1.70 [s], Freq 0, PWM 0
11:40:17 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":34,"Direction":-1,"Target":0}}
11:40:18 SHT: Shutter 1 Real 47000, Start 100000, Stop 0, Dir -1, Delay 0, Rtc 2.70 [s], Freq 0, PWM 0
11:40:18 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":24,"Direction":-1,"Target":0}}
11:40:18 APP: Button2 multi-press 1
11:40:19 SHT: Shutter 1 Real 27000, Start 100000, Stop 0, Dir -1, Delay 0, Rtc 3.70 [s], Freq 0, PWM 0
11:40:19 RSL: stat/tasmota_C2C9BC/RESULT = {"Shutter1":{"Position":14,"Direction":-1,"Target":0}}
11:40:19 SRC: Button
11:40:19 SHT: Stop Shutter 0 ..
11:40:19 RSL: stat/tasmota_C2C9BC/RESULT = {"POWER2":"OFF"}
11:40:19 RSL: stat/tasmota_C2C9BC/POWER2 = OFF
11:40:19 CFG: Saved to flash at FA, Count 186, Bytes 4096

blackscreener, 2020-09-26:
Hi all.
Yeah, there is a problem with up shutter function. Hardware and software up button pressed again stops and move down shutter to previous position. Also "ShutterStop1" does the same.

19:38:25 MQT: tele/roleta_kotlownia/STATE = {"Time":"2020-09-26T19:38:25","Uptime":"0T00:12:56","UptimeSec":776,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"OpenWrt","BSSId":"2C:30:33:22:4C:E8","Channel":8,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:38:25 MQT: stat/roleta_kotlownia/RESULT = {"POWER1":"ON"}
19:38:25 MQT: stat/roleta_kotlownia/POWER1 = ON
19:38:25 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Button1":1}}
19:38:25 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":27,"Direction":1,"Target":100}}
19:38:26 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":31,"Direction":1,"Target":100}}
19:38:27 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":36,"Direction":1,"Target":100}}
19:38:28 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":41,"Direction":1,"Target":100}}
19:38:29 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":46,"Direction":1,"Target":100}}
19:38:30 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":51,"Direction":1,"Target":100}}
19:38:31 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":56,"Direction":1,"Target":100}}
19:38:32 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":61,"Direction":1,"Target":100}}
19:38:33 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":66,"Direction":1,"Target":100}}
19:38:33 RUL: BUTTON3#STATE performs "ShutterStop1"
19:38:34 MQT: tele/roleta_kotlownia/STATE = {"Time":"2020-09-26T19:38:34","Uptime":"0T00:13:05","UptimeSec":785,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"OpenWrt","BSSId":"2C:30:33:22:4C:E8","Channel":8,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:38:34 MQT: stat/roleta_kotlownia/RESULT = {"POWER1":"OFF"}
19:38:34 MQT: stat/roleta_kotlownia/POWER1 = OFF
19:38:34 MQT: tele/roleta_kotlownia/STATE = {"Time":"2020-09-26T19:38:34","Uptime":"0T00:13:05","UptimeSec":785,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"OpenWrt","BSSId":"2C:30:33:22:4C:E8","Channel":8,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:38:34 MQT: stat/roleta_kotlownia/RESULT = {"POWER2":"ON"}
19:38:34 MQT: stat/roleta_kotlownia/POWER2 = ON
19:38:34 MQT: stat/roleta_kotlownia/RESULT = {"ShutterStop1":31}
19:38:34 MQT: stat/roleta_kotlownia/BUTTON3T = {"TRIG":"SINGLE"}
19:38:34 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":67,"Direction":-1,"Target":31}}
19:38:34 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":66,"Direction":-1,"Target":31}}
19:38:35 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":60,"Direction":-1,"Target":31}}
19:38:36 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":55,"Direction":-1,"Target":31}}
19:38:37 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":51,"Direction":-1,"Target":31}}
19:38:38 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":46,"Direction":-1,"Target":31}}
19:38:39 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":41,"Direction":-1,"Target":31}}
19:38:40 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":36,"Direction":-1,"Target":31}}
19:38:41 MQT: tele/roleta_kotlownia/STATE = {"Time":"2020-09-26T19:38:41","Uptime":"0T00:13:12","UptimeSec":792,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"OpenWrt","BSSId":"2C:30:33:22:4C:E8","Channel":8,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:38:41 MQT: stat/roleta_kotlownia/RESULT = {"POWER2":"OFF"}
19:38:41 MQT: stat/roleta_kotlownia/POWER2 = OFF
19:38:41 MQT: stat/roleta_kotlownia/SHUTTER1 = 31
19:38:41 MQT: stat/roleta_kotlownia/RESULT = {"Shutter1":{"Position":31,"Direction":0,"Target":31}}

stefanbode, 2020-09-27:
I have "fixed" the issue. One thing was an incomplete implementation for the shutterstop command. This was only working with timer shutters but not with the one using counters. This fixed the issue with the shutter starting again after already stopping. Anyhow, I had a reboot with shuttermotordelay 0 in this case. I'm still not found the root cause where the printf  command caused the exception. Even with stack traces quite hidden. As a working temporary workaround the shuttermotordelay is always 0.2 seconds and higher. In this case it works fine.

closed 2020-09-27 by arendst

blackscreener, 2020-09-27:
> I have "fixed" the issue. One thing was an incomplete implementation for the shutterstop command. This was only working with timer shutters but not with the one using counters. This fixed the issue [...]

Thx, now works fine. What about interlock time? It's too low..

stefanbode, 2020-09-27:
the interlock time is not shutter dependend. the interlock is a tasmota standard procedure. if you send normal command shutterxxx the shutter can take care that there is enough changeover time. this is not possible if you manually hit the relay. in this case tasmota controls the changeover. it is afaik 10ms.

PR #9406 (2020-09-27): Fix on Reboot and wrong operation during shutterstop
fix #9340

## Description:

**Related issue (if applicable):** fixes #<Tasmota issue number goes here>

## Checklist:
  - [ x] The pull request is done against the latest dev branch
  - [ x] Only relevant files were touched
  - [ x] Only one feature/fix was added per PR.
  - [x ] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.1
  - [x ] The code change is tested and works on core ESP32 V.1.12.2
  - [x ] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cl [... cut]
commit e8466371 (2020-09-27): Fix on Reboot and wrong operation during shutterstop

fix #9340