arendst/Tasmota #9343: Tasmota multiple sensor topics

fizroy, 2020-09-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

fizroy, 2020-09-18:
I have 3  LYWSD03MMC ble sensors conected to a esp32 with tasmota, but the data from them is sent to the same mqtt topic tele/tasmota/SENSOR...and not all toghether but in different lines...i olso use openhab to recive the data from the sensors and since they all use the same topic and have different id's i get a log full of warnigs.
[image]
[image]
 is it posible to have all the sensors maybe report in the same line or have different sensor topic like for zigbee sensors or like POWER???

schrej-zz, 2020-09-18:
You can use the REGEX transformation in the Openhab channels to prevent these warnings.

fizroy, 2020-09-18:
I already use json for the data ...any tip on the REGEX part?

fizroy, 2020-09-18:
Still it would be better if the dev's implement something like SetOption 89 but for ble devices.

schrej-zz, 2020-09-18:
See https://community.openhab.org/t/mqtt-2-5-problem-with-receiving-status-updates-on-some-things/90054

Jason2866, 2020-09-19:
Funny same answer i gave already in Discord before 😊

closed 2020-09-19 by ascillato2