arendst/Tasmota #9344: Tasmota-IR Buffer overrun

phinnay, 2020-09-18:
### PROBLEM DESCRIPTION
Getting consistent "buffer overrun messages in the console after flashing multiple devices with tasmota-ir

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Jinvoo [SM-AW502A](https://templates.blakadder.com/jinvoo_SM-AW502A.html)
- [ ] Tasmota binary firmware version number used: 8.50
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotarizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  "command unknown"

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
not using rules

```
- [ ] Provide the output of this command: `Status 0`:
```
  00:02:46 RSL: stat/tasmota_B559D2/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_B559D2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:02:46 RSL: stat/tasmota_B559D2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:01:43","StartupUTC":"2020-09-18T23:01:03","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2020-09-18T23:25:44","SaveCount":31,"SaveAddress":"F5000"}}
00:02:47 RSL: stat/tasmota_B559D2/STATUS2 = {"StatusFWR":{"Version":"8.5.0(ir)","BuildDateTime":"2020-09-09T11:43:50","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"348/699"}}
00:02:47 RSL: stat/tasmota_B559D2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["myssid",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
00:02:47 RSL: stat/tasmota_B559D2/STATUS4 = {"StatusMEM":{"ProgramSize":571,"Free":432,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F5A0582","04000001","00000002","00000000","00000000","00000000","00000000"],"Drivers":"1,2,4,5,7,9,10,12","Sensors":""}}
00:02:47 RSL: stat/tasmota_B559D2/STATUS5 = {"StatusNET":{"Hostname":"tasmota_B559D2-6610","IPAddress":"192.168.10.156","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.70","Mac":"50:02:91:B5:59:D2","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:02:47 RSL: stat/tasmota_B559D2/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B559D2","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:02:47 RSL: stat/tasmota_B559D2/STATUS7 = {"StatusTIM":{"UTC":"2020-09-18T23:02:47","Local":"2020-09-19T00:02:47","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:33","Sunset":"18:53"}}
00:02:47 RSL: stat/tasmota_B559D2/STATUS10 = {"StatusSNS":{"Time":"2020-09-19T00:02:47"}}
00:02:47 RSL: stat/tasmota_B559D2/STATUS11 = {"StatusSTS":{"Time":"2020-09-19T00:02:47","Uptime":"0T00:01:44","UptimeSec":104,"Vcc":3.466,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"myssd","BSSId":"24:A4:3C:35:F0:84","Channel":6,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:14"}}}
00:02:49 RSL: stat/tasmota_B559D2/RESULT = {"Command":"Unknown"}
00:02:49 CMD: Serial buffer overrun

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
 00:02:49 CMD: 00:02:49 RSL: stat/tasmota_B559D2/RESULT = {"Command":"Unknown"}
00:02:49 RSL: stat/tasmota_B559D2/RESULT = {"Command":"Unknown"}
00:02:49 CMD: Serial buffer overrun
00:02:49 CMD: Serial buffer overrun

```

### TO REPRODUCE
Install tasmota-ir and connect device to it's own power supply

### EXPECTED BEHAVIOUR
not get constant buffer overruns on its own power supply 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I have flashed two other devices from different manufacturers and all exhibit the same symptoms, as per this thread: https://github.com/arendst/Tasmota/issues/8829#issuecomment-691356755

I purchased this  Jinvoo device as it was the only one listed on the compatibility list available from Amazon

**(Please, remember to close the issue when the problem has been addressed)**

phinnay, 2020-09-18:
Jinvoo device:
[image]

[image]

Jason2866, 2020-09-19:
Check if rx and tx is not connected to anything. Probably the are.

phinnay, 2020-09-19:
> 
> 
> Check if rx and tx is not connected to anything. Probably the are.

If you mean connected to my programming dongle, no, they are not.
If you mean connected to other parts on the board, I'd have to assume so as the pads are soldered...

I've flashed 40+ things around my home with Tasmota and have not encountered buffer overruns on any of them, nor has de-soldering TX/RX from the board been a requirement - if this is specific to the "ir" flavor that's fine! I'm very curious why it is needed.

ascillato, 2020-09-19:
Please, try command `seriallog 0`

phinnay, 2020-09-19:
> 
> 
> Please, try command `seriallog 0`

No change, confirmed setting remained after reboot. 

[image]

ascillato, 2020-09-20:
From your logs, your TX and RX lines are in short circuit between them.

Please, disconnect them and try again. Thanks.

arendst, 2020-09-20:
Duplicate. Check your base module being 62 which disables serial.

closed 2020-09-20 by ascillato2

iz3man, 2021-06-05:
> Duplicate. Check your base module being 62 which disables serial.

Thanks. That solved my issue as well! (Came a little late to the show ;) )