arendst/Tasmota #9345: Tasmota support for LilyGO ttgo t-internet-poe

xaaron, 2020-09-19:
**Have you looked for this feature in other issues and in the docs?**  
Yes, I have looked and found similar support with the Olimex POE device, however this is a different more affordable board.  The need to run Tasmota on Ethernet-connected POE device - the LilyGO ttgo t-internet-poe similar to the Olimex POE esp32  at about half its price.  
.
**Is your feature request related to a problem? Please describe.**
No, this is a request to support a new module.  
I would like to see the LilyGO ttgo t-internet-poe supported in Tasmota.  Since a great deal of the work is already done with respect to the Ethernet connectivity already, I suspectmost of the remaining work will be with GPIO mapping/templates.

**Describe the solution you'd like**  
Please add support for this Lilygo t-internet-poe module, now that ESP32 platform and ethernet is supported by Tasmota.

**Describe alternatives you've considered**  
The Olimex is an alternative, but this presents a more affordable option.

**Additional context**  
_Add any other context or screenshots about the feature request here._
[image]
htps://www.aliexpress.comitem/4001122992446.html
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-19:
Since it uses the already Tasmota supported Ethernet chip 8720 this bord should work with Tasmota.
You have to configure a Template for this board.

ascillato2, 2020-09-19:
Please, as you have this board, check the gpio connections and make a template as explained in the docs. All components are supported, you just need to tell Tasmota where are they connected to.

Then, please, submit the template to the templates database (instructions in the docs)

If you need help, please, do not hesitate on using the Tasmota support chat. 

Thanks for sharing.

xaaron, 2020-09-19:
Thank you for the quick responses.  I will go through the process of creating a template for this board.  Once complete, I'll upload to the templates database.

xaaron, 2020-09-22:
Reopening because I am unable to get ethernet working.  Related to:  https://github.com/arendst/Tasmota/issues/9360

Jason2866, 2020-09-22:
You find the GPIO mapping here https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE/blob/master/example/eth/eth.ino

Jason2866, 2020-09-22:
The module has nearly no documentation!
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE/issues/2

xaaron, 2020-09-22:
Thank you Jason2866 for your help.  I am familiar with that ino, and
originally did base my pin mapping based on it, however, in further
diagnosing this board, I've discovered that the only way I could get the
ethernet to work using the esp-idf example/ethernet/basic and
example/ethernet/eth2ap working was to set the *Output RMII clock from
GPIO0 (Experimental!) *flag.  I was then able to also validate this in
esphome with the settings:
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_OUT
  phy_addr: 0

However to get the above working, I did need to disable the WIFI

This leads me to believe that the GPIO mapping here
https://gihub.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE/blob/master/example/eth/eh.ino
may be incorrect.

Jason2866, 2020-09-22:
Sorry we can not help any further. We dont have this board.

xaaron, 2020-09-22:
Ok, I understand.  Will keep tying to work around and post if I find a
solution.

xaaron, 2020-09-23:
The clock mode was indeed incorrect in the sketch i was referencing for this board.  I've submitted the change upstream so others don't have the same issues I did.

I finally got ethernet working with Tasmota32 on this board using the following settings combined with some changes to the code:

03:35:41 CMD: ethernet
03:35:41 RSL: stat/tasmota_56B35D/RESULT = {"Ethernet":"ON"}
03:36:08 CMD: EthAddress
03:36:08 RSL: stat/tasmota_56B35D/RESULT = {"EthAddress":0}
03:37:19 CMD: EthClockMode
03:37:19 RSL: stat/tasmota_56B35D/RESULT = {"EthClockMode":1}
03:37:39 CMD: EthType
03:37:39 RSL: stat/tasmota_56B35D/RESULT = {"EthType":0}

I did have to make a change in: tasmota/xdrv_82_ethernet.ino:#define ETH_CLKMODE       ETH_CLOCK_GPIO0_IN
to ETH_CLOCK_GPIO0_OUT
I thought that making this change in tasmota/my_user_config.h and then running pio run was all that was necessary, but I may have broken something that kept it from working correctly when I compiled last.

Will try to recreate solution with a clean environment when I have some time.

xaaron, 2020-09-23:
Template upload tool seems to think the template string below is not valid JSON.

{"NAME":"LilyGO ttgo-t-eth-poe","GPIO":[0,1,1,1,1,1,1,1,1,1,1,1,1,1,5600,1,0,1,1,5568,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}

Jason2866, 2020-09-23:
There should be no code change neccessary. 
There is a command for https://tasmota.github.io/docs/Commands/#esp32
`EthClockMode 1`

xaaron, 2020-09-23:
You are correct.  I was able to build cleanly and apply the template.  Now this board works well.  I am unable to upload the working template for this board, but it is posted earlier in this thread.
Thank you for all your help Jason2866, ascillato2 and the rest of the team.  I encourage you to keep up the good work.

closed 2020-09-23 by xaaron

Jason2866, 2020-09-23:
Thx, for your work. I added the template to the docs. Section ESP32

fromCologne, 2020-10-28:
There are different hardware versions of this board.
Mine is V1.0 and I have to set EthClockMode 3.

hideo213, 2020-12-13:
Hy

I am now for several hours trying to get the ethernet on this board working.
The pre-compiled binary don't seem to recognize the "ethernet" command.

So i compiled one by my own with "#define USE_ETHERNET " in my_user_config.h
now the commands are recognized but ethernet still won't work.
I used this template {"NAME":"LilyGO ttgo-t-eth-poe","GPIO":[0,1,1,1,1,1,1,1,1,1,1,1,1,1,5600,1,0,1,1,5568,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
and set "ethclockmode 1"

tried several different modes with
#define ETH_TYPE
#define ETH_ADDR 
#define ETH_CLKMODE
or setting them in the console. But no chance. I even tried to switch off wifi

is it possible that the board is not able to autonegotiate speed/duplex? If so i need to get another switch.

Jason2866, 2020-12-13:
For both board variants there is feedback Ethernet is working.
You can try my precompiled Ethernet build https://github.com/Jason2866/Tasmota-specials/tree/firmware

xaaron, 2020-12-13:
hideo213,
I have version 1.2 of the board and can confirm that it does auto-negotiate
speed/duplex.
I didn't have to compile the firmware, I was able to use the one
automatically built by Jason that he linked to above.
What version of the board are you using?

   1. I have validated the below process for getting mine working:
   2. Download firmware:
   https://github.com/arendst/Tasmota/blob/firmware/firmware/tasmota32/tasmota32.bin
   3. Download all additional files:
   https://github.com/Jason2866/Tasmota-specials/tree/firmware/firmware/tasmota32/ESP32_needed_files
   4. Connect and flash (make sure COM5 is the appropriate device for you.
   I'm on linux and used /dev/ttyUSB0):  esptool.py --chip esp32 --port COM5
   --baud 921600 --before default_reset --after hard_reset write_flash -z
   --flash_mode dout --flash_freq 40m --flash_size detect 0x1000
   bootloader_dout_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000
   tasmota32.bin
   5. After flashing, connect to the device via WiFI/Network, and enter
   command mode.  In the tasmota console menu, you can issue commands:
   https://tasmota.github.io/docs/Commands/
   6. There you should be able to issue the ethclockmode command.  Here,
   you use 1 if you have version 1.2 of the board and  and a 3 if you have
   version 1.0 of the board
   7. I was not able to get both WiFi and Eth working simultaneously.

fromCologne, 2020-12-14:
I use V1.0 and it´s working well. Eth, Wifi parallel too. I compiled my own firmware (8.5.1) with Platformio.

user_config_override.h
```
#define USE_ETHERNET
  // TTGO T-Internet PoE
  #define ETH_TYPE          0  
  #define ETH_ADDR          0
  #define ETH_CLKMODE       3 
``` 

`{"NAME":"LilyGO ttgo-t-eth-poe","GPIO":[0,1,1,1,1,1,1,1,1,1,1,1,1,1,5600,1,0,1,1,5568,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}`

No need to send console commands. Autoneg is working with my HP GigE-Switch.
With console
```
ethernet on/off
wifi on/off
```
I can change the active network.

hideo213, 2020-12-14:
Hy Guys
You are very quick. Thank you a lot.

I Tried the "ethernet-Tasmota32" Jason was providing.
Sadly it's still not working.
I have version 1.0 of the board so i used "ethclockmode 3" and wifi 0. restartet, but still no linkup.
Maybe i got a broken Board (DOA) i should get two others in a several days so i can compare.

hideo213, 2020-12-14:
tried it again with the infos from the last post from cologne
now it works. Thank you very much. It even works with eth/wifi togehter.

fromCologne, 2020-12-15:
Great!
The one I got was quiet bad soldered. I had to repair the right side of the ESP32-WROOM module.

ohhai-kthxbai, 2020-12-27:
Hi
> I use V1.0 and it´s working well. Eth, Wifi parallel too. I compiled my own firmware (8.5.1) with Platformio.
Does Bluetooth (BLE?) also work in parallel?

Jason2866, 2020-12-27:
BT work in parallel. You have to use latest release or development version.

dyde121, 2021-01-20:
> tried it again with the infos from the last post from cologne
> now it works. Thank you very much. It even works with eth/wifi togehter.

What did you do to make it work on 1.0?
Did you compile from "ethernet-Tasmota32" or "tasmota32"?

dalescott0, 2021-02-02:
Where can I find tasmota32-ethernet.bin, I've looked on the releases page, I loaded 9.2.0.4 however the ethclockmode is not available in console.

Jason2866, 2021-02-02:
@dalescott0  there is no official ethernet build variant. You have to self compile as described in the documentation or use a unofficial build from here https://github.com/Jason2866/Tasmota-specials/tree/firmware

dalescott0, 2021-02-02:
@Jason2866 Thanks so much it worked!!!

stoehrmark, 2021-05-28:
@Jason2866 tasmota32-ethernet.bin works great thanks. I have tried to compile my own with no luck Any way you can add    #define USE_WIEGAND  and recompile. Thanks

Jason2866, 2021-05-28:
Added #define USE_WIEGAND in build Ethernet. Dont ask for more ;-)

stoehrmark, 2021-05-29:
Amazing Thank You. I'm working on a Tasmota POE entry control door system. This really helps :)

Noschvie, 2021-05-29:
> 
> I'm working on a Tasmota POE entry control door system.

@stoehrmark : Do you want to share the concept with other users? I'm interested ... Thanks!

stoehrmark, 2021-05-29:
My plan is to have a php site connected to a MySQL db. That will manage the key fob users.  It will talk over mqtt or just html to the Tasmota LilyGO. The LilyGo will be connected to the key fob reader and a electric door strike. It will send the key fob ID to the DB for validation and then get back a command to open the strike. All with just 1 ethernet cable over POE. Just need to figure out how to get 8-12v out of the POE side of the LilyGO to power the strike.

Noschvie, 2021-05-29:
Are "Mifare DesFire" token supported by Tasmota ?

Noschvie, 2021-05-29:
> 
> 
> Just need to figure out how to get 8-12v out of the POE side of the LilyGO to power the strike.

Have a look to "PoE Splitter 12V 2A Output"

davy932, 2022-04-02:
> For both board variants there is feedback Ethernet is working. You can try my precompiled Ethernet build https://github.com/Jason2866/Tasmota-specials/tree/firmware

Hey Jason - Stupid Question but does the tasmota32-ethernet.bin still exist??  I cannot find it within the library you referenced.  Many thanks

Jason2866, 2022-04-02:
Ethernet is enabled ny defaultt in development Tasmota32

davy932, 2022-04-02:
> Ethernet is enabled ny defaultt in development Tasmota32

Got it

Thanks