arendst/Tasmota #9349: ESP32 MQTT Connect and disconnect loop

ryoma3111, 2020-09-19:
### PROBLEM DESCRIPTION
I'm trying to get Tasmota to work on an ESP32.
I flashed using the built tasmota-esp32.bin.
But my ESP32 keeps disconnecting and connecting MQTT.
Is there any solution?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [o] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [o] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [o] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [o] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [o] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [o] Device used (e.g., Sonoff Basic): **ESP32-WROOM-32**
- [o] Tasmota binary firmware version number used: **Version":"8.5.0.1(tasmota)**
  - [o] Pre-compiled
  - [-] Self-compiled
    - [-] IDE / Compiler used:
- [o] Flashing tools used: **esptool.py**
- [o] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
  {"Module":{"1":"ESP32-DevKit"}}
  {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [o] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
  {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
  {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [o] Provide the output of this command: `Status 0`:
```
STATUS = {"Status":{"Module":1,"DeviceName":"Sonoff2","FriendlyName":["Sonoff2"],"Topic":"sonoff26","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:28:22","StartupUTC":"2020-09-19T10:58:39","Sleep":0,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2020-09-19T09:16:37","SaveCount":15}}
STATUS2 = {"StatusFWR":{"Version":"8.5.0.1(tasmota)","BuildDateTime":"2020-09-19T06:03:53","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","CpuFrequency":80,"Hardware":"ESP32","CR":"457/699"}}
STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Ryoma Wi-Fi Secure",""],"TelePeriod":0,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000068","00006100","00000000"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":1051,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","00140001","000000CF","010013C0","40000981","00004000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29","Sensors":"1,2,3,5,6"}}
STATUS5 = {"StatusNET":{"Hostname":"Sonoff2","IPAddress":"192.168.0.236","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"C4:4F:33:60:94:05","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.4","MqttPort":1883,"MqttClientMask":"Sonoff2","MqttClient":"Sonoff2","MqttUser":"ryoma3111","MqttCount":830,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
STATUS7 = {"StatusTIM":{"UTC":"2020-09-19T11:27:02","Local":"2020-09-19T20:27:02","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+09:00","Sunrise":"14:33","Sunset":"02:53"}}
STATUS10 = {"StatusSNS":{"Time":"2020-09-19T20:27:02"}}
STATUS11 = {"StatusSTS":{"Time":"2020-09-19T20:27:02","Uptime":"0T00:28:23","UptimeSec":1703,"Heap":111,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":830,"Wifi":{"AP":1,"SSId":"Ryoma Wi-Fi Secure","BSSId":"B0:39:56:85:FD:7A","Channel":13,"RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [o] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
20:29:08 MQT: tele/sonoff26/LWT = Online (retained)
20:29:10 MQT: Attempting connection...
20:29:10 MQT: Connected
20:29:10 MQT: tele/sonoff26/LWT = Online (retained)
20:29:12 MQT: Attempting connection...
20:29:12 MQT: Connected
20:29:12 MQT: tele/sonoff26/LWT = Online (retained)
20:29:14 MQT: Attempting connection...
20:29:14 MQT: Connected
20:29:14 MQT: tele/sonoff26/LWT = Online (retained)
20:29:16 MQT: Attempting connection...
20:29:16 MQT: Connected
20:29:16 MQT: tele/sonoff26/LWT = Online (retained)
20:29:18 MQT: Attempting connection...
20:29:18 MQT: Connected
20:29:18 MQT: tele/sonoff26/LWT = Online (retained)
20:29:20 MQT: Attempting connection...
20:29:20 MQT: Connected
20:29:20 MQT: tele/sonoff26/LWT = Online (retained)
20:29:22 MQT: Attempting connection...
20:29:22 MQT: Connected
20:29:22 MQT: tele/sonoff26/LWT = Online (retained)

```

### TO REPRODUCE
Flash the Pre-compiled firmware.

### EXPECTED BEHAVIOUR
I don't kwon.

### SCREENSHOTS
[image]
### ADDITIONAL CONTEXT
None
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-19:
Do you have changed something in the sleep setup?
Your LoadAvg is 999. This is bad. There is something screwed up.
With this high value you have instable wifi and mqtt

ryoma3111, 2020-09-19:
I was changing Sleep.
Now back to the default of 50. LoadAvg is 19.
But no improvement. Thanks for your reply.

ascillato2, 2020-09-25:
Hi, any news on this? I have several Esp32 connected by mqtt and couldn't reproduce your issue.
Are you able to erase all flash and flash again?

ascillato2, 2020-09-26:
Please, when you have time, tell us if your issue remains after reflashing. Thanks.

closed 2020-09-26 by ascillato2