arendst/Tasmota #9351: Add the Rule (TimerRule)

zmdk0913, 2020-09-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When I typed Rule with RuleTimer) on console, occur an Error

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I typed this command on console.
Rule4 ON TuyaReceived#cmndData DO RuleTimer1 3 ENDON
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
00:00:05 RSL: stat/PirSensor/RESULT = {"Command":"Error"}}1

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I sucessed to upload the firmware on tuya motion pir sensor. So I just doing setting, I want to send mqtt when the motion detected after 3 seconds. Beacuse auto wifi disconnect.  So I had typed rule
 

**(Please, remember to close the issue when the problem has been addressed)**
Rule4 ON TuyaReceived#cmndData DO RuleTimer1 3 ENDON -> Rule4 1
Rule5 ON Rules#Timer=1 DO Publish PirSensor/tele/PirSensor %value%_sensor_Triggered ENDON -> Rule5 1

But occurred error when add Rule4 On ...
Please let me know Correct command.

Jason2866, 2020-09-20:
The syntax is wrong it should be `on TuyaReceived#data=55AA000700050D040001011E do .... endon` or to fire always when TuyaReceived any data `on TuyaReceived#data do .... endon`

Jason2866, 2020-09-20:
You can find a example config for a tuya pir sensor https://templates.blakadder.com/TYMC-1.html

zmdk0913, 2020-09-20:
umm.. So can I possible delay 3seconds and publish mqtt?

zmdk0913, 2020-09-20:
[image]
it seems error on RuleTimer

meingraham, 2020-09-20:
Rule4? There are only three rule sets. Each rule set can hold multiple rules.

zmdk0913, 2020-09-20:
Ah Thank you. I will set rule in three rules.

zmdk0913, 2020-09-20:
I added this rule.
Rule1 ON TuyaReceived#CmndData=65010001016604000101 DO ruletimer1 15 ENDON ON rules#timer=1 DO publish KitchenPir/tele/PirSensor Sensor_triggered ENDON ON TuyaReceived#CmndData=65010001006604000101 DO ruletimer2 15 ENDON ON rules#timer=2 DO publish KitchenPir/tele/PirSensor Sensor_Triggered_Cleared ENDON
But Here are output
 00:00:04 RUL: RULES#TIMER=1 performs "publish KitchenPir/tele/PirSensor Sensor_triggered"
00:00:04 RSL: KitchenPir/tele/PirSensor = Sensor_triggered
Not upload mqtt... Please Help me. And I don't need clear trigger.

ascillato, 2020-09-20:
RSL means that you are not connected to any mqtt broker. Please, go to configuration and config you mqtt broker (broker's IP, user password, etc)

meingraham, 2020-09-20:
`RSL` shows that you do not have an MQTT connection.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://tasmota.github.io/docs/#/Home) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-20 by ascillato2