arendst/Tasmota #9352: Sonoff ZBBridge connected to Home Assistant: ZbConfig command unknown

thatso, 2020-09-20:
### PROBLEM DESCRIPTION
I've flashed my Sonoff ZBBridge with Tasmota and successfully connected it to Home Assistant following [this](https://templates.blakadder.com/sonoff_ZBBridge.html) guide.
Conecting Zigbee devices works.
However, I see no way to configure channel, PanID etc.
As the Home Assistant UI is still quite rudimentary, I've tried to configure basic network parameters in the Tasmota console. However, `ZbConfig` results in `command unknown`.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff ZBBridge
- [X] Tasmota binary firmware version number used: 8.5.0(zbbridge)
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: VS Code 1.49.0
- [X] Flashing tools used: Tasmotize
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
 RSL: RESULT = {"NAME":"ZHA ZBBridge","GPIO":[56,208,0,209,59,58,0,0,0,0,0,0,17],"FLAG":0,"BASE":18}
 RSL: RESULT = {"Module":{"0":"ZHA ZBBridge"}}
 RSL: RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"208":"TCP Tx"},"GPIO2":{"0":"None"},"GPIO3":{"209":"TCP Rx"},"GPIO4":{"59":"Led4i"},"GPIO5":{"58":"Led3i"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"17":"Button1"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":37,"Free":474,"Rules":"ON System#Boot do TCPStart 8888 endon"}
RSL: RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [X] Provide the output of this command: `Status 0`:
```
RSL: STATUS = {"Status":{"Module":0,"DeviceName":"ZBBridge","FriendlyName":["livingroom"],"Topic":"tasmota_xxxxxx","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.lan/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:21:12","StartupUTC":"2020-09-20T14:15:33","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2020-09-15T18:51:13","SaveCount":27,"SaveAddress":"F6000"}}
RSL: STATUS2 = {"StatusFWR":{"Version":"8.5.0(zbbridge)","BuildDateTime":"2020-09-14T01:04:01","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"359/699"}}
RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mynetwork",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
RSL: STATUS4 = {"StatusMEM":{"ProgramSize":528,"Free":472,"Heap":27,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F100786","04500001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,21,23,41","Sensors":""}}
RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_xxxxxx-xxxx","IPAddress":"xxx.xxx.xxx.xxx","Gateway":"xxx.xxx.xxx.xxx","Subnetmask":"xxx.xxx.xxx.xxx","DNSServer":"xxx.xxx.x.xxx","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-09-20T14:36:45","Local":"2020-09-20T16:36:45","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"xx:xx","Sunset":"xx:xx"}}
RSL: STATUS10 = {"StatusSNS":{"Time":"2020-09-20T16:36:45"}}
RSL: STATUS11 = {"StatusSTS":{"Time":"2020-09-20T16:36:45","Uptime":"0T00:21:12","UptimeSec":1272,"Vcc":3.473,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"mynetwork","BSSId":"xx:xx:xx:xx:xx:xx","Channel":1,"RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
CMD: ZbConfig
SRC: WebConsole from xxx.xxx.xxx.xxx
CMD: Group 0, Index 1, Command "ZBCONFIG", Data ""
RSL: RESULT = {"Command":"Unknown"}
```

### TO REPRODUCE
Connect Sonoff ZBBridge to Home Assistant.
Try to setup Zigbee network parameters in Tasmota console with `ZbConfig`.

### EXPECTED BEHAVIOUR
Output of the current setting and the ability to change settings.

### SCREENSHOTS

### ADDITIONAL CONTEXT
Maybe I did understand something completely wrong and this is expected behavior?
Does this mean that currently there is no way to configure network parameters with this setup?

s-hadinger, 2020-09-20:
This repository is for Tasmota, not Home Assistant. Please refer to Home Assistant web site, we can't help you here.

thatso, 2020-09-20:
@s-hadinger did you even read the issue?
Its about a command in Tasmota, hence I believe this is exactly the right place to ask.

s-hadinger, 2020-09-20:
You have to chose whether you want to use Home Assistant or Zigbee2Tasmota to control the ZIgbee MCU. Since you are using Home Assistant, Tasmota only acts as a dumb TCP to Serial bridge.

`ZbConfig` is a command only active if you are using Zigbee2Tasmota (i.e. not Home Assistant).

You need to choose between both...

thatso, 2020-09-20:
Now that is a useful answer. I suspected  that ZHA bridge mode could cause this behavior, I just did not find this fact anywhere in the docs.

closed 2020-09-20 by thatso

ascillato2, 2020-09-20:
@thatso

Please, you should be a little more kind. Your lack of respect to people helping you is not tolerated.

You have been warned. Next time you will be banned and reported to github as explained in the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and in the [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

This issue is now locked.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)