arendst/Tasmota #9358: Zigbee2Tasmota support for Radiator Thermostat e.g. Spirit Zigbee SPZB0001

derFrickler, 2020-09-22:
**Have you looked for this feature in other issues and in the docs?**
  
yes, but did not find anything related.

**Is your feature request related to a problem? Please describe.**  

Yes, i use the new Sonoff Zigbee bridge with Tasmota and would like to connect my heating as well.
Did not find any MQTT-able radiator thermostats, but zigbee ones. They are supported by zigbee2mqtt, but not zigbee2tasmota. 

**Describe the solution you'd like**  

It would be great to have support for radiator thermostats in tasmotas zigbee implementation. Especially the 
Spirit Zigbee Radiator Thermostat Model SPZB0001  https://zigbee.blakadder.com/Eurotronic_SPZB0001.html
would be great as it is cheap and available from Amazon.

Is there a kind of tutorial on how to implement new devices into zigbee2tasmota or is there a way to "convert" the implementation from zigbee2mqtt?

**Describe alternatives you've considered**  

using zigbee2mqtt with several bridges to cover the whole building.

**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-09-22:
Oh, this one looks nice. We can indeed look at what Zigbee2MQTT did and transpose to Z2T. I think I'll buy one of these.

In the mean time, could you please try to pair with Z2T and paste the logs of pairing here?

derFrickler, 2020-09-22:
Wow, that sounds good! 
Don't have one yet, as it was not supported, just ordered on on Amazon ;_)

s-hadinger, 2020-09-23:
First quick test, it paired successfully but required a complete Reset first (pressing all 3 buttons for 10 seconds).

```
{"ZbState":{"Status":33,"Endpoint":"0x01","ProfileId":"0x0104","DeviceId":"0x0301","DeviceVersion":1"InClusters":["0x0000","0x0001","0x0003","0x0201","0x0019","0x000A"],"OutClusters":["0x0000","0x0001","0x0003","0x0004","0x0201","0x0019","0x000A"]}}
ZIG: auto-bind `ZbBind {"Device":"0x7985","Endpoint":1,"Cluster":"0x0001"}`
MQT: tele/tasmota/ZBBridge/RESULT = {"ZbBind":{"Device":"0x7985","Status":0,"StatusMessage":"SUCCESS"}}
ZIG: auto-bind `ZbSend {"Device":"0x7985","Config":{"BatteryVoltage":{"MinInterval":900,"MaxInterval":900,"ReportableChange":0.1},"BatteryPercentage":{"MinInterval":900,"MaxInterval":900,"ReportableChange":1}}}`
MQT: tele/tasmota/ZBBridge/SENSOR = {"0x7985":{"Device":"0x7985","ConfigResponse":{"BatteryVoltage":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"},"0001/0084":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"}},"Endpoint":1,"LinkQuality":163}}

{"ZbStatus2":[{"Device":"0x7985","IEEEAddr":"0x00158D0001FFF8B4","ModelId":"SPZB0001","Manufacturer":"Eurotronic","Endpoints":[1]}]}
```

It looks at a first glance that automatic battery status is not supported, not a big issue.

derFrickler, 2020-09-23:
That sound great! Will receive mine tomorrow to do some tests.

s-hadinger, 2020-09-23:
Good news, the Eurotronic documentation gives the detailed list of zigbee attributes supported. BatteryPercentage is reportable.

derFrickler, 2020-09-26:
Received mine and bound it to the bridge. Anything i can test yet?

s-hadinger, 2020-09-26:
Nothing specific for now.

s-hadinger, 2020-09-27:
I added support in #9405 for the Thermostat attributes. You can now do the following attribute reads:
```
ZbSend {"Device":"Radiator","Cluster":0,"Read":[0,1,2,3,4,5,6,7]}
MQT: tele/tasmota/ZBBridge/SENSOR = {"Radiator":{"Device":"0x7985","Name":"Radiator","ZCLVersion":2,"AppVersion":22,"StackVersion":5,"HWVersion":35,"Manufacturer":"Eurotronic","ModelId":"SPZB0001","DateCode":"20191014","PowerSource":3,"Endpoint":1,"LinkQuality":63}}

ZbSend {"Device":"Radiator","Cluster":1,"Read":[33]}
MQT: tele/tasmota/ZBBridge/SENSOR = {"Radiator":{"Device":"0x7985","Name":"Radiator","BatteryPercentage":100,"Endpoint":1,"LinkQuality":55}}

ZbSend {"Device":"Radiator","Cluster":"0x0201","Read":[0,8,16,18,20,21,22]}
MQT: tele/tasmota/ZBBridge/SENSOR = {"Radiator":{"Device":"0x7985","Name":"Radiator","LocalTemperature":20.5,"PIHeatingDemand":17,"LocalTemperatureCalibration":0,"OccupiedHeatingSetpoint":21,"UnoccupiedHeatingSetpoint":16,"MinHeatSetpointLimit":5,"MaxHeatSetpointLimit":30,"Endpoint":1,"LinkQuality":53}}

ZbSend {"Device":"Radiator","Cluster":"0x0201","Read":[26,27,28]}
MQT: tele/tasmota/ZBBridge/SENSOR = {"Radiator":{"Device":"0x7985","Name":"Radiator","LocalTemperature":20.5,"PIHeatingDemand":17,"LocalTemperatureCalibration":0,"OccupiedHeatingSetpoint":21,"UnoccupiedHeatingSetpoint":16,"MinHeatSetpointLimit":5,"MaxHeatSetpointLimit":30,"Endpoint":1,"LinkQuality":53}}
```

The `LocalTemperature` is shown in the web UI, although it will deserve a better thermostat specific UI.

There are cluster specific attributes but I can't find the relevant ManufCode (yet).

s-hadinger, 2020-09-27:
The manuf code is 0x1037.

The last attributes can be read with this:
```
ZbSend {"Device":"Radiator","Cluster":"0x0201","Manuf":"0x1037","Read":["0x4000","0x4001","0x4002","0x4003","0x4008"]}
MQT: tele/tasmota/ZBBridge/SENSOR = {"Radiator":{"Device":"0x7985","Name":"Radiator","TRVMode":2,"SetValvePosition":147,"ThErrors":0,"CurrentTemperatureSetPoint":20,"0201/4008":null,"Endpoint":1,"LinkQuality":81}}
```

closed 2020-09-27 by s-hadinger

derFrickler, 2020-10-01:
YES! that works. Thanks a lot!
Is there a way to make the thermostat report its values automatically if they change or hourly?
How to set the heating temp?

s-hadinger, 2020-10-01:
You can try the following command for attribute reporting (change the device name):
`ZbSend {"Device":"Radiator","Config":{"LocalTemperature":{"MinInterval":60,"MaxInterval":600,"ReportableChange":0.5},"OccupiedHeatingSetpoint":{"MinInterval":60,"MaxInterval":600,"ReportableChange":0.5},"PIHeatingDemand":{"MinInterval":60,"MaxInterval":600}}}`

Changing temp is simple:
`ZbSend {"Device":"Radiator","Write":{"OccupiedHeatingSetpoint":20.5}}`

derFrickler, 2020-10-02:
Thanks, setting the heating point works fine, but changing the config to report the values has no effect.
Thats what i get as answer from the device: 

`08:24:31 CMD: ZbSend {"Device":"Spirit","Config":{"LocalTemperature":{"MinInterval":60,"MaxInterval":600,"ReportableChange":0.5},"OccupiedHeatingSetpoint":{"MinInterval":60,"MaxInterval":600,"ReportableChange":0.5},"PIHeatingDemand":{"MinInterval":60,"MaxInterval":600}}}

08:24:31 MQT: stat/ZBBridgeEG/RESULT = {"ZbSend":"Done"}

08:24:32 MQT: tele/ZBBridgeEG/SENSOR = {"ZbReceived":{"Spirit":{"Device":"0xB8F3","Name":"Spirit","ConfigResponse":{},"Endpoint":1,"LinkQuality":87}}}`

What i do for now is request the values each hour via mqtt.

derFrickler, 2020-10-02:
BTW, i created a small Node-Red Dashboard to control and manage the Tasmota Zigbee bridges and devices, maybe its useful for some of you: https://github.com/derFrickler/tasmota-zigbee-admin

s-hadinger, 2020-10-02:
Aha. I forgot the binding, which is mandatory to receive the attribute reporting:

`ZbBind {"Device":"Radiator","Cluster":"LocalTemperature"}`

If you want to manually get the 3 main values, use the following:

`ZbSend {"Device":"Radiator","Read":{"LocalTemperature":true,"OccupiedHeatingSetpoint":true,"PIHeatingDemand":true}}`

s-hadinger, 2020-10-02:
I realized something was wrong with the `PIHeatingDemand` reporting that was sent every minutes thus draining battery. The right command is:

`ZbSend {"Device":"Radiator","Config":{"LocalTemperature":{"MinInterval":60,"MaxInterval":600,"ReportableChange":0.5},"OccupiedHeatingSetpoint":{"MinInterval":60,"MaxInterval":600,"ReportableChange":0.5},"PIHeatingDemand":{"MinInterval":60,"MaxInterval":600,"ReportableChange":5}}}`

and 

`ZbBind {"Device":"Radiator","Cluster":"LocalTemperature"}`

s-hadinger, 2020-10-02:
Last thing, latest #9453 now does auto-binding and auto-attribute-reporting during pairing time.

derFrickler, 2020-10-03:
Great! That works perfect!

galsom, 2020-10-15:
I am trying to add the same TRV to the Sonoff Zigbee Bridge, using Z2T 8.5.1 but am not able to get the TRV to pair with the bridge.

Other sensors such as Xiaomi Aqara temperature and door/window are able to pair just fine and are working properly.

The bridge has been flashed twice by now to make sure not is stuck there and the TRV has been reset multiple times by holding down the 3 buttons, wait for the count to 10, rebooting and pressing the boost button, but nothing changes.

The batteries were tested and even replaced with another new set, but that also did not have any impact.

Z2T disables the permission to join after 60 seconds and no other messages are displayed in the console, even with weblog level set to 4.

Do you have any suggestions of what to try?

Thank you in advance

derFrickler, 2020-10-15:
You will need the dev version 9.0.x.
It's not in the 8.5.

s-hadinger, 2020-10-15:
It should still pair with 8.5. 

I needed to reset the TRV first then enable pairing

galsom, 2020-10-15:
I assumed it should work indeed with 8.5. Tried also ZbReset and even a different flashed Sonoff Zigbee Bridge with the same Tasmota version, but still the same. 

In below you find the output between ZbPermitJoin 1 and the moment it is disabled:

```
21:26:00 CMD: Group 0, Index 1, Command "ZBPERMITJOIN", Data "1"
21:26:00 ZIG: ZbEZSPSend 22003C
21:26:00 ZIG: ZbEZSPSend 3600FCFF00003600000000000000011E0103013C01
21:26:00 RSL: tele/tasmota_D7F96C/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
21:26:00 RSL: stat/tasmota_D7F96C/RESULT = {"ZbPermitJoin":"Done"}
21:26:00 ZIG: {"ZbEZSPReceived":"220000"}
21:26:00 ZIG: {"ZbEZSPReceived":"360000C2"}
21:26:00 ZIG: {"ZbEZSPReceived":"45000500003600000000000000C2FF000000FFFF03013C01"}
21:26:00 ZIG: Internal ZDO message 0x0036 sent from 0x0000 (broadcast)
21:26:01 ZIG: {"ZbEZSPReceived":"3F0006FCFF00003600000000000000C2010000"}
21:27:00 RSL: tele/tasmota_D7F96C/RESULT = {"ZbState":{"Status":20,"Message":"Pairing mode disabled"}}
```
Not sure if this is helpful as nothing is appearing in the entire process and the valve is stuck in the `Jin` state with a blinking connection icon.

s-hadinger, 2020-10-16:
Unfortunately there is nothing in the logs. I'm afraid you will need a Zigbee sniffer to know what is going on. Ezsp does not provide any more logs when pairing fails

galsom, 2020-11-05:
Thank you for the assistance. I ended up returning the TRV and see online that other people have had mixed success with these valves. I ordered a Moes HY369RT on Aliexpress and that paired immediately. 

Now I'm figuring out how to integrate it properly in Home Assistant.

ghwaldo, 2024-05-03:
Hello,

new to Z2T.

I flashed ZBBridge and is working. I paired TRVZB, works fine.
But how can i set window open functions ? 
From https://www.zigbee2mqtt.io/devices/TRVZB.html i see, that itt exists. But in tasmota console i says unknown parameter. Tried many variants (window_open, windowopen, WindowOpen), but always sais unknown parameter.
Can it be done with a additional zb-file ?

PR #9405 (2020-09-27): Zigbee support for Thermostat cluster
## Description:

Added attributes from the Thermostat cluster and support for Eurotronic TRV.

The full detail of Eurotronic supported attributes is here: https://eurotronic.org/wp-content/uploads/2019/11/Spirit_ZigBee_BAL_web_EN_November-2019.pdf

**Related issue (if applicable):** fixes #9358 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.1
  - [x] The code chan [... cut]
PR #9408 (2020-09-27): Zigbee Thermostat Eurotronic fixes
## Description:

Fixes some issues with Eurotronic TRV:
- fix wrong name for attribute `RemoteSensing`
- change attribute name to `EurotronicErrors`
- fix wrong encoding of valve position: should be 0.100 and should not use the reserved value 255. But Eurotronic sends the attribute with values 0..255
- prepare fields in Device object for the Web UI

**Related issue (if applicable):** fixes #9358 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR. [... cut]
PR #9454 (2020-10-02): Zigbee add TRV to UI
## Description:

Add Thermostat and TRV to Web UI: shows current temperature, target temperature and valve position (%)
[image]

**Related issue (if applicable):** fixes #9358

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.1
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING. [... cut]