arendst/Tasmota #9359: Unable to send ir signal to Samsung TV

shermanchan, 2020-09-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have Tasmota 8.5.0 installed on ESP01, it is able to send ir signal with protocol set to NEC, however, it failed to triggered the ir led if i use SAMSUNG as protocol
cmnd/irsend10 -m '{"Protocol":"SAMSUNG","Bits":32,"Data":"0xE0E040BF"}'
I can see the ir led light up, with my phone camera, when i using NEC protocol, however I found no light if I using SAMSUNG protocol.  I know my ir code is correct as I use the same ir code to power up my TV on raspberry pi.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _EXP01 1M____
- [X] Tasmota binary firmware version number used: _8.5.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: OTA_____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
{"IRSend":"Done"}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

23:28:13 CFG: Saved to flash at F7, Count 139, Bytes 4096
23:28:16 MQT: Received Topic "Cave/LR-856B/cmnd/irsend10", Data Size 52, Data "{"Protocol":"SAMSUNG","Bits":32,"Data":"0xE0E040BF"}"
23:28:16 SRC: MQTT
23:28:16 CMD: Group 0, Index 10, Command "IRSEND", Data "{"Protocol":"SAMSUNG","Bits":32,"Data":"0xE0E040BF"}"
23:28:16 IRS: protocol_text NEC, protocol SAMSUNG, bits 32, data 3772793023 (0xE0E040BF), repeat 9, protocol_code -1
23:28:16 MQT: XXXX/YYYY/RESULT = {"IRSend":"Done"}
23:28:24 WIF: Checking connection...
23:28:27 MQT: XXXX/YYYY/STATE = {"Time":"2020-09-22T23:28:27","Uptime":"0T00:18:12","UptimeSec":1092,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"zzzzzzzz","BSSId":"AA:BB:CC:DD:EE:FF","Channel":4,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:06"}}
23:28:27 MQT: XXXX/YYYY/SENSOR = {"Time":"2020-09-22T23:28:27","SI7021":{"Temperature":22.2,"Humidity":41.0,"DewPoint":8.3},"TempUnit":"C"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Same with protocol NEC, the IR LED should light up with SAMSUNG protocol

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-22:
For Samsung protocol you have to use `tasmota-ir.bin`

ascillato2, 2020-09-22:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-22 by ascillato2

shermanchan, 2020-09-23:
Hi Team

Thanks for the quick response, do i still able to use DHT-22 sensor after I
changed to tasmota-ir version

Regards
Sherman

-- 
FetchTV Pty Ltd, Level 13, 100 Arthur St, North Sydney, NSW 2060
 

 
<http://www.fetchtv.com.au>

This email is sent by FetchTV Pty Ltd (ABN 36 
130 669 500). The contents of this communication may be 
confidential, 
legally privileged and/or copyright material. If you are not the intended 
recipient, any use, 
disclosure or copying of this communication is 
expressed prohibited. If you have received this email in error, 
please 
notify the sender and delete it immediately.

 
<http://facebook.com/fetchtv>

techman83, 2020-09-23:
@shantur I did this recently with my HVAC. I did have to use the sensors build with the following flags:
```
                             -DUSE_IR_REMOTE_FULL
                             -U_IR_ENABLE_DEFAULT_ 
```
I had to flash it via serial, as it was 760Kb (though I'm yet to figure out why the program space is only 1MB with 4MB of flash on the D1 Mini).

ascillato, 2020-09-23:
> though I'm yet to figure out why the program space is only 1MB with 4MB of flash on the D1 Mini

The esp8266 chip can ONLY handle 1MB flash for program. The rest can be used for OTA, for files, and for data, but not for program. That is a hardware limitation of the chip.

shermanchan, 2020-09-24:
Thanks all, I recompiled the tasmota with tasmota_ir and defined USE_DHT on
tasmota_configurations.h, now i got both Samsung IR and DHT supported with
size only 589k.

Best regards
Sherman

-- 
FetchTV Pty Ltd, Level 13, 100 Arthur St, North Sydney, NSW 2060
 

 
<http://www.fetchtv.com.au>

This email is sent by FetchTV Pty Ltd (ABN 36 
130 669 500). The contents of this communication may be 
confidential, 
legally privileged and/or copyright material. If you are not the intended 
recipient, any use, 
disclosure or copying of this communication is 
expressed prohibited. If you have received this email in error, 
please 
notify the sender and delete it immediately.

 
<http://facebook.com/fetchtv>