arendst/Tasmota #9360: Ethernet interface on Lilygo ttgo t-internet-poe with LAN8720A fails to activate

xaaron, 2020-09-22:
I've built tasmota32.bin with the following as required for this board.  
#define USE_ETHERNET                             // Add support for ethernet (Currently fixed for Olimex ESP32-PoE)
#define ETH_TYPE          0                    // [EthType] 0 = ETH_PHY_LAN8720
#define ETH_ADDR          1                    // [EthAddress] 0 = PHY0 .. 31 = PHY31, tried both 0 and 1 here since valid for lan8720
#define ETH_CLKMODE       1                    //  1 = ETH_CLOCK_GPIO0_OUT

However after flashing the device, I still am not able to bring up the ethernet interface.  I do get the web interface and a wifi ap to which I can connect and administer the device.  I can run the TASMOTA commands such as:
EthType0, EthClockMode 1, EthAddress 1, Ethernet 1 in that order.  However, interface still doesn't become active.  I am unable to get the mac address which makes me suspect the ETH_CLKMODE value may be where I need to focus my efforts to determine what is happening here.  Any help or guidance would be appreciated.

closed 2020-09-22 by close-issue-app[bot]

ascillato, 2020-09-22:
For that device, you need to set the correct gpio. There isn't any template yet, so you need to find the correct gpios and where are they connected to.

ascillato, 2020-09-22:
I think it is better to continue in your previous thread (#9345) so anyone searching for this device will find everything in the same issue.

As you have closed your previous issue, you can reopen if you think it is necessary.

xaaron, 2020-09-22:
Yes, I can re-open the other thread.  Is there documentation on setting the
gpio's that you can point me to?

ascillato, 2020-09-22:
Yes, please, search for templates on Tasmota.com