arendst/Tasmota #9362: Turning Sonoff switch into Latching Valve Time Controller for Irrigation

chamroeunbm, 2020-09-23:
Hi,
Please advice me on how to turn a Sonoff SV or Basic R3 or 4CHPro or any wifi switch into a latching valve controller with timing and cycle support. My use case is simple: turn on the irrigation water valve for 5-10 minutes, turn it off and turn back on again after 8hours and repeat this cycle.

A Latching Valve(different from solenoid valve) requires 50-100ms positive pulse to open it and 50-100ms negative pulse to close it(same wire).

Any advice is much appreciated and don't mind me if it's already post on this as I am new to this.

**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-09-23:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-23 by ascillato2

ascillato, 2020-09-23:
Hi, 

A Sonoff SV or a Sonoff Basic R3, are devices of just one relay. You need 2 relays. And a sonoff CH4 has 4 relays exceeding your requirement. 

It should be better to use a sonoff dual OR a shelly 2.5. If you use any of these last 2 devices, and flash Tasmota on it, you can use timers to turn on the opening relay and other timers to turn on the closing relay. Those timers can be configured using the Tasmota WebUI with the day of the week, hour, minutes and other options. 

Then to de-energize the opening and closing relays (to make the pulse) you can set the time by using commands:
```lua
pulsetime1 10
pulsetime2 10
```
These commands are just a config that must be entered only once in the Tasmota console. 

Also, you should check the `interlock 1` configuration command in order to not allow both relays to be turned on at the same time, so as to add more safety to your setup.

All these features and commands are explained in the Tasmota docs ( http://tasmota.com ) 

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato, 2020-09-23:
Hi, 

A Sonoff SV or a Sonoff Basic R3, are devices of just one relay. You need 2 relays. And a sonoff CH4 has 4 relays exceeding your requirement. 

It should be better to use a sonoff dual OR a shelly 2.5. If you use any of these last 2 devices, and flash Tasmota on it, you can use timers to turn on the opening relay and other timers to turn on the closing relay. Those timers can be configured using the Tasmota WebUI with the day of the week, hour, minutes and other options. 

Then to de-energize the opening and closing relays (to make the pulse) you can set the time by using commands:
```lua
pulsetime1 1
pulsetime2 1
```
These commands are just a config that must be entered only once in the Tasmota console. 

Also, you should check the `interlock 1` configuration command in order to not allow both relays to be turned on at the same time, so as to add more safety to your setup.

All these features and commands are explained in the Tasmota docs ( http://tasmota.com ) 

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)