arendst/Tasmota #9366: Add support for Xiaomi Aqara WXKG12LM wireless switch

derekoharrow, 2020-09-23:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
Device is not recognised when trying to pair.

**Describe the solution you'd like**  
Device recognised and paired/supported.

**Describe alternatives you've considered**  
N/A

**Additional context**  
Supported by Z2M - see https://www.zigbee2mqtt.io/devices/WXKG12LM.html

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-23:
Could you please try to pair with Z2T and paste the logs of pairing here?

derekoharrow, 2020-09-23:
I think it's the first line of this that you need:

```
12:08:10 ZIG: {"ZbEZSPReceived":"4500040000060000000001000015E8D602FFFFFF0962FDFF04010119000002"}
12:08:10 ZIG: Internal ZDO message 0x0006 sent from 0xFF02 (broadcast)
12:08:10 ZIG: {"ZbEZSPReceived":"3F0006FDFF0000060000000001000015000000"}
```

Thanks

derekoharrow, 2020-09-23:
I've just tried again and it seems to be working now! Sorry

closed 2020-09-23 by derekoharrow