arendst/Tasmota #9367: Add support for Samsung SmartThings plug with power measurement GP-WOU019BBDWG

derekoharrow, 2020-09-23:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
Unable to pair as device not recognised.

**Describe the solution you'd like**  
Device to be recognised and pair with Tasmota/Sonoff ZB Bridge

**Describe alternatives you've considered**  
N/A

**Additional context**  
Supported by Z2M - see https://www.zigbee2mqtt.io/devices/GP-WOU019BBDWG.html

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-23:
Could you please try to pair with Z2T and paste the logs of pairing here?

ascillato2, 2020-09-24:
Hi, any news on this? Can you share the logs?

derekoharrow, 2020-09-24:
I upgraded to the latest dev release and it seems to be supported in there. Thanks

closed 2020-09-24 by derekoharrow