arendst/Tasmota #9371: Esp32 iBeacon support

leranp, 2020-09-23:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Add support for iBeacon Tags , in my case i have NRF51822 Tags, the esp32 board flashed with EspHome, but i really want to move my firmware to tasmota like all of my devices.
Can i add the ibeacon future line and compile alone? 
I tried with tasmota32_sensor and i look like there is no iBeacon addon
Any help?
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-09-23:
According to the docs, iBeacon is already "confirmed to work" for ESP32, and included in the precompiled sensors build.
https://tasmota.github.io/docs/ESP32/

ascillato2, 2020-09-24:
Closing this issue as the requested feature is already implemented.

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-24 by ascillato2

leranp, 2020-09-26:
The iBeacon future is support only with a HM-11 board, the future request is for iBeacon that use the esp32 built in bluetooth