arendst/Tasmota #9372: Cannot change friendly names

XDM525, 2020-09-23:
### PROBLEM DESCRIPTION
I am using tasmota to control 4 relays through alexa however, when I go to change the friendly names from "tasmota, tasmota2 , tasmota3, tasmota4," to open, close, stop, unlock which are the functions of my relays it wont change on Alexa's end. I would delete the devices and try to discover them again the names are still labeled tasmota. No matter what I change the names to, they still come out as "tasmota". Also i noticed that it says they are unresponsive in the app but when i try to power them on anyways, the relays 1-3 turn on but I cant turn them off, relay 4 wont work at all through the app.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): __ESP8285___
- [x ] Tasmota binary firmware version number used: __8.5.0___
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used: _esptool____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
 22:41:39 CMD: backlog template
22:41:39 SRC: WebConsole from 192.168.1.192
22:41:39 CMD: Group 0, Index 1, Command "BACKLOG", Data "template"
22:41:39 SRC: Backlog
22:41:39 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
22:41:39 RSL: stat/tasmota_BBB64F/RESULT = {"NAME":"Alexa","GPIO":[0,0,0,0,21,22,0,0,23,0,0,24,0],"FLAG":0,"BASE":18}
22:41:42 UDP: Packet (119)

22:42:28 CMD: module
22:42:28 SRC: WebConsole from 192.168.1.192
22:42:28 CMD: Group 0, Index 1, Command "MODULE", Data ""
22:42:28 RSL: stat/tasmota_BBB64F/RESULT = {"Module":{"0":"Alexa"}}

22:43:10 CMD: GPIO 255
22:43:10 SRC: WebConsole from 192.168.1.192
22:43:10 CMD: Group 0, Index 1, Command "GPIO", Data "255"
22:43:10 RSL: stat/tasmota_BBB64F/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"22":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"23":"Relay3"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"24":"Relay4"},"GPIO16":{"0":"None"}}
22:43:11 WIF: Checking connection...

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  22:36:40 CMD: status 0
22:36:40 RSL: stat/tasmota_BBB64F/STATUS = {"Status":{"Module":0,"DeviceName":"device","FriendlyName":["Open","Close","Stop","Unlock"],"Topic":"tasmota_BBB64F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:36:40 RSL: stat/tasmota_BBB64F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:36","StartupUTC":"2020-09-23T21:36:04","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2020-09-23T21:49:01","SaveCount":62,"SaveAddress":"F6000"}}
22:36:40 RSL: stat/tasmota_BBB64F/STATUS2 = {"StatusFWR":{"Version":"8.5.0(tasmota)","BuildDateTime":"2020-09-09T11:41:02","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"367/699"}}
22:36:40 RSL: stat/tasmota_BBB64F/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Summit","Summit"],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
22:36:40 RSL: stat/tasmota_BBB64F/STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
22:36:40 RSL: stat/tasmota_BBB64F/STATUS5 = {"StatusNET":{"Hostname":"tasmota_BBB64F-5711","IPAddress":"192.168.1.188","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"68.105.28.16","Mac":"60:01:94:BB:B6:4F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:36:40 RSL: stat/tasmota_BBB64F/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BBB64F","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:36:40 RSL: stat/tasmota_BBB64F/STATUS7 = {"StatusTIM":{"UTC":"2020-09-23T21:36:40","Local":"2020-09-23T22:36:40","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:39","Sunset":"18:45"}}
22:36:40 RSL: stat/tasmota_BBB64F/STATUS10 = {"StatusSNS":{"Time":"2020-09-23T22:36:40"}}
22:36:40 RSL: stat/tasmota_BBB64F/STATUS11 = {"StatusSTS":{"Time":"2020-09-23T22:36:40","Uptime":"0T00:00:36","UptimeSec":36,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Summit","BSSId":"88:DC:96:7C:22:66","Channel":6,"RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
22:38:49 UDP: Packet (119)
22:38:49 UDP: Packet (119)
22:38:51 WIF: Checking connection...
22:38:54 HTP: Hue API (/bbb64f/lights/196830452) from 192.168.1.128
22:38:54 /lights path=/lights/196830452
22:38:54 HTP: Hue Result ({"state":{"on":false,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Unlock","modelid":"Alexa","manufacturername":"Tasmota","uniqueid":"60:01:94:bb:b6:4f:00:11-4"})
22:38:56 UDP: Packet (119)
22:38:57 UDP: Packet (119)
22:38:58 UDP: Packet (119)
22:38:59 UDP: Packet (119)
22:39:04 UDP: Packet (119)
22:39:04 UDP: Packet (119)
22:39:04 UDP: Packet (119)
22:39:11 WIF: Checking connection...
22:39:15 UDP: Packet (119)
22:39:15 UDP: Packet (119)
22:39:15 UDP: Packet (119)
22:39:17 UPP: Hue 3 response packets sent to 192.168.1.192:5826
22:39:19 UDP: Packet (119)
22:39:19 UDP: Packet (119)
22:39:19 UDP: Packet (119)
22:39:21 HTP: Hue API (/bbb64f/lights/196830452) from 192.168.1.128
22:39:21 /lights path=/lights/196830452
22:39:21 HTP: Hue Result ({"state":{"on":false,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Unlock","modelid":"Alexa","manufacturername":"Tasmota","uniqueid":"60:01:94:bb:b6:4f:00:11-4"})
22:39:28 UDP: Packet (119)
22:39:29 UDP: Packet (119)
22:39:30 UDP: Packet (119)
22:39:31 UDP: Packet (119)
22:39:31 WIF: Checking connection...
22:39:33 UDP: Packet (119)
22:39:34 UDP: Packet (119)
22:39:34 UDP: Packet (119)
22:39:34 UDP: Packet (119)
22:39:34 UDP: Packet (119)
22:39:35 UDP: Packet (119)
22:39:36 UDP: Packet (119)

```

### TO REPRODUCE

### EXPECTED BEHAVIOUR
The friendly names wont change in the alexa app and the devices are unresponsive

### SCREENSHOTS
[image]
They are all still labeled as "Tasmota" and they say unresponsive however, I can still turn on relays 1-3 i just cant turn them off. relay 4 is completely unresponsive and wont even turn on. 

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-09-24:
Hi,

From your status 0 output:

"FriendlyName":["Open","Close","Stop","Unlock"]

So, your friendly names are already set in your Tasmota.

Please, try again to delete your device from your Alexa, and discover your Tasmota again. Thanks.

Jason2866, 2020-09-24:
When friendly names are changed the procedure asicillato described is needed.
Alexa does not recognize changed names. A new device discover is always needed in this case.

XDM525, 2020-09-24:
oh yep I kept deleting and discovering and the names finally changed. thank you.

closed 2020-09-24 by XDM525