arendst/Tasmota #9374: Support of Zigbee Device " Aqara Water Leak Sensor" Xiaomi SJCGQ11LM

SirUli, 2020-09-24:
### PROBLEM DESCRIPTION
The Zigbee Device "Water Leak Sensor" Xiaomi SJCGQ11LM can be paired without any issues but then reports scrambled data on the water detection.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos D1 mini + 
- [X] Tasmota binary firmware version number used: 8.5.0
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: PlatformIO
- [X] Flashing tools used: Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
08:07:49 CMD: Backlog Template; Module; GPIO 255
08:07:49 MQT: stat/tasmota3/RESULT = {"NAME":"Zigbee","GPIO":[255,255,255,255,255,255,255,255,255,166,255,165,255],"FLAG":15,"BASE":18}
08:07:50 MQT: stat/tasmota3/RESULT = {"Module":{"0":"Zigbee"}}
08:07:50 MQT: stat/tasmota3/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"166":"Zigbee Rx"},"GPIO14":{"0":"None"},"GPIO15":{"165":"Zigbee Tx"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:

- [X] Provide the output of this command: `Status 0`:
```
08:08:43 CMD: Status 0
08:08:43 MQT: stat/tasmota3/STATUS = {"Status":{"Module":0,"DeviceName":"tasmota3","FriendlyName":["tasmota3"],"Topic":"tasmota3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:08:43 MQT: stat/tasmota3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:19:42","StartupUTC":"2020-09-24T06:49:01","Sleep":50,"CfgHolder":4617,"BootCount":31,"BCResetTime":"2020-08-01T20:22:33","SaveCount":50,"SaveAddress":"FA000"}}
08:08:43 MQT: stat/tasmota3/STATUS2 = {"StatusFWR":{"Version":"8.5.0(tasmota)","BuildDateTime":"2020-09-24T08:48:01","Boot":6,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"376/699"}}
08:08:43 MQT: stat/tasmota3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Halley",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["04008009","2805C8000100060000005A00000000000000","00000000","00006082","00000000"]}}
08:08:43 MQT: stat/tasmota3/STATUS4 = {"StatusMEM":{"ProgramSize":683,"Free":320,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAE797","04768001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,20,21,22,23,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
08:08:44 MQT: stat/tasmota3/STATUS5 = {"StatusNET":{"Hostname":"tasmota3-2271","IPAddress":"REMOVED","Gateway":"REMOVED","Subnetmask":"REMOVED","DNSServer":"REMOVED","Mac":"FC:F5:C4:AC:28:DF","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:08:44 MQT: stat/tasmota3/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt-server","MqttPort":1883,"MqttClientMask":"tasmota3","MqttClient":"tasmota3","MqttUser":"tasmotauser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:08:44 MQT: stat/tasmota3/STATUS7 = {"StatusTIM":{"UTC":"2020-09-24T07:08:44","Local":"2020-09-24T08:08:44","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:40","Sunset":"18:43"}}
08:08:44 MQT: stat/tasmota3/STATUS10 = {"StatusSNS":{"Time":"2020-09-24T08:08:44"}}
08:08:44 MQT: stat/tasmota3/STATUS11 = {"StatusSTS":{"Time":"2020-09-24T08:08:44","Uptime":"0T00:19:43","UptimeSec":1183,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"REMOVED","BSSId":"92:2A:A8:C1:9F:8F","Channel":11,"RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```

- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_

* I have first pushed the button on the device (sends the usual data on battery and a Xiaomi_64 event (any idea what that is?)
* The i submerged the sensor in a bit of water which led to: "0500<00":"000000FF0000" being sent. I think the `"ZoneStatusChange":1` means the sensor is submerged and the `"ZoneStatusChange":0` subsequently means that is is out again. Maybe you could bring some clarity to that? Thanks in advance!

```
08:11:32 CMD: weblog 4
08:11:32 MQT: stat/tasmota3/RESULT = {"WebLog":4}
08:11:33 CFG: Saved to flash at F9, Count 51, Bytes 4096
08:11:34 ZIG: {"ZbZNPReceived":"4481000000004A530101006500DCF63F0000431C5F11090A050042156C756D692E73656E736F725F776C65616B2E61713101FF42220121EF0B0328210421A81305210C0006240400010000082104020A2100006410004A531D"}
08:11:34 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0x534A","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":101,"securityuse":0,"seqnumber":0,"fc":"0x1C","manuf":"0x115F","transact":9,"cmdid":"0x0A","payload":"050042156C756D692E73656E736F725F776C65616B2E61713101FF42220121EF0B0328210421A81305210C0006240400010000082104020A210000641000"}}
08:11:34 ZIG: ZbZCLRawReceived: {"0x534A":{"0000/0005":"lumi.sensor_wleak.aq1","0000/FF01":"0121EF0B0328210421A81305210C0006240400010000082104020A210000641000","Endpoint":1,"LinkQuality":101}}
08:11:34 MQT: tele/tasmota3/534A/SENSOR = {"ZbReceived":{"0x534A":{"Device":"0x534A","ModelId":"lumi.sensor_wleak.aq1","BatteryVoltage":3.05,"BatteryPercentage":100,"Xiaomi_64":0,"Endpoint":1,"LinkQuality":101}}}
08:11:44 WIF: Checking connection...
08:11:48 ZIG: {"ZbZNPReceived":"4481000000054A530101003B005DA240000009190A00010000FF00004A531D"}
08:11:48 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0x534A","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":59,"securityuse":0,"seqnumber":0,"fc":"0x19","manuf":"0x0000","transact":10,"cmdid":"0x00","payload":"010000FF0000"}}
08:11:48 ZIG: ZbZCLRawReceived: {"0x534A":{"0500<00":"010000FF0000","ZoneStatusChange":1,"Endpoint":1,"LinkQuality":59}}
08:11:48 MQT: tele/tasmota3/534A/SENSOR = {"ZbReceived":{"0x534A":{"Device":"0x534A","0500<00":"010000FF0000","ZoneStatusChange":1,"Endpoint":1,"LinkQuality":59}}}
08:12:00 ZIG: {"ZbZNPReceived":"4481000000054A530101009E00AE2E41000009190B00000000FF00004A531D"}
08:12:00 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0x534A","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":158,"securityuse":0,"seqnumber":0,"fc":"0x19","manuf":"0x0000","transact":11,"cmdid":"0x00","payload":"000000FF0000"}}
08:12:00 ZIG: ZbZCLRawReceived: {"0x534A":{"0500<00":"000000FF0000","ZoneStatusChange":0,"Endpoint":1,"LinkQuality":158}}
08:12:00 MQT: tele/tasmota3/534A/SENSOR = {"ZbReceived":{"0x534A":{"Device":"0x534A","0500<00":"000000FF0000","ZoneStatusChange":0,"Endpoint":1,"LinkQuality":158}}}
```

### TO REPRODUCE

Connect [SJCGQ11LM](https://zigbee.blakadder.com/Xiaomi_SJCGQ11LM.html) to a Tasmota with ZigBee enabled.

### EXPECTED BEHAVIOUR

Water leak status is available.

### SCREENSHOTS

None.

### ADDITIONAL CONTEXT

None.

blakadder, 2020-09-24:
What are the "scrambled data" and what are you expectations on what it is supposed to report?

SirUli, 2020-09-24:
Sorry @blakadder you are right, so what my report is about:

I assume the ``"0500<00":"000000FF0000"`` and ``"0500<00":"010000FF0000"`` is supposed to tell something which isn't properly translated - i assume that isn't intended to be coming that way to the user and might required a fix. From what i read on various soruces, the ZoneStatusChange apparently means e.g. open or closed  in case of window contacts. That is why i deducted, that it might be "in water" or "out of water". Is that actually the case or should my payload be in ``"0500<00":"000000FF0000"`` for "not in water" and ``"0500<00":"010000FF0000"`` "in water" and just wasn't properly translated?

Thanks in advance!

blakadder, 2020-09-24:
you are correct, the payload corresponds to ZCL specs and the first byte is the status. technically it is properly translated and that translation is the ZoneStatusChange

SirUli, 2020-09-24:
Thanks @blakadder for your reply - wonderful :) 

So that leaves us with the point of not reporting the payload if it is translated anyway. Maybe here you or @s-hadinger can confirm this as intended or not?

blakadder, 2020-09-24:
it is intended as of this date. technically it is useable but maybe it will be improved upon in time, there's a lot of development going on in the zigbee space

SirUli, 2020-09-24:
Sweet - that closes this Issue then. Thanks so much for your help and that you have invested your precious time into this 👍

closed 2020-09-24 by SirUli

s-hadinger, 2020-09-25:
@SirUli as mentioned by blakadder, the plan is to have automatic sensor detection at pairing time (PIR, leak detector, gas detector...) and adapt the message accordingly. For now you will need to live with `ZoneStatusChange`

SirUli, 2020-09-25:
Thanks @s-hadinger for this comment, that clarifies the idea :)