arendst/Tasmota #9376: Problem with Alexa integration Tasmota 8.5.0.1

cybero2912, 2020-09-24:
Hello,
hopefully I collected all needed informations below
Kind Regards Robert

### PROBLEM DESCRIPTION
I upgraded two S20 to the latest Tasmota 8.5.0.1
unfortunatelly Alexa can't find them :(
I already enabled Wemo emulation and tried to search for new devices many times. I searched using Echo Dot and also using a Web Browser. The two S20 are reachable via Webif fine, I can toggle ON<>OFF fine and everything works well except Alexa intergration. Anybody else facing this problem with the latest Tasmota 8.5.0.1 ?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

all informations read before posting this and problem is only with Alexa detection/integration
I have other Tasmota enabled devices which are working with Alexa
```
  Configuration output here:
`Tasmota Version	8.5.0.1(tasmota)
Build-Datum & -Uhrzeit	2020.09.24 10:08:53
Core-/SDK-Version	2_7_4_1/2.2.2-dev(38a443e)
Laufzeit	0T00:15:32
Anz. Flash Schreibzugriffe	111 at 0xF6000
Anzahl Startvorgänge	31
Grund für Neustart	Power On
Name [friendly name] 1	S20-08
	 
AP1 SSID (RSSI)	************ (84%, -58 dBm)
Hostname	S20-08
MAC-Adresse	DC:4F:22:**:**:**
IP-Adresse (wifi)	192.168.*.*
	
Gateway	192.168.*.*
Subnetzmaske	255.255.255.0
DNS-Server	192.168.*.*
	 
MQTT Host	
MQTT Port	1883
MQTT Benutzer	DVES_USER
MQTT Client	DVES_37684C
MQTT Topic	tasmota_%06X
MQTT Group Topic 1	cmnd/tasmotas/
MQTT Full Topic	cmnd/tasmota_37684C/
MQTT Fallback-Topic	cmnd/DVES_37684C_fb/
MQTT No Retain	deaktiviert
	 
Emulation	Belkin WeMo
mDNS-Ermittlung	deaktiviert
	 
ESP Chip ID	3631180
Flash Chip ID	0x14405E
Realer Flash Speicher	1024kB
Ges. Flash Speicher	1024kB
Ben. Flash Speicher	599kB
Verf. Flash Speicher	404kB
Freier Arbeitsspeicher	26kB`

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
no rules used

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
`15:12:03 CMD: Status 0
15:12:03 RSL: stat/tasmota_37684C/STATUS = {"Status":{"Module":8,"DeviceName":"S20-08","FriendlyName":["S20-08"],"Topic":"tasmota_37684C","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:12:03 RSL: stat/tasmota_37684C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-DE.bin","RestartReason":"Power On","Uptime":"0T00:22:05","StartupUTC":"2020-09-24T13:49:58","Sleep":50,"CfgHolder":4617,"BootCount":31,"BCResetTime":"2020-05-14T12:40:04","SaveCount":111,"SaveAddress":"F6000"}}
15:12:03 RSL: stat/tasmota_37684C/STATUS2 = {"StatusFWR":{"Version":"8.5.0.1(tasmota)","BuildDateTime":"2020.09.24 10:08:53","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"373/699"}}
15:12:03 RSL: stat/tasmota_37684C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["**********",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
15:12:03 RSL: stat/tasmota_37684C/STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
15:12:03 RSL: stat/tasmota_37684C/STATUS5 = {"StatusNET":{"Hostname":"S20-08","IPAddress":"192.168.*.*","Gateway":"192.168.*.*","Subnetmask":"255.255.255.0","DNSServer":"192.168.*.*","Mac":"DC:4F:22:**:**:**","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:12:03 RSL: stat/tasmota_37684C/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_37684C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:12:03 RSL: stat/tasmota_37684C/STATUS7 = {"StatusTIM":{"UTC":"2020-09-24T14:12:03","Local":"2020-09-24T15:12:03","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:40","Sunset":"18:43"}}
15:12:03 RSL: stat/tasmota_37684C/STATUS10 = {"StatusSNS":{"Time":"2020-09-24T15:12:03"}}
15:12:03 RSL: stat/tasmota_37684C/STATUS11 = {"StatusSTS":{"Time":"2020-09-24T15:12:03","Uptime":"0T00:22:05","UptimeSec":1325,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"**********","BSSId":"00:3A:9A:**:**:**","Channel":4,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}}`

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
NA
`15:20:46 CMD: weblog 4
15:20:46 RSL: stat/tasmota_37684C/RESULT = {"WebLog":4}
15:20:46 CFG: in Flash gespeichert am F5, zählen 112, Bytes 4096`

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
install Tasmota 8.5.0.1 on Sonoff S20 enable Wemo emulation and try to integrate with Alexa

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
integration working in Alexa and new devices are detected

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
NA

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
NA

**(Please, remember to close the issue when the problem has been addressed)**
yes sure

s-hadinger, 2020-09-24:
What kind of Echo Dot do you have. If it's a 2ng Gen, then you now need to enable `SetOption109 1`

Jason2866, 2020-09-24:
I thought for WeMo the `SetOption109` 1 is not needed?

cybero2912, 2020-09-24:
Hello,
all my othe Tasmota device are with a dash in the friendly name (S20-01, S20-02, S20-03....) but I can try with another name without the dash.
I have both generations of the Dot - Gen2 and Gen3 and both are not able to find them

Regrads Robert

s-hadinger, 2020-09-24:
@Jason2866 Oh yes you're right. It's not needed for Wemo. I misread the issue.

The best approach is to enable `Weblog 4` and paste the logs here.

cybero2912, 2020-09-24:
Hello,
I changed friendly name and to a one without a dash but Alexa still does not find a new device

cybero2912, 2020-09-24:
the two problem devices with latest Tasmota are already deleted in Alexa, that was where the problem starts

ascillato, 2020-09-24:
Please, post the Tasmota console logs when you are discovering using Alexa.

cybero2912, 2020-09-24:
nothing in console during discovery :(
but it seems that the S20 is sending Wemo responses to various devices in my network:
`16:24:00 UPP: WeMo Type 2, Antwort gesendet zu 192.168.**.***:1900
16:24:16 WIF: Prüfe Verbindung...
16:24:36 WIF: Prüfe Verbindung...
16:24:56 WIF: Prüfe Verbindung...
16:25:16 WIF: Prüfe Verbindung...
16:25:36 WIF: Prüfe Verbindung...
16:25:56 WIF: Prüfe Verbindung...
16:26:16 WIF: Prüfe Verbindung...
16:26:36 WIF: Prüfe Verbindung...
16:26:56 WIF: Prüfe Verbindung...
16:27:16 WIF: Prüfe Verbindung...
16:27:36 WIF: Prüfe Verbindung...
16:27:56 WIF: Prüfe Verbindung...
16:28:16 WIF: Prüfe Verbindung...`

Jason2866, 2020-09-24:
If you have no UDP traffic messages in the Tasmota console the device does not get a discover UDP message. You have to check your network

s-hadinger, 2020-09-24:
Actually there is UDP traffic
`16:24:00 UPP: WeMo Type 2, Antwort gesendet zu 192.168.**.***:1900`

Is the source IP address one of your Echo device? Do you have multiple addresses showing up?

ascillato2, 2020-09-25:
@cybero2912 

Hi, any news on this?

ascillato2, 2020-09-26:
Please, when you have time, tell us is the source IP address one of your Echo device? Do you have multiple addresses showing up? Thanks.

closed 2020-09-26 by ascillato2

cybero2912, 2020-09-28:
Hello,
this is not the IP Of the Echo device and yes there are multiple addresses showing up, but none of them belongs to one of my Echos

s-hadinger, 2020-09-28:
So that's your problem. Something is happening on your local network. Disconnect all computers and try again

cybero2912, 2020-09-28:
I have approximately 50-60 devices connected to my network  :(

s-hadinger, 2020-09-28:
Not devices, computers. You may have installed software that interferes. Some users already reported similar issues with PC software installed

Jason2866, 2020-09-28:
Yes, i have to kill the Alexa App on my phones and tablets too. If one App is active discovery fails (for me).

cybero2912, 2020-09-30:
Hello,
one of the IP addresses does not belong to a computer it is the address of a fing box ;-)

cybero2912, 2020-09-30:
btw. thank you for closing this issue without a feedback from me
do you need to keep your bugtracker clean?

s-hadinger, 2020-09-30:
Nothing shows that it's a Tasmota issue and everything seems to indicate it's a local network ussue. I don't see the point in keeping this issue open except if you find an issue in Tasmota.

ymurawski, 2020-10-24:
Hello, I have the exact same behavior since 8.5.1....

When I change to hue emulator everything works perfectly.... But wemo can not find by alexa.