arendst/Tasmota #9379: Beca BAC-002-ALW fan-coil thermostat MCU commands (heating + cooling version, 2 pipes)

cyraxxx, 2020-09-24:
Hi,

I flashed the current development edition on this device and tested the dpId-s and values, based on an other firmware ( https://github.com/klausahrenberg/WThermostatBeca )

I can confirm the commands are the same that I found in the docs ( https://tasmota.github.io/docs/TuyaMCU/#becathermostat ), except for dpID 102. I would like you to extend the documentation with the following:

dpID 5:  "Eco mode", Issue and report, Function type 1 (boolean, Tuyasend1)
dpID 6:  "Locked mode", Issue and report, Function type 1 (boolean, Tuyasend1)
dpID 102: "System mode", Issue and report, Function type 4 (enum?, Tuyasend4), options: 0 cooling, 1 heating, 2 ventilating
dpID 103: "Fan speed", Issue and report, Function type 4 (enum?, Tuyasend4), options: 0 auto, 1 high, 2 medium, 3 low

Image of thermostat:
[image]

Image of ESP wired for flashing:
[image]

And also I think in the case of this thermostat the new fan controller options (61-64) cannot be used, as here the lowest fan speed is 3, not 0 (which is auto).

closed 2020-09-24 by close-issue-app[bot]

martin072, 2021-11-27:
@cyraxxx Did you get anywhere with this. I just got one of these devices (the BAC-002) and flashed it. Would like to run Tasmota on this. Did you get it up and running?

cyraxxx, 2021-11-27:
> @cyraxxx Did you get anywhere with this. I just got one of these devices (the BAC-002) and flashed it. Would like to run Tasmota on this. Did you get it up and running?

Yes, absolutely, that is how I was able to confirm the commands. Tasmota was fully functional and working well, however a bit overkill for my usecase (wanted to command 4 devices with MQTT) and felt complicated (for example, if you want to set 20C, you need to send TuyaSend2 2,40 - 2 is the target temperature's code, 40 is 2 times 20C, or TuyaSend4 102,0 for turning on cooling, TuyaSend4 102,1 for heating, etc..).  I was able to do that using Node-Red as converter but it wasn't very convenient. So after it I decided to move to an other firmware which is not that flexible in terms of Tasmota but very easy to use for MQTT commands (fashberg_wthermostat-1.19.beta1-fas)