arendst/Tasmota #9383: Zigbee Sonoff ZBBridge is dead?

abalashov78, 2020-09-25:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I made like on this page https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html?m=1
with both latest tasmota-zbbridge.bin  release and developer versions
but I can't pair any devices with Sonoff ZBBridge with Tasmota FW.
I tested with both file ncp-uart-sw_6.7.6_115200.ota  or ncp-uart-sw_6.5.5_115200.ota 

in log i have only this message 

2020-09-25 08:37:30 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'665721575485ea4da66bb5da55aac820fb0c42ceaca20be29ca800eccc47fb2c7e'
2020-09-25 08:37:30 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'6757a15754851563917e'
2020-09-25 08:37:30 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2020-09-25 08:37:30 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 175 (addTransientLinkKey) received: b'00'

in tasmota console:
06:28:10 TCP: from MCU: 5646A1575479158BF57E
06:28:10 TCP: to MCU/1: 8610BE7E6647215754790FA25954927E
06:28:10 TCP: from MCU: 6747A1575479152FB97E
06:28:11 TCP: to MCU/1: 87009F7E
06:28:11 TCP: to MCU/1: 7744215754797D33A25912387E
06:28:11 TCP: from MCU: 7044A157547915CF667E
06:28:11 TCP: to MCU/1: 8070787E
06:28:11 TCP: to MCU/1: 0045215754797D38B75943EC7E
06:28:11 TCP: from MCU: 0145A157
06:28:11 TCP: from MCU: 547915B6C67E
06:28:11 TCP: to MCU/1: 8160597E
06:28:11 TCP: to MCU/1: 7D314A2157547937B0597D31937E
06:28:11 CFG: Saved to flash at F6, Count 16, Bytes 4096
06:28:11 TCP: from MCU: 124AA1575479159C3C7E
06:28:11 TCP: to MCU/1: 82503A7E224B2157547910A259E9417E
06:28:12 TCP: from MCU: 234BA1575479
06:28:12 TCP: from MCU: 1538707E
06:28:12 TCP: to MCU/1: 83401B7E
06:28:12 TCP: to MCU/1: 3348215754790CB059CAFA7E
06:28:12 TCP: from MCU: 3448A157547915D8
06:28:12 TCP: from MCU: AF7E
06:28:12 TCP: to MCU/1: 8430FC7E
06:28:12 TCP: to MCU/1: 44492157547919B0593FCA7E
06:28:12 TCP: from MCU: 4549A157547915A10F7E
06:28:12 TCP: to MCU/1: 8520DD7E
06:28:12 TCP: to MCU/1: 554E2157547907B2478DF07E
06:28:12 TCP: from MCU: 564EA15754791586B77E
06:28:12 TCP: to MCU/1: 8610BE7E
06:28:12 TCP: to MCU/1: 664F215754793FB159D9CD7E
06:28:12 TCP: from MCU: 674FA1
06:28:12 TCP: from MCU: 5754791522FB7E
06:28:13 TCP: to MCU/1: 87009F7E
06:28:13 TCP: to MCU/1: 774C21575479144D5989FE7E
06:28:13 TCP: from MCU: 704CA157547915
06:28:13 TCP: from MCU: C2247E
06:28:13 TCP: to MCU/1: 8070787E
06:28:13 TCP: to MCU/1: 004D215754781622677E
06:28:13 TCP: from MCU: 014DA1575478
06:28:13 TCP: from MCU: 15B85939377E
06:28:13 TCP: to MCU/1: 8160597E
06:28:13 TCP: to MCU/1: 7D31522157542814B6587BF425AA54924C3E0B7E
06:28:13 TCP: from MCU: 1252A157542815B6747E
06:28:13 TCP: to MCU/1: 82503A7E
06:28:13 TCP: to MCU/1: 2253215754211422307E
06:28:13 TCP: from MCU: 2353A157542105
06:28:13 TCP: from MCU: 4DA66BB5DA55AA6DB663B1D8541231986B317E
06:28:14 TCP: to MCU/1: 83401B7E
06:28:14 TCP: to MCU/1: 3350215754211753A97E
06:28:14 TCP: from MCU: 3450A1575421054D
06:28:14 TCP: from MCU: A66BB5DA55AA6DB663B1D854123198C63F7E
06:28:14 TCP: to MCU/1: 8430FC7E
06:28:14 TCP: to MCU/1: 4451215754800487317E
06:28:14 TCP: from MCU: 4551A157548015B5E9954C20AF55384F2C7E
06:28:14 TCP: to MCU/1: 8520DD7E
[... 97 lines cut ...]
06:28:58 TCP: from MCU: 6757A157548515
06:28:58 TCP: from MCU: 63917E
06:28:58 TCP: to MCU/1: 87009F7E

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

use https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html?m=1 
with latest tasmota-zbbridge.bin  and ncp-uart-sw_6.7.6_115200.ota  or ncp-uart-sw_6.5.5_115200.ota 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

06:26:37 RSL: tele/tasmota_DBF93B/INFO1 = {"Module":"Sonoff ZbBridge","Version":"8.5.0.1(zbbridge)","FallbackTopic":"cmnd/DVES_DBF93B_fb/","GroupTopic":"cmnd/tasmotas/"}
06:26:37 RSL: tele/tasmota_DBF93B/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_DBF93B-6459","IPAddress":"192.168.1.155"}

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-25:
Please use no template. Firmware has all settings and functions (per default) which are needed to flash the EFR32 MCU with the ncp firmware. Do a `reset 1` and flash the NCP firmware again. After reset you should see similar (`weblog 4` enabled)

```
16:50:37 ZIG: Resetting EZSP device
16:50:38 MQT: tele/sonoff-064A39/RESULT = {"ZbState":{"Status":1,"Message":"EFR32 booted","RestartReason":"Power-on","Code":2}}
16:50:38 ZIG: ZbEZSPSend 000008
16:50:38 ZIG: {"ZbEZSPReceived":"000008026067"}
16:50:38 MQT: tele/sonoff-064A39/RESULT = {"ZbState":{"Status":55,"Version":"6.7.6.0","Protocol":8,"Stack":2}}
16:50:38 MQT: tele/sonoff-064A39/RESULT = {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}}
16:50:38 ZIG: ZbEZSPSend 53001E0200
16:50:38 ZIG: {"ZbEZSPReceived":"530000"}
16:50:38 ZIG: ZbEZSPSend 5300061000
16:50:38 ZIG: {"ZbEZSPReceived":"530000"}
16:50:38 ZIG: ZbEZSPSend 53000C0200
16:50:38 ZIG: {"ZbEZSPReceived":"530000"}
16:50:38 ZIG: ZbEZSPSend 53000D0500
16:50:38 ZIG: {"ZbEZSPReceived":"530000"}
16:50:38 ZIG: ZbEZSPSend 5300112000
16:50:38 ZIG: {"ZbEZSPReceived":"530000"}
16:50:38 ZIG: ZbEZSPSend 530012001E
16:50:38 ZIG: {"ZbEZSPReceived":"530000"}
16:50:38 ZIG: ZbEZSPSend 5300190200
16:50:38 ZIG: {"ZbEZSPReceived":"530000"}
16:50:38 ZIG: ZbEZSPSend 53001A1000
16:50:38 ZIG: {"ZbEZSPReceived":"530000"}
16:50:38 ZIG: ZbEZSPSend 53001E0400
16:50:39 ZIG: {"ZbEZSPReceived":"530000"}
16:50:39 ZIG: ZbEZSPSend 5300220200
16:50:39 ZIG: {"ZbEZSPReceived":"530000"}
16:50:39 ZIG: ZbEZSPSend 53002A0300
16:50:39 ZIG: {"ZbEZSPReceived":"530000"}
16:50:39 ZIG: ZbEZSPSend 53002D0100
16:50:39 ZIG: {"ZbEZSPReceived":"530000"}
16:50:39 ZIG: ZbEZSPSend 530001FF00
16:50:39 ZIG: {"ZbEZSPReceived":"000000"}
16:50:39 ZIG: ZbEZSPSend 02000104010500000000
16:50:39 ZIG: {"ZbEZSPReceived":"020000"}
16:50:39 ZIG: ZbEZSPSend 02000B04010500000000
16:50:39 ZIG: {"ZbEZSPReceived":"020000"}
16:50:39 ZIG: ZbEZSPSend 100000F9FF58020807020500
16:50:39 ZIG: {"ZbEZSPReceived":"100000"}
16:50:39 ZIG: ZbEZSPSend 55000003
16:50:39 ZIG: {"ZbEZSPReceived":"550000"}
16:50:39 ZIG: ZbEZSPSend 55000220
16:50:39 ZIG: {"ZbEZSPReceived":"550000"}
16:50:39 ZIG: ZbEZSPSend 55000330
16:50:39 ZIG: {"ZbEZSPReceived":"550000"}
16:50:39 ZIG: ZbEZSPSend 55000551
16:50:39 ZIG: {"ZbEZSPReceived":"550000"}
16:50:39 ZIG: ZbEZSPSend 55000660
16:50:39 ZIG: {"ZbEZSPReceived":"550000"}
16:50:39 ZIG: ZbEZSPSend 17000000
16:50:39 ZIG: {"ZbEZSPReceived":"170093"}
16:50:39 MQT: tele/sonoff-064A39/RESULT = {"ZbState":{"Status":2,"Message":"Resetting configuration"}}
16:50:50 ZIG: Resetting EZSP device
16:50:51 MQT: tele/sonoff-064A39/RESULT = {"ZbState":{"Status":1,"Message":"EFR32 booted","RestartReason":"Power-on","Code":2}}
16:50:51 ZIG: ZbEZSPSend 000008
16:50:51 ZIG: {"ZbEZSPReceived":"000008026067"}
16:50:51 MQT: tele/sonoff-064A39/RESULT = {"ZbState":{"Status":55,"Version":"6.7.6.0","Protocol":8,"Stack":2}}
16:50:51 MQT: tele/sonoff-064A39/RESULT = {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}}
16:50:51 ZIG: ZbEZSPSend 53001E0200
16:50:51 ZIG: {"ZbEZSPReceived":"530000"}
16:50:51 ZIG: ZbEZSPSend 5300061000
16:50:51 ZIG: {"ZbEZSPReceived":"530000"}
16:50:51 ZIG: ZbEZSPSend 53000C0200
16:50:51 ZIG: {"ZbEZSPReceived":"530000"}
16:50:51 ZIG: ZbEZSPSend 53000D0500
16:50:51 ZIG: {"ZbEZSPReceived":"530000"}
16:50:51 ZIG: ZbEZSPSend 5300112000
16:50:51 ZIG: {"ZbEZSPReceived":"530000"}
16:50:51 ZIG: ZbEZSPSend 530012001E
16:50:51 ZIG: {"ZbEZSPReceived":"530000"}
16:50:51 ZIG: ZbEZSPSend 5300190200
16:50:52 ZIG: {"ZbEZSPReceived":"530000"}
16:50:52 ZIG: ZbEZSPSend 53001A1000
16:50:52 ZIG: {"ZbEZSPReceived":"530000"}
16:50:52 ZIG: ZbEZSPSend 53001E0400
16:50:52 ZIG: {"ZbEZSPReceived":"530000"}
16:50:52 ZIG: ZbEZSPSend 5300220200
16:50:52 ZIG: {"ZbEZSPReceived":"530000"}
16:50:52 ZIG: ZbEZSPSend 53002A0300
16:50:52 ZIG: {"ZbEZSPReceived":"530000"}
16:50:52 ZIG: ZbEZSPSend 53002D0100
16:50:52 ZIG: {"ZbEZSPReceived":"530000"}
16:50:52 ZIG: ZbEZSPSend 530001FF00
16:50:52 ZIG: {"ZbEZSPReceived":"000000"}
16:50:52 ZIG: ZbEZSPSend 02000104010500000000
16:50:52 ZIG: {"ZbEZSPReceived":"020000"}
16:50:52 ZIG: ZbEZSPSend 02000B04010500000000
16:50:52 ZIG: {"ZbEZSPReceived":"020000"}
16:50:52 ZIG: ZbEZSPSend 100000F9FF58020807020500
16:50:52 ZIG: {"ZbEZSPReceived":"100000"}
16:50:52 ZIG: ZbEZSPSend 55000003
16:50:52 ZIG: {"ZbEZSPReceived":"550000"}
16:50:52 ZIG: ZbEZSPSend 55000220
16:50:52 ZIG: {"ZbEZSPReceived":"550000"}
16:50:52 ZIG: ZbEZSPSend 55000330
16:50:52 ZIG: {"ZbEZSPReceived":"550000"}
16:50:52 ZIG: ZbEZSPSend 55000551
16:50:52 ZIG: {"ZbEZSPReceived":"550000"}
16:50:52 ZIG: ZbEZSPSend 55000660
16:50:52 ZIG: {"ZbEZSPReceived":"550000"}
16:50:52 ZIG: ZbEZSPSend 68000C035A6967426565416C6C69616E63653039A1403D4AC82B9606A1403D4AC82B9606000000000000000000
16:50:52 ZIG: {"ZbEZSPReceived":"680000"}
16:50:52 ZIG: ZbEZSPSend 1E00C82B9606CCCCCCCCC82B140B00FFFF0000000000
16:50:53 ZIG: {"ZbEZSPReceived":"1E0000"}
16:50:53 ZIG: {"ZbEZSPReceived":"190090"}
16:50:53 ZIG: ZbEZSPSend 2600
16:50:53 ZIG: {"ZbEZSPReceived":"2600D4A642FEFF23A460"}
16:50:53 ZIG: ZbEZSPSend 2700
16:50:53 ZIG: {"ZbEZSPReceived":"27000000"}
16:50:53 ZIG: Subscribe to group 0 'ZbListen0 0'
16:50:53 ZIG: ZbEZSPSend 64000000000100
16:50:53 ZIG: {"ZbEZSPReceived":"640000"}
16:50:53 MQT: tele/sonoff-064A39/RESULT = {"ZbState":{"Status":0,"Message":"Started"}}
16:50:53 ZIG: Zigbee started
16:50:53 ZIG: Zigbee signature in Flash: 3167697A - 59
16:50:53 ZIG: Zigbee devices data in Flash (59 bytes)
```

If zigbee started appears the NCP firmware is working.

abalashov78, 2020-09-25:
> 
> 
> Please use no template. Firmware has all settings and functions (per default) which are needed to flash the EFR32 MCU with the ncp firmware. Do a `reset 1` and flash the NCP firmware again. After reset you should see similar (`weblog 4` enabled)
> 

> If zigbee started appears the NCP firmware is working.

Thank you for answer. 

After resetting device i still can't pair my snzb-02 with zbbrudge

Resetting EZSP device
07:41:00 RSL: tele/tasmota_DBF93B/RESULT = {"ZbState":{"Status":1,"Message":"EFR32 booted","RestartReason":"Power-on","Code":2}}
07:41:00 RSL: tele/tasmota_DBF93B/RESULT = {"ZbState":{"Status":55,"Version":"6.7.6.0","Protocol":8,"Stack":2}}
07:41:00 RSL: tele/tasmota_DBF93B/RESULT = {"ZbState":{"Status":3,"Message":"Configured, starting coordinator"}}
07:41:02 ZIG: {"ZbEZSPReceived":"1E0000"}
07:41:02 ZIG: Subscribe to group 0 'ZbListen0 0'
07:41:02 RSL: tele/tasmota_DBF93B/RESULT = {"ZbState":{"Status":0,"Message":"Started"}}
07:41:02 ZIG: Zigbee started
07:41:02 ZIG: No zigbee devices data in Flash
07:41:42 CMD: ZbPermitJoin 1
07:41:42 RSL: tele/tasmota_DBF93B/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
07:41:42 RSL: stat/tasmota_DBF93B/RESULT = {"ZbPermitJoin":"Done"}
07:41:42 ZIG: Internal ZDO message 0x0036 sent from 0x0000 (broadcast)

Here i push button on my device for pairing

07:42:42 RSL: tele/tasmota_DBF93B/RESULT = {"ZbState":{"Status":20,"Message":"Pairing mode disabled"}}
07:42:59 CMD: ZbConfig 
07:42:59 RSL: stat/tasmota_DBF93B/RESULT = {"ZbConfig":{"Channel":11,"PanID":"0x0FA4","ExtPanID":"0xCCCCCCCCDB12CFA4","KeyL":"0xDB12CFA4F93F40A1","KeyH":"0xDB12CFA4F93F40A1","TxRadio":20}}

and i have no pairing device

abalashov78, 2020-09-25:
It's magic
Few minutes later I try pairing device again and it's work!
Thank you so mutch! 

Using no template is correct colution for my problem.

abalashov78, 2020-09-25:
But i still have problem with ZHA's integration in HA.
Next question is not clear for me. 

Zbbridge is work fine with ncp-uart-sw_6.7.6_115200.ota  and i pair device. 
But with ncp-uart-sw_6.5.5_115200.ota in Home Assistant i can't pair it.

2020-09-25 10:19:30 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'477121575408290d457e'
2020-09-25 10:19:30 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'7571a157540815e1247e'
2020-09-25 10:19:30 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2020-09-25 10:19:30 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 34 (permitJoining) received: b'00'
2020-09-25 10:19:39 DEBUG (MainThread) [bellows.ezsp.protocol] Send command nop: ()
2020-09-25 10:19:39 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'50762157542f23d57e'
2020-09-25 10:19:39 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'0676a157542f03987e'
2020-09-25 10:19:39 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2020-09-25 10:19:39 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 5 (nop) received: b''
2020-09-25 10:19:49 DEBUG (MainThread) [bellows.ezsp.protocol] Send command nop: ()
2020-09-25 10:19:49 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'61772157542fe3a87e'
2020-09-25 10:19:49 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'1777a157542ff6ed7e'
2020-09-25 10:19:49 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2020-09-25 10:19:49 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 5 (nop) received: b''

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

backlog rule1 on system#boot do TCPStart 8888 endon ; rule1 1 ; template {"NAME":"Sonoff ZHABridge","GPIO":[56,208,0,209,59,58,0,0,0,0,0,0,17],"FLAG":0,"BASE":18} ; module 0

if i don't use template - i can't start tcp server on 8888 for connect HA to bridge.

s-hadinger, 2020-09-25:
@abalashov78 It looks like the issue is now on HA side. I'm afraid we can't help here. I suggest you post the issue in their Github site.

abalashov78, 2020-09-25:
correct template for latest versions of HA is:

backlog rule1 on system#boot do TCPStart 8888 endon ; rule1 1 ; template {"NAME":"ZHA ZBBridge","GPIO":[56,208,0,209,59,58,0,0,0,0,0,0,17],"FLAG":0,"BASE":18} ; module 0

ascillato2, 2020-09-25:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-25 by ascillato2

lehighkid, 2021-01-12:
I know this is closed but did want to share that I had the same exact issue following the same guide.  I was not able to pair any devices either via Tasmota or HA.  I was concerned my hardware was damaged after trying different firmware versions, repeating the process verbatim.  Eventually I found it wasn't after finding this thread.  

I started over completely - refreshed Tasmota via serial (likely not necessary) - did not use a template, flashed the radio firmware and then used this template above.  

```backlog rule1 on system#boot do TCPStart 8888 endon ; rule1 1 ; template {"NAME":"ZHA ZBBridge","GPIO":[56,208,0,209,59,58,0,0,0,0,0,0,17],"FLAG":0,"BASE":18} ; module 0```

```Weblog 4``` was very helpful as well to confirm ZB radio function

All is working well now via HA - much, much better than my HUSBZ1 stick in terms of range, responsiveness and so far stickiness.  Thank you.