arendst/Tasmota #9385: Tasmota 8.5.1 Script Problem

renegat56, 2020-09-25:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
the script works for 2 or 3 times before it is disabled automatically (in the web gui the enable box is unchecked again)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): ESP-01
- [x ] Tasmota binary firmware version number used: 8.5.0.1
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [ ] IDE / Compiler used: VSCode PlatformIO 0.5
- [x ] Flashing tools used: IDE
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
when it works:

00:00:00 CFG: Loaded from flash at F7, Count 125
00:00:00 QPC: Count 1
00:00:00 Script: nv=0, tv=1, vns=6, ram=45
00:00:00  RESULT
00:00:00 Project iris Iris Version 8.5.0.1(tasmota)-2_7_4_1
00:00:00 Script: performs "deepsleeptime 0"
00:00:00 RSL: /homenet/stat/iris/RESULT = {"DeepSleepTime":0}
00:00:00 Script: performs "power1 1"
00:00:00 RSL: /homenet/stat/iris/RESULT = {"POWER1":"ON"}
00:00:00 RSL: /homenet/stat/iris/POWER1 = ON
00:00:00  BOOT
00:00:00 WIF: Connecting to AP1 Komiker Channel 13 BSSId 34:31:C4:3F:51:D0 in mode 11N as iris-4226...
00:00:01  RESULT
00:00:01 RSL: /homenet/stat/iris/RESULT = {"POWER1":"OFF"}
00:00:01 RSL: /homenet/stat/iris/POWER1 = OFF
00:00:01  RESULT
00:00:01 RSL: /homenet/stat/iris/RESULT = {"POWER2":"OFF"}
00:00:01 RSL: /homenet/stat/iris/POWER2 = OFF
00:00:01 WIF: Connected
10:56:37 HTP: Web server active on iris-4226 with IP address 192.168.0.157
10:56:37 UPP: Multicast (re)joined
10:56:37 Script: performs "power2 1"
10:56:37 RSL: /homenet/stat/iris/RESULT = {"POWER2":"ON"}
10:56:37 RSL: /homenet/stat/iris/POWER2 = ON
10:56:37 Script: performs "websend [192.168.0.78:778] /dophp/iris.php?do1=1&do2=1&ubat=3.30"
10:56:37 RSL: /homenet/stat/iris/RESULT = DO1=1DO2=1UBAT=3.30
10:56:37  RESULT
10:56:37 RSL: /homenet/stat/iris/RESULT = {"WebSend":"Done"}
10:56:37  RESULT
10:56:37 Script: performs "deepsleeptime 30"
10:56:37 RSL: /homenet/stat/iris/RESULT = {"DeepSleepTime":30}
10:56:37 DSL: Reset wrong settings wakeup: 0, slip 10000
10:56:37 RSL: /homenet/stat/iris/RESULT = {"DeepSleep":{"Time":"2020-09-25T10:57:00","Epoch":1601024220}}
10:56:38 MQT: Attempting connection...
10:56:38 MQT: Connected
10:56:38 MQT: /homenet/tele/iris/LWT = Online (retained)
10:56:38 MQT: /homenet/cmnd/iris/POWER = 
10:56:38 MQT: /homenet/stat/iris/RESULT = {"POWER1":"OFF"}
10:56:38 MQT: /homenet/stat/iris/POWER1 = OFF
10:56:38 MQT: /homenet/stat/iris/RESULT = {"POWER2":"ON"}
10:56:38 MQT: /homenet/stat/iris/POWER2 = ON
10:56:38 MQT: homeassistant/discovery/483FDA00D082/config = {"ip":"192.168.0.157","dn":"Iris","fn":["fenster dunkel","fenster hell",null,null,null,null,null,null],"hn":"iris-4226","mac":"483FDA00D082","md":"IRIS","ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"8.5.0.1","t":"iris","ft":"/homenet/%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,1,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"btn":[0,0,0,0],"so":{"11":0,"13":0,"17":0,"20":0,"30":0,"37":1,"68":0,"73":0,"80":0},"lt_st":0,"ver":1} (retained)
10:56:38  RESULT
10:56:38 MQT: /homenet/stat/iris/RESULT = {"POWER2":"OFF"}
10:56:38 MQT: /homenet/stat/iris/POWER2 = OFF
10:56:39 UPP: Multicast (re)joined

```

### TO REPRODUCE
enable script again and save

### EXPECTED BEHAVIOUR
the script stays enabled

### SCREENSHOTS
[screenshot_script_tasmota.pdf](https://github.com/arendst/Tasmota/files/5281641/screenshot_script_tasmota.pdf)

### ADDITIONAL CONTEXT
I tried with savdata 0 and restart 1, but also with savedata 1 -> same result, after script works 2 or 3 times after deepsleep, it suddenly stops, but deepsleeptime 30 is still active

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-25:
`delay(100)` is a long time.

ascillato, 2020-09-25:
Hi, you need to disable the bootloop protection.

Go to the console and type:

```lua
Setoption36 0
Setoption65 1
```

And please try again. Thanks.

kugelkopf123, 2020-09-25:
Yes. Try without Delay like some nonblocking style:
```
>D
cnt=0
>F
if cnt==0{
=>power2 1
;WEBSEND STUFF
}
if cnt==1{
=>deepsleeptime 30
}

cnt+=1

if cnt=>2{
cnt=0
}
```
And with the delay in the >E section the same. 
Also have a look on Subroutines that may help you too.

renegat56, 2020-09-25:
@ascillato thanks, the option36 0 and option65 1 seems to solve the problem. Can you please explain?
If possible, just another question : how can I use the websend response in my script (as example, I like to pulse GPIO0 low, if DO1=1 in response/RESULT topic)?
RSL: /homenet/stat/iris/RESULT = DO1=1DO2=1UBAT=3.21

ascillato, 2020-09-25:
If your device go to deepsleep and wakeup, it boots up again and if the time that is awake is less than 10 seconds, the bootloop protection will start counting failed boots cycles. When it reaches its maximum allowed it will start disabling features until the boot loop problem is disabled in order to complete the boot cycle and leave you in control again.

Your use case makes Tasmota wake up and go to deepsleep in less than 10 seconds, so, you don't want the bootloop protection for your use case.

The command `websend` don't have answers, but from your example seems that you want something else different than the websend command. Please explain further or you can address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4) too. The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-25 by ascillato2

renegat56, 2020-09-25:
Thanks again, the response comes from my php file on my linux server, which I request with websend (echo "something" in php file, when it is requested by the websend command), so I send like http://myserver/php/iris.php?ubat=3.45 and because I enabled USE_WEBSEND_RESPONSE in Tasmota source, I get in return

RSL: /homenet/stat/iris/RESULT = DO1=1DO2=1UBAT=3.21 

or whatever I want. UBAT is the ESP-01 Vcc and not really needed in the response, but DO1 and DO2 should control the ESP-01 GPIO's, but for that I need to analyse the RSL/RESULT line. Is there a way to extract DO1=1 and DO2=1 with a script out of the response/payload and as example switch GPIO0 alias POWER1 alias DO1 and GPIO2 alias POWER2 alias DO2?

There might be also another way with mqtt, like check on my linux server, if ESP-01 is awake (via LWT topic) and then subscribe or publish something from/to the ESP, but this seems to be time critical.

I have the php file structure already. The php file writes and reads two other files in shared memory of my linux server-> written file contains the name=values pairs attached to the request and the read file the commands for the ESP returned as response. 
On the server side it is handled by apache and php file and on the ESP runs lua and it works mostly, but sometimes the ESP just stopped reacting. So I try it with Tasmota, because it works in other commercial plugs very reliable and I can learn more :-)

Sorry that I post here again, but in the chat I didn't got reply.

Jason2866, 2020-09-25:
Considering to install mqtt? It is easy to install, lightweigt and Tasmota is designed for.
Probably the goal you want to achive is easier and more reliable with mqtt.

renegat56, 2020-09-25:
@Jason2866 I have mqtt broker running on my linux server and I use it for commercial plugs with Tasmota and mosquitto or paho/python. But I like the other approach with php and webserver, because I think it is more flexible and I have just to serve a php file. The apache2 server is always listening and I use it also to access my plugs (and other modules/PC/NAS etc.) also from remote with https and dyn.IP. With reverse proxy this is very simple and secure. But maybe if I solved it my way, I want to use for it mqtt :-)

kugelkopf123, 2020-09-26:
@renegat56 What prints the console after you fired your websend command?

renegat56, 2020-09-26:
@kugelkopf123 

RSL: /homenet/stat/iris/RESULT = DO1=1DO2=1UBAT=3.21

Jason2866, 2020-09-26:
I said considering mqtt 😀. Speaking from PHP and secure is... brave
Securing mqtt is waaaay easier

kugelkopf123, 2020-09-26:
Tried something like:
```

>D
V1=““

;your websend stuff 

>E
V1=/homenet/stat/iris/RESULT#DO1

```

kugelkopf123, 2020-09-26:
Write it by yourself. Because the quotation marks are special chars as I post this from my mobile.

renegat56, 2020-09-26:
@kugelkopf123 

I tried it before and now again, unfortunately it is not working :-(

kugelkopf123, 2020-09-26:
Is there something in the v1 var after that?

Perhaps @gemu2015 has an idea.

gemu2015, 2020-09-26:
@renegat56 
your php must return a valid Jason string. (which your current is not)
then in >E you may read a variable from that json string (numeric or string)

like
/homenet/stat/iris/RESULT = {"DO1":0,"POWER":"on"}

\>D
v1=0
v2=""

....

\>E
v1=DO1
print %v1%
v2=POWER
print %v2%

renegat56, 2020-09-26:
@kugelkopf123 it was empty or contained the initial string 
>D
v1="X"

@gemu2015 I will try asap, up to now I manged it with a mixture of websend and paho python program which reacts on the topic message /homenet/tele/iris/LWT Online and publishs the desired state of power1 alias DO1 and power2 alias DO2. It also takes care of DeepSleepTime 0 and finally DeepSleepTime 60 to go to sleep for 60sec
I will report if your suggestion works, thanks

renegat56, 2020-09-26:
@gemu2015 

even the RSL: line looks now correct with json string, it is not working. There seems to be no way to use RESULT topics with scripting.
So I still don't know what the prupose of #define USE_WEBSEND_RESPONSE is.

kugelkopf123, 2020-09-26:
> @gemu2015
> 
> even the RSL: line looks now correct with json string, it is not working. There seems to be no way to use RESULT topics with scripting.
> So I still don't know what the prupose of #define USE_WEBSEND_RESPONSE is.

Could you please Post the Log from the Console with loglevel 4? 
Type: weblog 4 into the console. Then run your script. 
Perhaps that makes it easier to follow your problem.

gemu2015, 2020-09-27:
i verified with current firmware and it works like expected:
i sent a websend to a led stripes dimmer

\>D
cnt=25
wst=""

\>E
wst=WebSend
print >>> E:  %wst%

\>S
if upsecs%10==0
then
=>websend [192.168.178.84] dimmer %cnt%
cnt+=25
if cnt>=100
then
cnt=0
endif
endif

console output

06:54:02 Script: performs "websend [192.168.178.84] dimmer 25.00"
06:54:03 MQT: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":25,"Color":"00000040","HSBColor":"0,0,0","White":25,"Channel":[0,0,0,25]}
06:54:03 >>> E:  
06:54:03 MQT: stat/sonoff/RESULT = {"WebSend":"Done"}
06:54:03 >>> E:  Done

renegat56, 2020-09-27:
@gemu2015 

Thanks a lot for your efforts, but I do not see a which point you use in your script the response of a remote webserver after you send a websend to this server.

Websend command works also in my script, I also get the local RESULT  {"WebSend":"Done"}
My remote webserver (php script) also get the information of the websend command.

**Script: performs "websend [192.168.0.78:778] /dophp/iris.php?do1=1&do2=1&ubat=3.30"**

See console output in my first post. At that time the response looked like

**RSL: /homenet/stat/iris/RESULT = DO1=1DO2=1UBAT=3.30**

It returns to the tasmota esp-01 now a correct json string {"DO1":1,"DO2":1}, which apears in the ESP-01 as

**RSL: /homenet/stat/iris/RESULT = {"DO1":1,"DO2":1}**

I want to use this result and react on the value=1 of DO1 and the value=1 of DO2

Your line starts different as **MQT:** (mqtt), my starts with **RSL:** (whatever that means, maybe result)
Maybe this makes the difference.

gemu2015, 2020-09-27:
@renegat56 
please post your current script and console log
MQT or RSL should not make a difference

PS
i edited my previous post

renegat56, 2020-09-27:
@gemu2015 

here the script

>D
p:cnt=0
do1=0
do2=0
done=0

>B
cnt=cnt+1
=>print Counter %cnt%

>S
if wific>0
then
=>websend [192.168.0.78:778] /dophp/iris_json.php?ubat=%adc(10)%&info=%cnt%
done=1
endif

>E
if done==1
then
do1=/homenet/stat/iris/RESULT#DO1
print %do1%
do2=/homenet/stat/iris/RESULT#DO2
print %do2%
if do1==1
then
=>power1 1
endif
if do2==1
=>power2 1
endif
done=0
endif

Sorry, the console output I don't have right now, because I have now the workaround with mqtt broker running, see
https://github.com/arendst/Tasmota/issues/9385#issuecomment-699499310

In general it still looks like in first post, with the difference that the RSL line is now

RSL: /homenet/stat/iris/RESULT = {"DO1":1,"DO2":1}

Do you have any further suggestion regarding my script?

gemu2015, 2020-09-27:
look at my script!
your syntax is wrong!
do1=/homenet/stat/iris/RESULT#DO1
should be 
do1=D01

renegat56, 2020-09-27:
Sorry, this was suggestion of kugelkopf https://github.com/arendst/Tasmota/issues/9385#issuecomment-699470024
I think I tried both ways, but I will do again, thanks!

renegat56, 2020-09-28:
@gemu2015 
Thanks a lot for your support, you are right, with do1=DO1 in >E it is working! Now I know more how to script :-)

>D
p:cnt=0
do1=0
do2=0
dst=0
x=0

>S
x=x+1
if x==2
then
=>websend [192.168.0.78:778] /dophp/lilie_json.php?ubat=%adc(10)%&DST=%dst%&info=%cnt%
endif

if x==3
then
if do1==1
then
=>power1 1
endif
if do2==1
then
=>power2 1
endif
endif

if x==4
then
=>power1 0
=>power2 0
=>deepsleeptime %dst%
endif

>BS
=>deepsleeptime 0

>E
do1=DO1
do2=DO2
dst=DST

kugelkopf123, 2020-09-28:
backslash 
\>